aws4embeddedlinux/meta-aws #5: worker failed to initialize when NewDeployment

Jesse-Chang, 2020-01-14:
Hi,

I am following gg guide to setup ggc to device(qualcomm s820/yocto).
check_ggc_dependencies passed **except** optional package **python3.7/java8**.
I've disable stream manager since not support jdk8.
Also, select python2.7 when upload lambda hellow_world example.
When starting deployment, it returns **Deployment xxx of type NewDeployment for group xxx failed error: worker with xxx failed to initialize**
runtime.log show some Worker killed ungracefully and failed to restart.
already searching for related keywords, but still not workable solution for me.

check_ggc_dependency result and runtime.log as attached.
1. I tried config group connection settings from auto to manually but deployment stuck in status on going. runtime.log did not update anything.
2. device not support docker, so I set lambda container= none.
3. try disable ipv6 and enable ipv4 .ip_forward
4. add iotHttpPort=443 and ggHttpPort=443 to config.json still got same result.

[check_ggc_dependency.txt](https://github.com/aws/meta-aws/files/4056933/check_ggc_dependency.txt)
[runtime.log](https://github.com/aws/meta-aws/files/4056935/runtime.log)
[GGTES.log](https://github.com/aws/meta-aws/files/4056936/GGTES.log)

Thanks.
jc.

rpcme, 2020-01-14:
Could you let me know the BSP you are using?  Are you building in meta-aws or is this error from running on a baked image without meta-aws?

In the meantime, I will look at the runtime.log for hints.

Jesse-Chang, 2020-01-15:
Hi, rpcme

BSP version: yocto2.3 for apq8096
Since aws-meta got build error, I am using gg v1.10 from aws website.

thanks.
runtime log:  [runtime.log](https://github.com/aws/meta-aws/files/4062589/runtime.log)

Jesse-Chang, 2020-01-15:
**update**
I just build gg v1.9.2 from meta-aws success.
It seems same error with gg v1.10.
logs: 
[GGConnManager.log](https://github.com/aws/meta-aws/files/4062724/GGConnManager.log)
[runtime.log](https://github.com/aws/meta-aws/files/4062725/runtime.log)

rpcme, 2020-01-15:
Thank you very much.  Can you also let me know the hardware kit you are using?  I will try to acquire it in an expedited fashion.  If you are using custom hardware just let me know,

Is this the snapdragon 820a? 

Also, can you please give me `lsblk` output or something similar, as well as fstab?  I have a guess that it might be a filesystem related issue.

rpcme, 2020-01-15:
I just want to double confirm @Jesse-Chang that you are using meta-qcom and the machine definition is here.  http://git.yoctoproject.org/cgit/cgit.cgi/meta-qcom/tree/conf/machine/dragonboard-820c.conf?h=warrior (or that module matches the module you're using) - if so, I will 1/ ensure you can build with meta-aws with meta-qcom and 2/ figure out what is going on with Greengrass not running properly.

Jesse-Chang, 2020-01-16:
1. Yes, it's custom board with 820a, but i think it's close to Qualcomm reference design.
2. lsblk result: [lsblk.txt](https://github.com/aws/meta-aws/files/4068919/lsblk.txt)
3. machine definition: meta-qti-bsp/conf/machine/8x96auto.conf

Thanks.
jc.

rpcme, 2020-01-17:
Thank you for the information.  

ggc_user does not need permission.  It is used to operate Greengrass in a bounded security context.  Another option you can try for now is running Greengrass in "containerless" mode, which means the lambdas run as pure linux processes and do not run using cgroups and namespaces.  See https://docs.aws.amazon.com/greengrass/latest/developerguide/lambda-group-config.html and "Running a Lambda Function as Root".  This obviously is not ideal but we can then understand if it's something like cgroups or namespaces causing the problem.

I have ordered the mini reference kit https://shop.intrinsyc.com/products/adp-mini-mojave and will build the image next week given I am traveling today.  I appreciate your patience on this.  We will get to resolution on it.

rpcme, 2020-01-17:
Another thing @Jesse-Chang if you desire please connect to me separately at https://www.linkedin.com/in/richardelberger/ so we can speak more privately on your project.

Jesse-Chang, 2020-01-18:
Hi, since device not support docker, greengrass already run as containerless.
Will try run lambda function as root on monday and update result ASAP.
Thanks.

rpcme, 2020-01-22:
Offline, it was confirmed that no-container mode is working.  Once hardware received, will look for root cause of why cgroups are not configured properly in order for Greengrass lambdas to run.

rpcme, 2020-02-07:
CONFIG_ANDROID_PARANOID_NETWORK=n needs to be set in defconfig for qualcomm builds in order for greengrass to work in context of Qualcomm bsp.  A pull request will be issued soon.

rpcme, 2020-02-19:
@Jesse-Chang  - fixed by commit #45

rpcme, 2020-02-19:
I have put this on zeus and master branches, and will be put back farther later in the week.

closed 2020-02-19 by rpcme