aws4embeddedlinux/meta-aws #13: Add Greengrass HSI support for IoT Greengrass recipe

rpcme, 2020-02-07:
IoT Greengrass implements a feature named Hardware Security Integration, which enables Greengrass integration with qualified TPMs.  The configuration needs to be associated with configurable elements for a given implementation and the config.json needs to be created and staged for the desired given implementation.

PeterHuewe, 2020-09-18:
+1

Upstream Yocto already has the necessary TPM Software (tpm2-tss, tpm2-tools and tpm2-pkcs11) 
http://git.yoctoproject.org/cgit/cgit.cgi/meta-security/tree/meta-tpm/recipes-tpm2

rpcme, 2021-06-04:
Closing this issue given no new work will be done for Greengrass v1.x and new Greengrass recipe maintainers will handle when Greengrass 2.x supports secure hw

closed 2021-06-04 by rpcme