aws4embeddedlinux/meta-aws #18: Yocto recipe for Firecracker

rpcme, 2020-02-07:
Add recipe to build and install Firecracker.

Firecracker is an open source project supported by AWS.  From their landing page:

Firecracker is an open source virtualization technology that is purpose-built for creating and managing secure, multi-tenant container and function-based services that provide serverless operational models. Firecracker runs workloads in lightweight virtual machines, called microVMs, which combine the security and isolation properties provided by hardware virtualization technology with the speed and flexibility of containers.

Note: The Arm build is in preview.

https://github.com/firecracker-microvm/firecracker

bbros-dev, 2020-05-03:
The [`meta-virtualization`](http://git.yoctoproject.org/cgit/cgit.cgi/meta-virtualization/) layer may be of interest to others coming along this path.

HTH

jwinarske, 2021-05-23:
Firecracker Recipe: https://github.com/jwinarske/meta-microvm

Builds on NVIDIA Xavier NX BSP (aarch64)

rpcme, 2021-05-23:
@jwinarske nice! is there a reason why you just didn't do a PR to this layer?

jwinarske, 2021-05-23:
Keep in mind it brings in layer dependencies of meta-clang, and meta-rust.  I'm finding I like the approach of having them separate when target doesn't need rust.  Something like meta-aws-rs might be good.  But I can PR into here.

rpcme, 2021-05-23:
We can loosely couple the dependencies through the layer config - like we have for greengrass when wanting docker, specific runtimes, or other heavyweight dependencies.

Super appreciate it :)

jwinarske, 2021-05-24:
https://github.com/aws/meta-aws/pull/128

rpcme, 2021-06-10:
Fixed by #128 thank you @jwinarske!

closed 2021-06-10 by rpcme