aws4embeddedlinux/meta-aws #31: AWS Device SDK for C++ V1

rpcme, 2020-02-07:
Implement recipe for the AWS Device SDK for C++ V1

https://github.com/aws/aws-iot-device-sdk-cpp

BlessyNavara, 2020-08-26:
Hi.
Im building meta-aws layer onto imx-yocto-bsp image.

Toolchain :
bitbake - 1.44
Yocto distro - 5.4.24
added meta-aws layer
building for core-image-minimal target
machine - imx8m

All dependent layers(openembedded-core, meta-oe, meta-yocto-bsp, meta-filesystems, meta-python) are added to bblayers.conf.
IMAGE_INSTALL_append = " aws-iot-device-sdk-cpp-v2 aws-crt-cpp" added to layers.conf

    Is meta-aws having support for imx machines? Guide at github shows that in poky-zeus only qemu is supported. If yes, how to bitbake for imx machines?

following is the error log file content when running bitbake.(attached only 2/313 errors).

DEBUG: Executing shell function do_compile
NOTE: VERBOSE=1 cmake --build /home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/build --target all -- -j 8
[1/313] /home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++ -DAWS_AUTH_USE_IMPORT_EXPORT -DAWS_CAL_USE_IMPORT_EXPORT -DAWS_COMMON_USE_IMPORT_EXPORT -DAWS_COMPRESSION_USE_IMPORT_EXPORT -DAWS_CRT_CPP_EXPORTS -DAWS_CRT_CPP_USE_IMPORT_EXPORT -DAWS_HTTP_USE_IMPORT_EXPORT -DAWS_IO_USE_IMPORT_EXPORT -DAWS_MQTT_USE_IMPORT_EXPORT -DAWS_MQTT_WITH_WEBSOCKETS -DAWS_USE_EPOLL -DCJSON_HIDE_SYMBOLS -Daws_crt_cpp_EXPORTS -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-http/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-io/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/s2n -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/s2n/api -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot-native/usr/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-common/include -Iaws-common-runtime/aws-c-common/generated/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-compression/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-mqtt/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-cal/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-auth/include -mcpu=cortex-a53+crc+crypto -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0=/usr/src/debug/aws-crt-cpp/0.8.3-r0 -fdebug-prefix-map=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0=/usr/src/debug/aws-crt-cpp/0.8.3-r0 -fdebug-prefix-map=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot= -fdebug-prefix-map=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot-native= -fvisibility-inlines-hidden -mcpu=cortex-a53+crc+crypto -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot -O2 -g -DNDEBUG -fPIC -Wall -Wno-long-long -pedantic -Werror -DS2N_NO_PQ_ASM -DS2N_HAVE_EXECINFO -fPIC -pthread -std=gnu++11 -MD -MT CMakeFiles/aws-crt-cpp.dir/source/StringUtils.cpp.o -MF CMakeFiles/aws-crt-cpp.dir/source/StringUtils.cpp.o.d -o CMakeFiles/aws-crt-cpp.dir/source/StringUtils.cpp.o -c /home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/source/StringUtils.cpp
[2/313] /home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++ -DAWS_AUTH_USE_IMPORT_EXPORT -DAWS_CAL_USE_IMPORT_EXPORT -DAWS_COMMON_USE_IMPORT_EXPORT -DAWS_COMPRESSION_USE_IMPORT_EXPORT -DAWS_CRT_CPP_EXPORTS -DAWS_CRT_CPP_USE_IMPORT_EXPORT -DAWS_HTTP_USE_IMPORT_EXPORT -DAWS_IO_USE_IMPORT_EXPORT -DAWS_MQTT_USE_IMPORT_EXPORT -DAWS_MQTT_WITH_WEBSOCKETS -DAWS_USE_EPOLL -DCJSON_HIDE_SYMBOLS -Daws_crt_cpp_EXPORTS -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-http/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-io/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/s2n -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/s2n/api -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot-native/usr/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-common/include -Iaws-common-runtime/aws-c-common/generated/include -I/home/imx-yocto-bsp/build_di [... cut ...] DEBUG -fPIC -Wall -Wno-long-long -pedantic -Werror -DS2N_NO_PQ_ASM -DS2N_HAVE_EXECINFO -fPIC -pthread -std=gnu++11 -MD -MT CMakeFiles/aws-crt-cpp.dir/source/Api.cpp.o -MF CMakeFiles/aws-crt-cpp.dir/source/Api.cpp.o.d -o CMakeFiles/aws-crt-cpp.dir/source/Api.cpp.o -c /home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/source/Api.cpp

ninja: build stopped: subcommand failed.
WARNING: exit code 1 from a shell command.

@rpcme:
Any working solution is much appreciated.

rpcme, 2021-02-14:
Hello @BlessyNavara I am so sorry I missed your message!  We will not be supporting v1 for new designs.  I am going to close this out. If you have any issue or need this supported on a specific release branch, please let me know.  Please use the V2 SDK.

closed 2021-02-14 by rpcme