aws4embeddedlinux/meta-aws #41: Greengrass dependency error for linux-ti-staging_%.bbappend

kennu, 2020-02-12:
Hello. I get this error when trying to bitbake my project. I use the meta-aws layer and try to install the greengrass package in my IMAGE_INSTALL_append.

    ERROR: No recipes available for:
    meta-aws/recipes-greengrass/greengrass-core/linux-ti-staging_%.bbappend

I have included these layers in my project BBLAYERS, as instructed in the README, along with several other layers that I need.

    meta-yocto/meta-yocto-bsp
    meta-openembedded/meta-oe
    meta-openembedded/meta-filesystems
    meta-openembedded/meta-python

By searching for the missing package, I found https://git.yoctoproject.org/cgit/cgit.cgi/meta-ti/tree/recipes-kernel/linux/linux-ti-staging_5.4.bb but it's not mentioned in Greengrass instructions.

Should the **meta-ti** layer be documented as an additional dependency?

rpcme, 2020-02-12:
Hello @kennu - could you let me know if you're building in the context of TI Processor SDK?  This should be picked up only when running in that context.

If you're running in that context, then yes it should be a documented dependency (and should be documented anyway), but it should also already exist in your source checkout.

For the non TI case, this bbappend should not be used. I might then have missing configuration to ensure the patch doesn't get run when building outside the TI processor sdk.

kennu, 2020-02-12:
I don't think my project has anything related to TI in it. The target machine is SolidRun CuBox (i.MX6). The other layers being used look like this:

    ${BSPDIR}/sources/poky/meta \
    ${BSPDIR}/sources/poky/meta-poky \
    ${BSPDIR}/sources/meta-freescale \
    ${BSPDIR}/sources/meta-freescale-3rdparty \
    ${BSPDIR}/sources/meta-freescale-distro \
    ${BSPDIR}/sources/meta-openembedded/meta-oe \
    ${BSPDIR}/sources/meta-openembedded/meta-multimedia \
    ${BSPDIR}/sources/meta-openembedded/meta-networking \
    ${BSPDIR}/sources/meta-openembedded/meta-filesystems \
    ${BSPDIR}/sources/meta-openembedded/meta-python \
    ${BSPDIR}/sources/meta-virtualization \
    ${BSPDIR}/sources/meta-yocto/meta-yocto-bsp \

rpcme, 2020-02-12:
Ok, I understand.  I will reproduce with the layers you have in the above list and provide the necessary fix. Will not be able to reproduce it until my tomorrow morning - GMT+5. You should be able to continue by removing linux-ti-staging_%.bbappend.  The linux-yocto_%.bbappend file would be applicable in your case.

Thank you very much for reporting - I know others have built for Qualcomm and QEMU but maybe they "fixed it" without reporting.

kennu, 2020-02-13:
Thanks, my build actually worked by adding the *meta-ti* layer temporarily (I guess it doesn't affect anything when no packages from it are used). I'll remove it later on.

rpcme, 2020-02-13:
Thank you for the clarification.  The patch should only be used when the distro is arago.  I am glad you were able to continue.

I will take some time to identify a strong approach to choosing the correct append file.  It will likely revolve around DISTRO being arago.

Until I verify this approach, I won't yet taint the documentation and will leave this issue open.

kennu, 2020-02-18:
Strange thing happened: As I'm building my project today, the meta-ti layer now complains that a zeus branch doesn't exist and my workaround stopped working.  Not sure how it worked in the first place, unless someone actually deleted that branch from meta-ti during the weekend.

For now I changed my workaround to this in my Dockerfile:

RUN rm -f ${YOCTO_INSTALL_PATH}/fsl-community-bsp/sources/meta-aws/recipes-greengrass/greengrass-core/linux-ti-staging_%.bbappend

rpcme, 2020-02-19:
Hello @kennu, I believe I have resolved this problem.  Please use the zeus branch.  I will be proliferating the change across branches once confirmed.  I checked this with a plain poky build.

kennu, 2020-02-20:
My build is working now with the zeus branch without any workarounds, thanks @rpcme.

closed 2020-02-20 by kennu