aws4embeddedlinux/meta-aws #50: Deployment container test failed - mkdir tmp permission denied

awsblake, 2020-03-25:
Chipset: Qualcomm QCS603
BSP: Yotco Linux with Qualcomm BSP QCS605
Greengrass version:  greengrass-linux-armv7l-1.10.0.tar.gz
Fail case: deploymentcontainer
Failure log: mkdir /testGreengrass/greengrass/ggc/packages/1.10.0/rootfs/merged/var/volatile/tmp: permission denied

workaround: create an actual /tmp directory
unlink /tmp
mkdir /tmp
chmod 777 /tmp

rpcme, 2020-03-26:
@awsblake thank you for filing this issue.  I am currently evaluating for a proper fix - we want to ensure that `tmp` remains a `tmpfs` to promote flash longevity.

rpcme, 2020-04-09:
A fix will be made to the Greengrass version-specific recipe.  It was verified that the /tmp directory mount isn't being used so there will be no flash impact.

AltekCllin, 2020-04-10:
Other features in our project use the tmp a symbolic link, the workaround(unlink tmp) solution is high risk.
So please kindly provide other solutions.
Thanks.

rpcme, 2020-04-10:
CLLin,

Excellent feedback.  We do not want to put your project in a high risk situation.  Thank you for your valuable feedback!

jbutler, 2020-04-10:
Greengrass team member here. We have root caused this issue and are working on a proper fix. In the meantime, another workaround is to ensure the directory the tmp symlink points to exists on the root filesystem itself, rather than just within the tmpfs. 

For example:

```
# WORKAROUND - create tmp directory before mounting tmpfs
mkdir -p /var/volatile/tmp

# Set up tmp symlinks
ln -s /var/tmp /tmp  # /tmp -> /var/tmp
ln -s /var/volatile/tmp /var/tmp  # /var/tmp -> /var/volatile/tmp

# Finally, mount tmpfs and recreate tmp
mount -t tmpfs tmpfs /var/volatile
mkdir /var/volatile/tmp
```

Thanks,
-joe

rpcme, 2020-04-10:
@AltekCllin can you let us know which possible workaround works for you? We will add it to the recipe.

AltekCllin, 2020-04-13:
If greengrass team has co-worked with QTI Team to get any workaround.I can accept any workaround. Or I hope to keep the original tmp and not apply workaround on tmp .

Because I do not know why normal deploy is OK but container deploy fails and greengrass use tmp to do what,I do not know which possible workaround is low risk.

Could greengrass team could co-work with QTI Team to obtain proper fix or workaround?

rpcme, 2020-04-13:
@AltekCllin Thank you for your response. Your assertions will be embraced when implementing the solution to the recipe.

rpcme, 2020-04-20:
@AltekCllin I have confirmed with QTI that the approach by @jbutler is agreeable. A fix will be applied for specific Greengrass versions.

AltekCllin, 2020-04-20:
If the new Greengrass version recipe is released,please inform me to download.
Thanks you.

rpcme, 2020-09-16:
@AltekCllin @jbutler  @awsblake  This should be fixed for 1.11, which is on /dunfell/ and /master/ branches.

closed 2020-09-16 by rpcme