aws4embeddedlinux/meta-aws #56: dunfell branch

codrin989, 2020-04-27:
Hi @rpcme ,

Now with the yocto dunfell release, do you plan to create a new dunfell branch? I know it's an extra branch to maintain, but the master branch will be used with newer yocto packages and meta-aws might need new patches to support them, while dunfell will get only fixes.
Thanks!

rpcme, 2020-04-28:
Hello @codrin989 Sorry for the delay, I have been on leave.

Yes will do, lovely that the release happened when I was on leave :) thanks for the heads up. When I get to my desk this morning, I will fork.

codrin989, 2020-04-28:
There is no delay, I just opened the 'iissue' last day. Thanks! :)

rpcme, 2020-04-28:
This branch has now been created.

closed 2020-04-28 by rpcme

codrin989, 2020-04-28:
Thanks @rpcme !