aws4embeddedlinux/meta-aws #59: warrior: No recipes available for linux-ti-staging_%.bbappend

mkozjak, 2020-05-04:
@rpcme: I'm tracking warrior and I just got this on first pass with `bitbake core-image-base`:

```
ERROR: No recipes available for:
  /home/user/yocto/sources/meta-aws/recipes-greengrass/greengrass-core/linux-ti-staging_%.bbappend
```

If I remove the file it passed and then failed on:

```
ERROR: Nothing RPROVIDES 'aws-iot-device-sdk-python-v2' (but /home/user/yocto/sources/poky/meta/recipes-core/images/core-image-base.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'aws-iot-device-sdk-python-v2' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['aws-iot-device-sdk-python-v2']
ERROR: Required build target 'core-image-base' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-base', 'aws-iot-device-sdk-python-v2']
```

The `aws-iot-device-sdk-python-v2` recipe is mandatory for me and yet it's not in `warrior`.

My `IMAGE_INSTALL_append` includes these:
```
aws-iot-device-sdk-python-v2 amazon-cloudwatch-publisher
```

rpcme, 2020-05-06:
@mkozjak I have fixed the staging file issue.

The cloudwatch recipe depends on 'python3-boto3' and I am not sure where you intended to get that?

mkozjak, 2020-05-06:
@rpcme thanks! I use this one for `python3-boto3`: https://github.com/bachp/meta-homeassistant/blob/master/recipes-devtools/python/python3-boto3_1.12.29.bb.

rpcme, 2020-05-06:
Interesting thanks.  We intend to have this sdk in this meta layer since it's AWS delivered software, so we should probably weight it accordingly.  Ideally, we would want to simplify the number of layers required by customers.  Do you see an issue with that?

mkozjak, 2020-05-06:
Yup, it'd be nice to have those here. `python3-boto3`, `python3-botocore`, `python3-jmespath`, `python3-s3transfer` and `python3-timeloop` and the needed ones for the cloudwatch recipe, for example.

mkozjak, 2020-05-17:
@rpcme any updates on this one?

rpcme, 2020-05-17:
@mkozjak it's on my list for this week.  We had some refactoring to do for x86_64 that stole some cycles this week.  I do intend to pull any dependent recipes not resident in meta-python to this repo, and then over time upstream those not directly resulting from aws delivered software. I apologize if I didn't pick up on the urgency of this request.

rpcme, 2020-08-29:
@mkozjak I am going to close this and create new issue for tracking ok?

closed 2020-08-29 by rpcme