aws4embeddedlinux/meta-aws #60: greengrassSystemComponents needs ELF patch for x86_64

kriskep, 2020-05-06:
Binary file `ggc/core/lambda/GreengrassSystemComponents/greengrassSystemComponents`  needs patchelf to --set-interpreter to /lib/... instead of /lib64/... for x86_64 arch.

rpcme, 2020-05-06:
Hi @kriskep, please check 1.10.1 that I just committed. there is an issue with the previous fix where it was specific to qemu.    Right now I have it only on master but can pull it down to zeus or whatever, depending on the immediate need.

kriskep, 2020-05-11:
Hi @rpcme, there is second binary that needs patchelf. Please check my pull request #65

rpcme, 2020-05-11:
Thank you! I think this should resolve extending issues on this topic. If there's more issues, please share.

closed 2020-05-11 by rpcme

kriskep, 2020-05-13:
Hi @rpcme, unfortunately, there is yet another binary that needs this patch:
`/greengrass/ggc/packages/1.10.1/lambda/arn.aws.lambda...function.GGIPDetector.1`

It only shows up when you start to deploy things. 
**Maybe it is safer to create a symlink in /lib64. What do you think?** It would also save us from future changes in the binary package.