aws4embeddedlinux/meta-aws #69: Worker is ungracefully killed issue, lambda function not running properly after deployed

guntshi, 2020-07-01:
Hi,

I am using instructions from 
https://github.com/aws-samples/meta-aws-demos/blob/master/am572x_idk/aws_iot_greengrass/README.md

to build and setup a new variant of a target (AM437x EVM) with Yocto "zeus" based Greengrass. The greengrassd is starting correctly. Using the GG Getting Started Docs Module 2 and 3 and deploying a group with the HelloWorld Lambda function causes an issue, the var/log/system/runtime.log shows "Worker is ungracefully killed". 

Using GGC 1.10.1-RC3, the greengrassd runs from root on the target.

From the runtime.log
[2020-07-01T19:58:08.063Z][INFO]-Created worker.	{"functionArn": "arn:aws:lambda:us-west-2:630221750224:function:Greengrass_HelloWorld:2", "workerId": "d174f4c7-3b26-4a7d-6e1e-35d7a0de9e12", "pid": 9037}
[2020-07-01T19:58:08.789Z][INFO]-Worker Memory Usage (KB).	{"workerId": "d174f4c7-3b26-4a7d-6e1e-35d7a0de9e12", "funcArn": "arn:aws:lambda:us-west-2:630221750224:function:Greengrass_HelloWorld:2", "memSize": 16384, "memUsed": 12144}
[2020-07-01T19:58:08.789Z][ERROR]-Worker is ungracefully killed.	{"workerId": "d174f4c7-3b26-4a7d-6e1e-35d7a0de9e12", "funcArn": "arn:aws:lambda:us-west-2:630221750224:function:Greengrass_HelloWorld:2", "state": "exit status 1"}

From the GGTES.log
root@am437x-evm:/greengrass/ggc/var/log/system# less GGTES.log 
[2020-07-01T17:40:07.505Z][INFO]----- Lambda runtime configuration ----
[2020-07-01T17:40:07.505Z][INFO]-AWS_GG_HTTP_ENDPOINT: greengrass-ats.iot.us-west-2.amazonaws.com:8443
[2020-07-01T17:40:07.505Z][INFO]-GG_HTTP_TIMEOUT: 60
[2020-07-01T17:40:07.969Z][INFO]-function is initializing ...   {"functionName": "arn:aws:lambda:::function:GGTES"}
[2020-07-01T17:40:07.986Z][INFO]-function is initialized, starting to process work items ...    {"functionName": "arn:aws:lambda:::function:GGTES"}
[2020-07-01T17:40:09.291Z][INFO]-Caught signal, stopping runtime        {"signal": "terminated"}

Attaching the runtime.log

[runtime.log.am4376-zeus-ggc110-python37.txt](https://github.com/aws/meta-aws/files/4860340/runtime.log.am4376-zeus-ggc110-python37.txt)

Do you have any pointers of what could be happening?

Thanks!
--Gunter

rpcme, 2020-07-01:
Hello Gunter, I have the part on hand.  Let me see if I can reproduce the issue. I can start building the image today but I won't be able to go through the tutorial until tomorrow.  Note that the AWS IoT Device Tester for IoT Greengrass was run against the image which tests similar Lambda deployment scenarios.

guntshi, 2020-07-01:
Thanks, Richard for testing this! 
Please let us know if you need additional HW from TI, I can help there.

Also, let me check if I can get the IOT Device Tester (IDT) going on this target.

Regards,
--Gunter

guntshi, 2020-07-07:
Hi Richard, just wanted to check if you were able to re-produce the issue on your side?

Thanks,
--Gunter

rpcme, 2020-07-08:
Hello @guntshi - I walked through the tutorial twice - once last Thursday and again this morning just to make sure I didn't go too fast and miss something.  The tutorial succeeded.  Here is what I did:

1. Built image
2. Flashed image
3. Booted image.
4. From workstation, create a new Greengrass group.  NOTE.  Do not follow tutorial Module 1 and Module 2.  The software is already installed.  
    1. You need to just to Greengrass Group quick create from the AWS Console.  Unfortunately I can no longer find the instruction for that in our documentation.
    2. When you do quick create, you download put it on USB, put USB to TI board, and copy credential.  Of course other procedures apply when doing this at scale/in production, but this is for prototyping.  When I did this, I downloaded the credential, put in the usb, and invoked these commands:

        ```bash
tar -xzvf ~/Downloads/7681c8a294-setup.tar.gz
wget https://www.amazontrust.com/repository/AmazonRootCA1.pem
        ```

    3. Then, after ejecting USB and putting it to the EVK, I issued the following:

        ```bash
cp /run/media/sda1/certs/* /greengrass/certs
cp /run/media/sda1/config/* /greengrass/config/
cp /run/media/sda1/AmazonRootCA1.pem /greengrass/certs/root.ca.pem
/greengrass/ggc/core/greengrassd start
        ```

5.  Follow Module 3 part 1 explicitly.  Turn off Stream Manager (bottom of https://docs.aws.amazon.com/greengrass/latest/developerguide/config-lambda.html) because, by default, Java 8 is not built.

rpcme, 2020-07-08:
Okay to be more clear about (4). Please do not do the "Quick Start" https://docs.aws.amazon.com/greengrass/latest/developerguide/quick-start.html

Module 1 is covered off by the yocto build.
Module 2 is where you should start.  This is the instruction that I said was "missing" but it wasn't. Sorry for the confustion.

guntshi, 2020-07-08:
Hi Richard, thanks for the instructions and testing. Good to know that it works on your side. Are you running on an AM437x EVM or Starterkit?

Looking in detail at your instructions, this looks like exactly what I have done on my side as well.
After building the Yocto image with greengrass recipes included, I have also completely skipped Module 1, since the greengrassd is installed.
Started with Module 2 to create the Group and downloaded the tar ball with credentials and transferred to the target including the Root CA, and copied into the relevant directories certs and config. (This seems to work since the AWS IOT Core during deployment does connect to the GG Group later). The greengrassd starts correctly.
Then in detail went through Module 3 Part 1 as you said, to make sure I did not miss a step. Again all this goes well including the deployment of the Group and Lambda function. AWS IOT reports "successfully completed" after deployment.

The problem is when I test the Lambda function by subscribing to the topic hello/world, the messages do not show up.

Looking at the runtime.log, the lambda function was getting this "ungracefully killed" error.

Could you send your runtime.log for a successful run?

I will try to go through every step again in detail.

Thanks,
--Gunter

rpcme, 2020-07-08:
I used the TMDSIDK574 kit for this test. But I don't think it has anything to do with the Yocto build.

It might be that the zip file is not formed correctly.  When doing this walk-through, I realized that the instruction for creating the Lambda Zip isn't entirely clear.  What I did was this:

```bash
cd ~/Downloads
wget https://github.com/aws/aws-greengrass-core-sdk-python/archive/v1.5.0.tar.gz
tar xzf v1.5.0.tar.g
mkdir HelloWorld
cp aws-greengrass-core-sdk-python-1.5.0/examples/HelloWorld/greengrassHelloWorld.py HelloWorld/
cp -r aws-greengrass-core-sdk-python-1.5.0 HelloWorld/greengrasssdk
cd HelloWorld/
zip -r hello_world_python_lambda.zip greengrasssdk greengrassHelloWorld.py
```

It might pop if you don't build the lambda correctly, and then invoke a function entry point. 

I already put the board back in storage.  Hopefully this is the issue, if not I will have to pull it out to get the logs.

rpcme, 2020-07-17:
Hello @guntshi do you have any update on progress?  I would like to know if you figured it out or if you're still stuck.

guntshi, 2020-07-20:
Hi Richard, thanks again for testing this on your side. Carefully reviewing your steps, my steps were as far as I could tell exactly the same. I was still running into the issue on the AM437x EVM Starterkit though. So I decided to test on the AM57xx IDK HW with the am57xx-evm yocto build just as you did to see if there are any differences. Doing this today, hopefully will have more results shortly.

rpcme, 2020-07-29:
Hello @guntshi I'm looking forward to your result.

guntshi, 2020-07-31:
Hi Richard, (due to the COVID restrictions) I had an issue getting access to the AM57xx kit, which slowed me down unfortunately, but as of today I have access to it now and verified it is booting correctly. Sorry it took longer ... Going now exactly through your steps to reproduce your result and will update you as soon as I can.

rpcme, 2020-07-31:
Sounds great @guntshi - looking forward to seeing positive resolution :)

guntshi, 2020-08-03:
Hi Richard,

unfortunately I am getting the same behavior with the AM57xx EVM setup, and Yocto build for that. Everything works going carefully through all the steps on the AWS docs, including deploying the Group with Lambda function (getting successfully deployed). But subscribing to the topic hello/world does not show any MQTT messages from the Core and the runtime.log looks the same, see here:

root@am57xx-evm:/greengrass/ggc/var/log/system# less runtime.log
[2020-08-03T23:07:31.085Z][INFO]-===========================================
[2020-08-03T23:07:31.087Z][INFO]-Greengrass Version: 1.10.1-RC3
[2020-08-03T23:07:31.087Z][INFO]-Greengrass Root: /greengrass
[2020-08-03T23:07:31.087Z][INFO]-Greengrass Write Directory: /greengrass/ggc
[2020-08-03T23:07:31.087Z][INFO]-Group File Directory: /greengrass/ggc/deployment/group
[2020-08-03T23:07:31.087Z][INFO]-Default Lambda UID: 998
[2020-08-03T23:07:31.087Z][INFO]-Default Lambda GID: 998
[2020-08-03T23:07:31.091Z][INFO]-===========================================
[2020-08-03T23:07:31.091Z][INFO]-The current core is using the AWS IoT certificates with fingerprint.   {"fingerprint": "111e3d16638a73eac591a71b1fc4c166d5cdfcef703f237ecd7e528984402bd9"}
[2020-08-03T23:07:31.091Z][INFO]-Will persist worker process info.      {"dir": "/greengrass/ggc/packages/1.10.1/var/worker/processes"}
[2020-08-03T23:07:31.092Z][INFO]-Will persist worker process info.      {"dir": "/greengrass/ggc/packages/1.10.1/var/worker/processes"}
[2020-08-03T23:07:31.141Z][INFO]-Subscription added to the system       {"source": "arn:aws:lambda:us-west-2:630221750224:function:Greengrass_HelloWorld:4", "sourceType": 0, "subject": "hello/world", "tar
get": "cloud", "targetType": 2}
[2020-08-03T23:07:31.173Z][INFO]-No proxy URL found.
[2020-08-03T23:07:31.195Z][INFO]-The current core is using the AWS IoT certificates with fingerprint.   {"fingerprint": "111e3d16638a73eac591a71b1fc4c166d5cdfcef703f237ecd7e528984402bd9"}
[2020-08-03T23:07:31.197Z][INFO]-Setting up overlay tmpfs mounts
[2020-08-03T23:07:31.298Z][INFO]-Finding mounted cgroups.
[2020-08-03T23:07:31.298Z][WARN]-Link to DNS resolve file does not exist.       {"errorString": "lstat /etc/../run/systemd/resolve/resolv.conf: no such file or directory"}
[2020-08-03T23:07:31.312Z][INFO]-Starting file sync.    {"src": "/greengrass/ggc/resolvtmp.conf", "dst": "/greengrass/ggc/packages/1.10.1/dns/etc/resolv.conf"}
[2020-08-03T23:07:31.326Z][INFO]-Started Deployment Agent to listen for updates.
[2020-08-03T23:07:31.349Z][INFO]-Connecting with MQTT.  {"endpoint": "aokw42bc0infj-ats.iot.us-west-2.amazonaws.com:8883", "clientId": "MyAm571xGroup_Core"}
[2020-08-03T23:07:31.377Z][WARN]-MQTT[client]   dial tcp: lookup aokw42bc0infj-ats.iot.us-west-2.amazonaws.com on [::1]:53: read udp [::1]:43226->[::1]:53: read: connection refused

[2020-08-03T23:07:31.377Z][WARN]-MQTT[client]   Failed to connect to a broker

[2020-08-03T23:07:32.326Z][INFO]-Found DNS resolve file.
[2020-08-03T23:07:32.327Z][INFO]-Stop syncing file.     {"src": "/greengrass/ggc/resolvtmp.conf", "dst": "/greengrass/ggc/packages/1.10.1/dns/etc/resolv.conf"}
[2020-08-03T23:07:32.338Z][INFO]-Starting file sync.    {"src": "/etc/resolv.conf", "dst": "/greengrass/ggc/packages/1.10.1/dns/etc/resolv.conf"}
[2020-08-03T23:07:32.545Z][INFO]-Received file notify event: "/etc/resolv.conf": CHMOD
[2020-08-03T23:07:32.545Z][INFO]-Received file notify event: "/etc/resolv.conf": REMOVE
[2020-08-03T23:07:32.667Z][INFO]-Received file notify event: "/etc/resolv.conf": CHMOD
[2020-08-03T23:07:32.667Z][INFO]-Received file notify event: "/etc/resolv.conf": REMOVE
[2020-08-03T23:07:34.782Z][INFO]-Created worker.        {"functionArn": "arn:aws:lambda:::function:GGShadowService", "workerId": "cb627f3c-c9cf-47eb-6602-80f42f4faed2", "pid": 581}
[2020-08-03T23:07:35.583Z][INFO]-Created worker.        {"functionArn": "arn:aws:lambda:::function:GGShadowSyncManager", "workerId": "65897a52-7895-4aaa-5408-80110dfa13e3", "pid": 615}
[2020-08-03T23:07:36.294Z][INFO]-Received file notify event: "/etc/resolv.conf": CHMOD
[2020-08-03T23:07:36.294Z][INFO]-Received file notify event: "/etc/resolv.conf": REMOVE
[2020-08-03T23:07:36.828Z][INFO]-Received file notify event: "/etc/resolv.conf": CHMOD
[2020-08-03T23:07:36.828Z][INFO]-Received file notify event: "/etc/resolv.conf": REMOVE
[2020-08-03T23:07:37.753Z][INFO]-Received file notify event: "/etc/resolv.conf": CHMOD
[2020-08-03T23:07:37.753Z][INFO]-Received file notify event: "/etc/resolv.conf": REMOVE
[2020-08-03T23:07:51.864Z][INFO]-Received file notify event: "/etc/resolv.conf": CHMOD
[2020-08-03T23:07:51.864Z][INFO]-Received file notify event: "/etc/resolv.conf": REMOVE
[2020-08-03T23:07:53.166Z][INFO]-Received file notify event: "/etc/resolv.conf": CHMOD
[2020-08-03T23:07:53.167Z][INFO]-Received file notify event: "/etc/resolv.conf": REMOVE
[2020-08-03T23:07:53.396Z][INFO]-Created worker.        {"functionArn": "arn:aws:lambda:::function:GGDeviceCertificateManager", "workerId": "8fd88ecd-3169-422b-45e4-64f1d2595432", "pid": 811}
[2020-08-03T23:07:53.422Z][INFO]-Created worker.        {"functionArn": "arn:aws:lambda:::function:GGCloudSpooler:1", "workerId": "b18205a6-59c5-47d2-7e65-667eca901278", "pid": 846}
[2020-08-03T23:07:53.49Z][INFO]-Created worker. {"functionArn": "arn:aws:lambda:::function:GGTES", "workerId": "64e52661-8d1e-4b1c-73a4-da0b3862a664", "pid": 852}
[... cut ...]
[2020-08-03T23:07:55.355Z][INFO]-Worker Memory Usage (KB).      {"workerId": "4d1cb286-65f4-44f9-5854-fb58e1af2178", "funcArn": "arn:aws:lambda:us-west-2:630221750224:function:Greengrass_HelloWorld:4", "m
emSize": 16384, "memUsed": 12148}
[2020-08-03T23:07:55.355Z][ERROR]-Worker is ungracefully killed.        {"workerId": "4d1cb286-65f4-44f9-5854-fb58e1af2178", "funcArn": "arn:aws:lambda:us-west-2:630221750224:function:Greengrass_HelloWorl
d:4", "state": "exit status 1"}

Is there anything that looks suspicious here, in comparison to your setups?

Thanks!
--Gunter

rpcme, 2020-08-13:
@guntshi Sorry for the late response, I was negatively impacted by the storms on the east coast US last week.

For the lambda function, is there any user lambda function log?  It is bizarre that hello world would fail.  Usually some kind of die on impact there would be interpreter related, which would mean somehow you missed getting Python installed?

And the resolv.conf file thing is a little strange, I don't recall that but usually I don't check the log unless there are some challenges.

guntshi, 2020-08-25:
@rpcme Sorry it took me longer to recreate everything, but this time I created a doc to show all the steps being taken, which I am attaching below. 

For the lambda function on the target I don't see anything being printed. 
The python 3.7 is installed
root@am437x-evm:/etc# python3 --version
Python 3.7.7

The nameserver seems to be working too, as I can e.g. ping google.com

Could you take a look at my greengrassInstallation.doc file, to see if there is anything out of the order. This time I have gone through every step again in detail to make sure nothing is missing and logged everything. Please let me know if you see something.

It would be great to see a runtime.log from a working setup, to see where the differences are.

Regards,
--Gunter

[greengrassInstallation.tar.gz](https://github.com/aws/meta-aws/files/5121309/greengrassInstallation.tar.gz)

guntshi, 2020-08-25:
Hi @rpcme 
I realized, the attached doc format was corrupted, so I am re-attaching that document. Please let me know if there are any problems accessing it.

Thanks!
--Gunter

[greengrassInstallation.zip](https://github.com/aws/meta-aws/files/5126411/greengrassInstallation.zip)

rpcme, 2020-09-01:
@guntshi sorry I am on vacation so responses will be delayed. I won't be back on my workstation until the weekend. I will be able to look at it at that time.

guntshi, 2020-09-10:
Hi @rpcme 
a colleague of mine in parallel went through all the steps on his side in detail and was able to replicate the same problem, I had documented in the doc I had attached earlier. This was done on an independent setup, so it seems something fundamental is missing. 

By the fact that we are building the GG core into the Yocto filesystem with meta-aws, we are wondering if something is missing there. I am thinking that we have a running Greengrass Core device on the target, but e.g. the Module 3 Part 1 has a "Tip" in the very beginning. https://docs.aws.amazon.com/greengrass/latest/developerguide/module3-I.html
TIP:
"Or, to use a script that sets up your core device for you, see Quick start: Greengrass device setup. The script can also create and deploy the Lambda function used in this module."

Is this something that needs to be done?

Thanks, 
--Gunter

guntshi, 2020-09-17:
Hi @rpcme 
it is working now!
I was able to figure it out, there was a dependency problem with python3 and missing files in /usr/lib/python3.7 on the target. This manifested itself in the Lambda logs which e.g. showed missing base64.py, which should have been installed by default, but wasn't. Since I did not trust the python3 yocto RDEPENDS in your meta-aws greengrass recipe, I made the following change to add full python3:

diff --git a/recipes-greengrass/greengrass-core/greengrass_1.10.1.bb b/recipes-greengrass/greengrass-core/greengrass_1.10.1.bb
index 1ed4082..9385e83 100644
--- a/recipes-greengrass/greengrass-core/greengrass_1.10.1.bb
+++ b/recipes-greengrass/greengrass-core/greengrass_1.10.1.bb
@@ -29,7 +29,7 @@ SRC_URI[x86-64.sha256sum]  = "8fded584f9291510ee91fe98cfd8bc69e01d1b8e4147f24fa1
 
 # Release specific configuration
 
-RDEPENDS_${PN} += "ca-certificates python3-json python3-numbers sqlite3"
+RDEPENDS_${PN} += "ca-certificates python3 python3-json python3-numbers sqlite3"
 
 do_install_append_x86-64() {
     # create symbolic link /lib64/ld-linux-x86-64.so.2 to enable loading the binary

Rebuilding with that, the filesystem now has everything in python3 usr libs and greengrass gets everything it needs from python. With that it started working.

Regards,
--Gunter

rpcme, 2020-11-25:
Thank you for relaying that information.  I will cross check this across all branches!

closed 2020-11-25 by rpcme

RX1208, 2021-11-16:
@rpcme 
excuse me,
Because I don’t speak English, I use Google Translate now.
Please allow me to describe my problem
I am also using AWS IoT Greengrass to deploy lambda 
but  Worker is ungracefully killed
Is there any relevant help documentation?
Below is my log

[2021-11-15T16:44:37.669-10:00][INFO]-Created worker.   {"functionArn": "arn:aws:lambda:ap-northeast-1:291538252294:function:test1:2", "workerId": "177c1397-*******-713a-ee43cf90d46f", "pid": 18799}
[2021-11-15T16:44:38.316-10:00][INFO]-Worker Memory Usage (KB). {"workerId": "177c1397-*******-713a-ee43cf90d46f", "funcArn": "arn:aws:lambda:ap-northeast-1:*********:function:test1:2", "memSize": 102400, "memUsed": 15204}
[2021-11-15T16:44:38.316-10:00][ERROR]-Worker is ungracefully killed.   {"workerId": "177c1397-*******-713a-ee43cf90d46f", "funcArn": "arn:aws:lambda:ap-northeast-1:*********:function:test1:2", "state": "exit status 1"}
[2021-11-15T16:44:38.477-10:00][INFO]-Created worker.   {"functionArn": "arn:aws:lambda:ap-northeast-1:***********:function:test1:2", "workerId": "177c1397-*******-713a-ee43cf90d46f", "pid": 18810}
[2021-11-15T16:44:39.118-10:00][INFO]-Worker Memory Usage (KB). {"workerId": "177c1397-*******713a-ee43cf90d46f", "funcArn": "arn:aws:lambda:ap-northeast-1:***********:function:test1:2", "memSize": 102400, "memUsed": 14952}
[2021-11-15T16:44:39.118-10:00][ERROR]-Worker is ungracefully killed.   {"workerId": "177c1397*******-713a-ee43cf90d46f", "funcArn": "arn:aws:lambda:ap-northeast-1:********:function:test1:2", "state": "exit status 1"}