aws4embeddedlinux/meta-aws #71: meta-aws for imx-yocto-bsp

BlessyNavara, 2020-08-26:
Toolchain : 
   bitbake - 1.44
   Yocto distro - 5.4.24
added meta-aws layer
building for core-image-minimal target
machine  - imx8m

All dependent layers(openembedded-core, meta-oe, meta-yocto-bsp, meta-filesystems, meta-python) are added to bblayers.conf.
IMAGE_INSTALL_append = " aws-iot-device-sdk-cpp-v2 aws-crt-cpp" added to layers.conf

1. **Is meta-aws having support for imx machines? Guide at github shows that in poky-zeus only qemu is supported. If yes, how to bitbake for imx machines?**

following is the error log file content when running bitbake.(attached only 2/313 errors).

DEBUG: Executing shell function do_compile
NOTE: VERBOSE=1 cmake --build /home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/build --target all -- -j 8
[1/313] /home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++  -DAWS_AUTH_USE_IMPORT_EXPORT -DAWS_CAL_USE_IMPORT_EXPORT -DAWS_COMMON_USE_IMPORT_EXPORT -DAWS_COMPRESSION_USE_IMPORT_EXPORT -DAWS_CRT_CPP_EXPORTS -DAWS_CRT_CPP_USE_IMPORT_EXPORT -DAWS_HTTP_USE_IMPORT_EXPORT -DAWS_IO_USE_IMPORT_EXPORT -DAWS_MQTT_USE_IMPORT_EXPORT -DAWS_MQTT_WITH_WEBSOCKETS -DAWS_USE_EPOLL -DCJSON_HIDE_SYMBOLS -Daws_crt_cpp_EXPORTS -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-http/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-io/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/s2n -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/s2n/api -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot-native/usr/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-common/include -Iaws-common-runtime/aws-c-common/generated/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-compression/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-mqtt/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-cal/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-auth/include -mcpu=cortex-a53+crc+crypto -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0=/usr/src/debug/aws-crt-cpp/0.8.3-r0                      -fdebug-prefix-map=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0=/usr/src/debug/aws-crt-cpp/0.8.3-r0                      -fdebug-prefix-map=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -mcpu=cortex-a53+crc+crypto -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot -O2 -g -DNDEBUG -fPIC   -Wall -Wno-long-long -pedantic -Werror -DS2N_NO_PQ_ASM -DS2N_HAVE_EXECINFO -fPIC -pthread -std=gnu++11 -MD -MT CMakeFiles/aws-crt-cpp.dir/source/StringUtils.cpp.o -MF CMakeFiles/aws-crt-cpp.dir/source/StringUtils.cpp.o.d -o CMakeFiles/aws-crt-cpp.dir/source/StringUtils.cpp.o -c /home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/source/StringUtils.cpp
[2/313] /home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++  -DAWS_AUTH_USE_IMPORT_EXPORT -DAWS_CAL_USE_IMPORT_EXPORT -DAWS_COMMON_USE_IMPORT_EXPORT -DAWS_COMPRESSION_USE_IMPORT_EXPORT -DAWS_CRT_CPP_EXPORTS -DAWS_CRT_CPP_USE_IMPORT_EXPORT -DAWS_HTTP_USE_IMPORT_EXPORT -DAWS_IO_USE_IMPORT_EXPORT -DAWS_MQTT_USE_IMPORT_EXPORT -DAWS_MQTT_WITH_WEBSOCKETS -DAWS_USE_EPOLL -DCJSON_HIDE_SYMBOLS -Daws_crt_cpp_EXPORTS -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-http/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-io/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/s2n -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/s2n/api -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot-native/usr/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-common/include -Iaws-common-runtime/aws-c-common/generated/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-compression/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-mqtt/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-cal/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-auth/include -mcpu=cortex-a53+crc+crypto -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0=/usr/src/debug/aws-crt-cpp/0.8.3-r0                      -fdebug-prefix-map=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0=/usr/src/debug/aws-crt-cpp/0.8.3-r0                      -fdebug-prefix-map=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -mcpu=cortex-a53+crc+crypto -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/recipe-sysroot -O2 -g -DNDEBUG -fPIC   -Wall -Wno-long-long -pedantic -Werror -DS2N_NO_PQ_ASM -DS2N_HAVE_EXECINFO -fPIC -pthread -std=gnu++11 -MD -MT CMakeFiles/aws-crt-cpp.dir/source/Api.cpp.o -MF CMakeFiles/aws-crt-cpp.dir/source/Api.cpp.o.d -o CMakeFiles/aws-crt-cpp.dir/source/Api.cpp.o -c /home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/source/Api.cpp

ninja: build stopped: subcommand failed.
WARNING: exit code 1 from a shell command.

rpcme, 2020-09-01:
Hello, I am on vacation so responses will be delayed, sorry.

I do not have any reason to believe there would be anything wrong with using this bsp.  We've built on many more platforms than Poky.

I will have to reproduce it myself but I won't be able to try until the weekend.

rpcme, 2020-09-14:
Hello @BlessyNavara - coincidentally I am working on another issue (that is not documented in this repo) for imx machines.
 
What I am finding is aws-crt-cpp has a build issue on 'zeus'.  You can see how CMAKE_PREFIX_PATH is not being resolved in the following include:

... -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-common/include ***-Iaws-common-runtime/aws-c-common/generated/include** -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-compression/include -I/home/imx-yocto-bsp/build_dir/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/git/aws-common-runtime/aws-c-mqtt/include ...

The problem becomes immediately clear after jumping to `devshell` on this target.

I am working on a resolution.

BlessyNavara, 2020-09-15:
Hi @rpcme 
I'm glad you are working on this.

But the errors in my log seem to originate from missing "immintrin.h" and "intrin.h" headers.
Specifically, the errors come from the AwsSIMD.cmake file.

Also, Im attaching the Errorlog for your reference.
[CMakeError.log](https://github.com/aws/meta-aws/files/5222511/CMakeError.log)

rpcme, 2020-09-17:
@BlessyNavara thank you for your note. I am back on this issue today.  Going to try this in a clean environment.

rpcme, 2020-09-17:
@BlessyNavara How did you get to that point? I am on clean environment and received the same error I had before.

I committed the recipes to the zeus branch for convenience.  To use this, you will need to manually delete meta-aws/recipes-containers/docker/docker-ce_%.bbappend unless you cloned meta-virtualization and added it to bblayers.conf. However, the build issue needs to be resolved. I will be working with the CRT team internally on it.

BlessyNavara, 2020-09-18:
> @BlessyNavara How did you get to that point? I am on clean environment and received the same error I had before.
> 
> I committed the recipes to the zeus branch for convenience. To use this, you will need to manually delete meta-aws/recipes-containers/docker/docker-ce_%.bbappend unless you cloned meta-virtualization and added it to bblayers.conf. However, the build issue needs to be resolved. I will be working with the CRT team internally on it.
> 
> Also, dependencies are set so you should only have to do:
> 
> IMAGE_INSTALL_append = " aws-iot-device-sdk-cpp-v2"

@rpcme Earlier, I added the dependencies by following this https://layers.openembedded.org/layerindex/branch/master/layer/meta-aws/. Yes, you are right, I added the meta-virtualization to sources and set it in bblayers.conf.

rpcme, 2020-09-22:
Hello @BlessyNavara , this is finally fixed. Sorry for the delay. I have committed the fix to branch 'zeus' and will cherry pick to master later.  Due to the version of the imx sdk you're using, you will want to use zeus.

effectively:

1/ clone meta-aws to ~/imx-yocto-bsp/sources/meta-aws
2/ add the directory to conf/bblayers.conf
3/ add the sdk target to local.conf (or whatever else mechanism you are using)

I do see value in adding this to the generated OE SDK target. This will need to be done later (and I need to identify the right developer experience for that).

Fixed with https://github.com/aws/meta-aws/commit/b5a1707cd23d462e963958fcfda59b56a08e6710

closed 2020-09-22 by rpcme

BlessyNavara, 2020-10-29:
@rpcme Thanks for the fix.