aws4embeddedlinux/meta-aws #75: AGL qemux86_64 dunfell

jwinarske, 2020-09-02:
I'm experiencing a user/group problem with Automotive Grade Linux (AGL) master branch (dunfell).

```
ERROR: Nothing RPROVIDES 'greengrass' (but /builds/workspace_agl/meta-agl-demo/recipes-platform/images/agl-demo-platform.bb RDEPENDS on or otherwise requires it)
greengrass was skipped: greengrass - greengrass: system username ggc_user does not have a static ID defined. Add ggc_user to one of these files: /builds/workspace_agl/meta-agl/meta-agl-profile-core/files/passwd
greengrass was skipped: greengrass - greengrass: system username ggc_user does not have a static ID defined. Add ggc_user to one of these files: /builds/workspace_agl/meta-agl/meta-agl-profile-core/files/passwd
NOTE: Runtime target 'greengrass' is unbuildable, removing...

```

Adding values to file then results in later build break:
```
ERROR: Nothing RPROVIDES 'greengrass' (but /builds/workspace_agl/meta-agl-demo/recipes-platform/images/agl-demo-platform.bb RDEPENDS on or otherwise requires it)
greengrass was skipped: greengrass - greengrass: system groupname ggc_group does not have a static ID defined. Add ggc_group to one of these files: /builds/workspace_agl/meta-agl/meta-agl-profile-core/files/group
greengrass was skipped: greengrass - greengrass: system groupname ggc_group does not have a static ID defined. Add ggc_group to one of these files: /builds/workspace_agl/meta-agl/meta-agl-profile-core/files/group
NOTE: Runtime target 'greengrass' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['greengrass']
ERROR: Required build target 'agl-demo-platform' has no buildable providers.
Missing or unbuildable dependency chain was: ['agl-demo-platform', 'greengrass']
```

Adding group value to file then continues until this:
```

ERROR: greengrass-1.10.1-r0 do_prepare_recipe_sysroot: greengrass: useradd command did not succeed.
ERROR: greengrass-1.10.1-r0 do_prepare_recipe_sysroot: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:useradd_sysroot_sstate(d)
     0003:
File: '/builds/workspace_agl/external/poky/meta/classes/useradd.bbclass', lineno: 152, function: useradd_sysroot_sstate
     0148:        bb.build.exec_func("useradd_sysroot", d)
     0149:    elif task == "prepare_recipe_sysroot":
     0150:        # Used to update this recipe's own sysroot so the user/groups are available to do_install
     0151:        scriptfile = d.expand("${RECIPE_SYSROOT}${bindir}/postinst-useradd-${PN}")
 *** 0152:        bb.build.exec_func("useradd_sysroot", d)
     0153:    elif task == "populate_sysroot":
     0154:        # Used when installed in dependent task sysroots
     0155:        scriptfile = d.expand("${SYSROOT_DESTDIR}${bindir}/postinst-useradd-${PN}")
     0156:
File: '/builds/workspace_agl/external/poky/bitbake/lib/bb/build.py', lineno: 251, function: exec_func
     0247:    with bb.utils.fileslocked(lockfiles):
     0248:        if ispython:
     0249:            exec_func_python(func, d, runfile, cwd=adir)
     0250:        else:
 *** 0251:            exec_func_shell(func, d, runfile, cwd=adir)
     0252:
     0253:    try:
     0254:        curcwd = os.getcwd()
     0255:    except:
File: '/builds/workspace_agl/external/poky/bitbake/lib/bb/build.py', lineno: 452, function: exec_func_shell
     0448:    with open(fifopath, 'r+b', buffering=0) as fifo:
     0449:        try:
     0450:            bb.debug(2, "Executing shell function %s" % func)
     0451:            with open(os.devnull, 'r+') as stdin, logfile:
 *** 0452:                bb.process.run(cmd, shell=False, stdin=stdin, log=logfile, extrafiles=[(fifo,readfifo)])
     0453:        finally:
     0454:            os.unlink(fifopath)
     0455:
     0456:    bb.debug(2, "Shell function %s finished" % func)
File: '/builds/workspace_agl/external/poky/bitbake/lib/bb/process.py', lineno: 182, function: run
     0178:        if not stderr is None:
     0179:            stderr = stderr.decode("utf-8")
     0180:
     0181:    if pipe.returncode != 0:
 *** 0182:        raise ExecutionError(cmd, pipe.returncode, stdout, stderr)
     0183:    return stdout, stderr
Exception: bb.process.ExecutionError: Execution of '/builds/workspace_agl/build/tmp/work/corei7-64-agl-linux/greengrass/1.10.1-r0/temp/run.useradd_sysroot.2068776' failed with exit code 1:
/builds/workspace_agl/build/tmp/work/corei7-64-agl-linux/greengrass/1.10.1-r0/recipe-sysroot-native/usr/sbin/useradd
Running groupadd commands...
Running useradd commands...
useradd: group '1008' does not exist
WARNING: /builds/workspace_agl/build/tmp/work/corei7-64-agl-linux/greengrass/1.10.1-r0/temp/run.useradd_sysroot.2068776:1 exit 1 from 'exit 1'

ERROR: Logfile of failure stored in: /builds/workspace_agl/build/tmp/work/corei7-64-agl-linux/greengrass/1.10.1-r0/temp/log.do_prepare_recipe_sysroot.2068776
ERROR: Task (/builds/workspace_agl/meta-aws/recipes-greengrass/greengrass-core/greengrass_1.10.1.bb:do_prepare_recipe_sysroot) failed with exit code '1'

```

rpcme, 2020-09-14:
Hello @jwinarske I just got off vacation today - just wanted to let you know this is on my radar ok?

rpcme, 2020-09-14:
Hello @jwinarske I am having trouble checking out the jellyfish release which is aligned with dunfell.  This is what I'm getting:

```bash
$ repo init -b jellyfish -m jellyfish_9.99.3.xml -u https://gerrit.automotivelinux.org/gerrit/AGL/AGL-repo
Downloading Repo source from https://gerrit.googlesource.com/git-repo
remote: Counting objects: 1, done
remote: Finding sources: 100% (31/31)
remote: Total 31 (delta 11), reused 31 (delta 11)
Unpacking objects: 100% (31/31), done.
Downloading manifest from https://gerrit.automotivelinux.org/gerrit/AGL/AGL-repo
fatal: Couldn't find remote ref refs/heads/jellyfish
manifests:
fatal: Couldn't find remote ref refs/heads/jellyfish
...
```

Do I need to use some secret sauce to get a hold of this release?  Please let me know, I would like to fix this problem.  I haven't yet had a chance to use jumping jellyfish so this is my first time encountering this problem.

I can also ask the question to the AGL developer group but you might have a more timely answer.

jwinarske, 2020-09-15:
Hi @rpcme,

This was using tip of tree:

    repo init -u https://gerrit.automotivelinux.org/gerrit/AGL/AGL-repo
    repo sync

rpcme, 2020-09-16:
Hello @jwinarske I have reproduced the problem and am working on it today.

rpcme, 2020-09-17:
@jwinarske 

Unfortunately, the recipe won't be able to fix this for now. I repro that adding the user and group works.  This is the way AGL manages it. In my humble opinion, this is horrible because every distro (yes, this means your company's distro) will need to maintain their own passwd and group file and not do it dynamically like in normal Yocto builds.

I will attempt to get ggc_user and ggc_group in those files through a PR but it's doubtful.

For user:

```bash
ubuntu@ip-172-31-15-243:~/agl-jellyfish-9.99.3/build$ tail ../meta-agl/meta-agl-profile-core/files/passwd 
vmail::958:958:::
opensaf::957:957:::
lldpd::956:956:::
cyrus::955:8:::
display::200:200:::
systemd-network::1005:1005:::
systemd-resolve::1006:1006:::
mosquitto::1007:1007:::
ggc_user::10000:10000:::
```
And for group:

```bash
ubuntu@ip-172-31-15-243:~/agl-jellyfish-9.99.3/build$ tail ../meta-agl/meta-agl-profile-core/files/group 
lldpd::956:
postdrop::954:
display::200:
agl-driver::1001:
agl-passenger::1002:
systemd-network::1005:
systemd-resolve::1006:
mosquitto::1007:
ggc_group::10000:
```

rpcme, 2020-09-17:
Note that Greengrass 1.11 is out so please pull that recipe down.  It supports the dunfell release.

I will be noting this issue in the root readme for AGL users.  Closing.

closed 2020-09-17 by rpcme