aws4embeddedlinux/meta-aws #86: Error:mtk-image-1.0-r0 do_rootfs: Could not invoke dnf.

vijayrakesh13, 2020-11-20:
Hi, I'm trying to install AWS IOT SDK CPP V2 recipe, but I'm facing below error.

ERROR: mtk-image-1.0-r0 do_rootfs: Could not invoke dnf. Command '/home/bl-docker/rity/bsp/tmp/work/i500_pumpkin-poky-linux/mtk-image/1.0-r0/recipe-sysroot-native/usr/bin/dnf -v --rpmverbosity=info -y -c /home/bl-docker/rity/bsp/tmp/work/i500_pumpkin-poky-linux/mtk-image/1.0-r0/rootfs/etc/dnf/dnf.conf --setopt=reposdir=/home/bl-docker/rity/bsp/tmp/work/i500_pumpkin-poky-linux/mtk-image/1.0-r0/rootfs/etc/yum.repos.d --installroot=/home/bl-docker/rity/bsp/tmp/work/i500_pumpkin-poky-linux/mtk-image/1.0-r0/rootfs --setopt=logdir=/home/bl-docker/rity/bsp/tmp/work/i500_pumpkin-poky-linux/mtk-image/1.0-r0/temp --repofrompath=oe-repo,/home/bl-docker/rity/bsp/tmp/work/i500_pumpkin-poky-linux/mtk-image/1.0-r0/oe-rootfs-repo --nogpgcheck install aws-iot-device-sdk-cpp-v2 optee-client packagegroup-audio packagegroup-base packagegroup-core-boot packagegroup-core-full-cmdline packagegroup-core-ssh-dropbear packagegroup-zeroconf run-postinsts v4l-utils wayland weston weston-init locale-base-en-us locale-base-en-gb' returned 1:
DNF version: 4.2.2
cachedir: /home/bl-docker/rity/bsp/tmp/work/i500_pumpkin-poky-linux/mtk-image/1.0-r0/rootfs/var/cache/dnf
Added oe-repo repo from /home/bl-docker/rity/bsp/tmp/work/i500_pumpkin-poky-linux/mtk-image/1.0-r0/oe-rootfs-repo
repo: using cache for: oe-repo
not found other for:
not found modules for:
not found deltainfo for:
not found updateinfo for:
oe-repo: using metadata from Fri 20 Nov 2020 07:08:00 AM UTC.
Last metadata expiration check: 0:00:01 ago on Fri 20 Nov 2020 07:08:10 AM UTC.
No module defaults found
No match for argument: aws-iot-device-sdk-cpp-v2
Error: Unable to find a match

Thanks & Regards,
VR

rpcme, 2020-11-20:
@vijayrakesh13 could you kindly let me know the release branch and bsp? thank you.  It seems like this https://www.mediatek.com/products/iot/aiot/pumpkin-software but please let me know details so I can reproduce.

vijayrakesh13, 2020-11-20:
@rpcme  I'm using Dunfell 3.1.1 release branch and mediatek bsp. Thank you.

rpcme, 2020-11-23:
@vijayrakesh13 I am not familiar with this BSP so I am working with Mediatek directly.  Can you please let me know / verify:

> Is it a customer who got BayLibre SDK and Pumpkin i500?

Thank you!

vijayrakesh13, 2020-11-23:
@rpcme Yes, it is i500-Pumpkin with Rity SDK v20.1. 

Thank you.

nirantarashwin, 2020-11-24:
Hi @vijayrakesh13, 

The issue you are facing is a common one and it is being referenced in a [Technical FAQ Section of Yocto](https://wiki.yoctoproject.org/wiki/Technical_FAQ#I_see_a_recipe_built.2C_but_building_an_image_containing_the_corresponding_package_fails_at_do_rootfs_because_it_can.27t_find_the_package._How_does_this_happen.3F)

_Usually this is because the recipe claimed to provide the specified package (via PACKAGES or PACKAGES_DYNAMIC) but it wasn't actually produced, possibly because it ended up empty (since by default empty packages aren't produced), but the image or some other package still has a dependency that pulls in the specified package. If this is a recipe you are writing yourself the probable cause is your recipe isn't installing any files and thus the main package for the recipe is empty._

So I tried to see what was wrong in the recipe and I can see that the recipe does not install any files on the rootfs hence the build fails. See the recipe file [here](https://github.com/aws/meta-aws/blob/master/recipes-sdk/aws-iot-device-sdk-cpp-v2/aws-iot-device-sdk-cpp-v2.inc#L41)

The only library files that are built are static library files '.a'. No '.so' files are built by the recipe. I personally tried to build it by following instructions [here](https://github.com/aws/aws-iot-device-sdk-cpp-v2#automatically-build-and-install-aws-dependencies): 

Attaching a screenshot of the output of library files built: 

[image]

So if you update the recipe to include these files then it should work. 

Let me know if you have any questions. 

Ashwin Nirantar

vijayrakesh13, 2020-11-25:
Hi Ashwin Nirantar,

As you suggested, shall I use SRC_URI to include all these files?

Thanks & Regards,
Vijay

rpcme, 2020-11-25:
@vijayrakesh13 @nirantarashwin this is a surprise to me  - the .so files should be put to rootfs for applications using shared library.  Let me reproduce this morning via Poky x86_64.  I will fix as necessary.

The .a files will be shipped eventually when we build for eSDK, but this work has not been completed yet.

@vijayrakesh13 can you please let me know which branch you're working from as well?

vijayrakesh13, 2020-11-25:
@rpcme I’m working from master branch on Yocto Dunfell 3.1.1.

rpcme, 2020-11-25:
hello @vijayrakesh13 I have verified the problem.  There are two problems actually.

When I worked on the recipe last time, I accidentally removed an EXTRA_OECMAKE to build the shared libraries.  You can see the flag on zeus branch. 

Even after fixing this, I noted that the package in the package feed is empty, which is causing the symptom you are seeing.

I am pushing everything aside to focus on this and hopefully fix today.  I really apologize for this problem.

vijayrakesh13, 2020-11-25:
@rpcme Thank you!

rpcme, 2020-11-25:
@vijayrakesh13 the fix is pushed to master. could you kindly confirm the fix?
Note that the libs with 1.0.0 suffix do not have a corresponding .so symlink - but I think that's ok?

rpcme, 2020-11-25:
Note that I see we are terribly behind in versions so getting you the latest up there ok?

vijayrakesh13, 2020-11-25:
@rpcme  Fix is working. Build is successful, thank you!

closed 2020-11-25 by vijayrakesh13