aws4embeddedlinux/meta-aws #88: Greengrass Core: OTA update

Damien-Mgt, 2020-11-25:
Hi, I'm trying to update the Greengrass Core software with the OTA update agent. The execution of the update task always ends with a failure.  

I'm using a Raspberry Pi 4 with a Yocto image to install Greengrass Core and other user applications. I use the **zeus** branch of the meta-aws layer. So the version of Greengrass Core is v1.10.1 and I'd like to be able to update it remotely with **AWS IoT Jobs**.

As mentioned [in the doc](https://docs.aws.amazon.com/greengrass/latest/developerguide/what-is-gg.html#gg-platforms), I took care to **not use the BusyBox shell commands variants**. I use the _core-image-full-cmdline_ reference image (coreutils shell commands version) and I added the wget package in my Yocto configuration file.

I then configure my OTA update task with _openwrt_ as OS and _armv7l_ as architecture. Here the **update agent logs** (_/var/log/greengrass/ota/ggc_ota.txt_) : [ggc_ota.txt](https://github.com/aws/meta-aws/files/5596175/ggc_ota.txt)

To go further, I used the **strace** command to find out a bit more about what is going on. Here output file as well : [ggc_ota_rpi_strace_output.txt](https://github.com/aws/meta-aws/files/5596183/ggc_ota_rpi_strace_output.txt)

I can't figure out why it doesn't work. Do you have an idea?

Thanks & Regards.

rpcme, 2020-11-25:
Hello Damien,

It is unexpected that the OTA feature would be used for Yocto baked distributions.  Usually, customers are doing system upgrade using swupd or similar.  I am concerned what would happen during swupd if greengrass was modified side-by-side.

Also, I should caution that OTA does not work with trusted boot.

Because of these practices and limitations, I never considered the OTA feature as applicable to Embedded Linux.

I will check with my colleagues internal AWS to see what can be uncovered.

jwinarske, 2020-12-11:
@rpcme Can you explain the limitation with trusted boot?

rpcme, 2020-12-15:
Please see under Requirements, the fifth bullet point.

https://docs.aws.amazon.com/greengrass/latest/developerguide/core-ota-update.html

My understanding is it changes the filesystem signature. Of course, you might be able to get around it by separating the boot partition from other goo, but I haven't seen people do this often.

rpcme, 2020-12-16:
Since the OP hasn't responded for a few weeks, I'm closing this issue.

closed 2020-12-16 by rpcme