aws4embeddedlinux/meta-aws #89: (meta-aws/recipes-crt/aws-crt-cpp/aws-crt-cpp_0.8.3.bb:do_compile) failed with exit code '1'

vijayrakesh13, 2020-11-26:
Hi,

After installing the aws-iot-device-sdk-cpp the dependence package aws-crt-cpp do_compile failed. Here is the log below.

| ninja: build stopped: subcommand failed.
| WARNING: /home/rity/bsp/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/temp/run.do_compile.1622:1 exit 1 from 'eval ${DESTDIR:+DESTDIR=${DESTDIR} }VERBOSE=1 cmake --build '/home/bl-docker/rity/bsp/tmp/work/aarch64-poky-linux/aws-crt-cpp/0.8.3-r0/build' "$@" -- ${EXTRA_OECMAKE_BUILD}'
|
ERROR: Task (/home/rity/src/meta-aws/recipes-crt/aws-crt-cpp/aws-crt-cpp_0.8.3.bb:do_compile) failed with exit code '1'
Summary: 1 task failed:
  /home/bl-docker/rity/src/meta-aws/recipes-crt/aws-crt-cpp/aws-crt-cpp_0.8.3.bb:do_compile

Thanks & Regards,
VR

vijayrakesh13, 2020-11-26:
@rpcme Kindly please look into the issue. Thank you!

rpcme, 2020-11-26:
Hello @vijayrakesh13  - There is no more dependency between the two packages.  Due to complexities that the repo maintainer but on the build environment, the two recipes are separate and the device sdk already includes CRT.  I am trying to make the CRT recipes palatable on their own but since the higher level recipes already include the CRT it is of lesser priority.

I do still need to propagate the fixes to branches so you could be getting an old error on dunfell branch?

vijayrakesh13, 2020-12-01:
@rpcme Sorry for the late reply. Yes, there is no dependency between two packages, but I was saying CRT is dependence package to run basic subpub example. 

Yes, I'm getting an error on dunfell branch, thank you.

rpcme, 2020-12-02:
Sorry for the late reply, I will try to get the changes propagated to dunfell (and others) tomorrow morning.  I am unfortunately running out of runway today.

vijayrakesh13, 2020-12-02:
@rpcme Thank you! Is there any recipe for aws-iot-device-sdk-embedded-C also? If yes, Please share me the link.

vijayrakesh13, 2020-12-04:
@rpcme I'm waiting for your response. Kindly please update the changes at the earliest.

rpcme, 2020-12-04:
@vijayrakesh13 it is planned but it hasn't happened yet. the new v4 is now ready and we should be able to pick it up.  The last time I checked it doesn't produce a shared library, so this would be provided as a demo recipe and eSDK provider.  I would have to cross check this but this is my understanding right now.

closed 2020-12-08 by vijayrakesh13

yourslab, 2020-12-28:
Hi @vijayrakesh13 @rpcme

`aws-iot-device-sdk-embedded-C` now supports installation.

Documentation can be found in the [README](https://github.com/aws/aws-iot-device-sdk-embedded-C#installation).