aws4embeddedlinux/meta-aws #93: neo-ai-dlr build failure

dhwalters423, 2020-12-23:
Build configuration:
```
Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "imx8mqevk"
DISTRO               = "fsl-imx-xwayland"
DISTRO_VERSION       = "5.4-zeus"
TUNE_FEATURES        = "aarch64 cortexa53 crc crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:5408e48d374fc741fa46b57ddfd8a537843bda6f"
meta-oe              
meta-multimedia      
meta-python          = "HEAD:2b5dd1eb81cd08bc065bc76125f2856e9383e98b"
meta-freescale       = "HEAD:14f1a630a47375432f93c556927b879b51d84c4e"
meta-freescale-3rdparty = "HEAD:dbcc686f52c3c84db8cb86aa8973a4e373651b98"
meta-freescale-distro = "HEAD:ca27d12e4964d1336e662bcc60184bbff526c857"
meta-aws             = "zeus:3e79237b1495b4d957dca5d7467131de787e02bc"
meta-virtualization  = "zeus:48ec1b49871fef7d331ac1d0680a48d362c7482b"
meta-bsp             
meta-sdk             
meta-ml              = "HEAD:780b023a4c77ac9c0011d858ad377f4a9593bfea"
meta-nxp-demo-experience = "HEAD:c7263d9f3cc7bbf44e7164ffeda494cf283d3dec"
meta-browser         = "HEAD:ee3be3b5986a4aa0e73df2204a625ae1fe5df37e"
meta-rust            = "HEAD:7f235b6f8973cc5269448375f2a8f9867bb2a369"
meta-clang           = "HEAD:711e593d5984aad3bf35c51b7ac4482982bc16c7"
meta-gnome           
meta-networking      
meta-filesystems     = "HEAD:2b5dd1eb81cd08bc065bc76125f2856e9383e98b"
meta-qt5             = "HEAD:5144a99a696a8a72c0765ea0e8be8ceafaa78f85"
meta-python2         = "HEAD:4400f9155ec193d028208cf0c66aeed2ba2b00ab"
```

local.conf includes the following:
``IMAGE_INSTALL_append = " neo-ai-dlr"``

Error:
```
WARNING: neo-ai-dlr-1.1-r0 do_fetch: Failed to fetch URL git://github.com/neo-ai/tvm;protocol=https;branch=stable;destsuffix=/home/ubuntu/environment/imx-yocto-bsp/build-yocto/tmp/work/aarch64-poky-linux/neo-ai-dlr/1.1-r0/git/3rdparty/tvm;name=neo-ai-tvm, attempting MIRRORS if available
ERROR: neo-ai-dlr-1.1-r0 do_fetch: Fetcher failure: Unable to find revision 44779571412930681eef9e8b9d32aa845b8cc5ad in branch stable even from upstream
ERROR: neo-ai-dlr-1.1-r0 do_fetch: Fetcher failure for URL: 'git://github.com/neo-ai/tvm;protocol=https;branch=stable;destsuffix=/home/ubuntu/environment/imx-yocto-bsp/build-yocto/tmp/work/aarch64-poky-linux/neo-ai-dlr/1.1-r0/git/3rdparty/tvm;name=neo-ai-tvm'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/ubuntu/environment/imx-yocto-bsp/build-yocto/tmp/work/aarch64-poky-linux/neo-ai-dlr/1.1-r0/temp/log.do_fetch.1138
ERROR: Task (/home/ubuntu/environment/imx-yocto-bsp/sources/meta-aws/recipes-neo-ai/neo-ai-dlr/neo-ai-dlr.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 9255 tasks of which 9254 didn't need to be rerun and 1 failed.
```

Suggest changing https://github.com/aws/meta-aws/blob/b29e614e76fd61548bb3883a9b3294f789c5c678/recipes-neo-ai/neo-ai-dlr/neo-ai-dlr.bb#L22 to:
``git://github.com/neo-ai/tvm;protocol=https;branch=release-1.7.0;destsuffix=${S}/3rdparty/tvm;name=neo-ai-tvm \``

stable branch does not exist in the neo-ai/tvm repo.

rpcme, 2020-12-23:
Hello @dhwalters423 this recipe is terribly outdated and I am currently upgrading it to the 1.7.0 tag.  No ETA at this point, there's a few bumps, but working on it right now.

rpcme, 2020-12-23:
Hello @dhwalters423 the changes have been applied to master. Kindly pull and switch meta-aws branch to master.

dhwalters423, 2021-01-04:
I pulled and checked out master

Added in local.conf:
``IMAGE_INSTALL_append = " neo-ai-dlr"``

Received the following error:
```
ERROR: neo-ai-dlr-1.7.0-r0 do_install: 'python3 setup.py install --root=/home/ubuntu/environment/imx-yocto-bsp/build-yocto/tmp/work/aarch64-poky-linux/neo-ai-dlr/1.7.0-r0/image     --prefix=/usr     --install-lib=/usr/lib/python3.7/site-packages     --install-data=/usr/share' execution failed.
ERROR: neo-ai-dlr-1.7.0-r0 do_install: Execution of '/home/ubuntu/environment/imx-yocto-bsp/build-yocto/tmp/work/aarch64-poky-linux/neo-ai-dlr/1.7.0-r0/temp/run.do_install.29252' failed with exit code 1:
/home/ubuntu/environment/imx-yocto-bsp/build-yocto/tmp/work/aarch64-poky-linux/neo-ai-dlr/1.7.0-r0/recipe-sysroot-native/usr/bin/python3-native/python3: can't open file 'setup.py': [Errno 2] No such file or directory
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/ubuntu/environment/imx-yocto-bsp/build-yocto/tmp/work/aarch64-poky-linux/neo-ai-dlr/1.7.0-r0/temp/log.do_install.29252
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/home/ubuntu/environment/imx-yocto-bsp/sources/poky/meta/recipes-core/glibc/glibc_2.30.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/environment/imx-yocto-bsp/sources/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/environment/imx-yocto-bsp/sources/poky/meta/recipes-devtools/ninja/ninja_1.9.0.bb:do_populate_sysroot', '/home/ubuntu/environment/imx-yocto-bsp/sources/poky/meta/recipes-devtools/gcc/gcc-runtime_9.2.bb:do_populate_sysroot', '/home/ubuntu/environment/imx-yocto-bsp/sources/poky/meta/recipes-devtools/python/python3-setuptools_41.2.0.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/environment/imx-yocto-bsp/sources/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/environment/imx-yocto-bsp/sources/poky/meta/recipes-devtools/python/python3_3.7.8.bb:do_populate_sysroot', '/home/ubuntu/environment/imx-yocto-bsp/sources/poky/meta/recipes-devtools/gcc/gcc-cross_9.2.bb:do_populate_sysroot', '/home/ubuntu/environment/imx-yocto-bsp/sources/meta-imx/meta-bsp/recipes-test/googletest/googletest_git.bb:do_populate_sysroot', '/home/ubuntu/environment/imx-yocto-bsp/sources/poky/meta/recipes-devtools/python/python3_3.7.8.bb:do_populate_sysroot', '/home/ubuntu/environment/imx-yocto-bsp/sources/poky/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot', '/home/ubuntu/environment/imx-yocto-bsp/sources/poky/meta/recipes-devtools/cmake/cmake-native_3.15.3.bb:do_populate_sysroot', 'virtual:native:/home/ubuntu/environment/imx-yocto-bsp/sources/poky/meta/recipes-devtools/python/python3-setuptools_41.2.0.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['glibc', 'patch-native', 'ninja-native', 'gcc-runtime', 'python3-setuptools', 'pseudo-native', 'python3-native', 'gcc-cross-aarch64', 'googletest', 'python3', 'quilt-native', 'cmake-native', 'python3-setuptools-native', 'linux-libc-headers', 'gnu-config-native', 'autoconf-native', 'automake-native', 'libtool-native', 'attr-native', 'libgcc', 'sqlite3-native', 'libtirpc-native', 'readline-native', 'libnsl2-native', 'zlib-native', 'xz-native', 'libffi-native', 'gdbm-native', 'pkgconfig-native', 'bzip2-native', 'openssl-native', 'util-linux-native', 'zlib', 'libxcrypt', 'gdbm', 'opkg-utils', 'xz', 'libnsl2', 'readline', 'openssl', 'sqlite3', 'bzip2', 'libtirpc', 'util-linux', 'libffi', 'expat-native', 'ncurses-native', 'curl-native', 'unzip-native', 'binutils-cross-aarch64', 'gmp-native', 'texinfo-dummy-native', 'libmpc-native', 'flex-native', 'mpfr-native', 're2c-native', 'm4-native', 'gettext-minimal-native', 'libcap-ng-native', 'ncurses', 'cryptodev-linux', 'libcap-ng', 'bash-completion', 'libpam', 'cracklib', 'flex']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_install
| /home/ubuntu/environment/imx-yocto-bsp/build-yocto/tmp/work/aarch64-poky-linux/neo-ai-dlr/1.7.0-r0/recipe-sysroot-native/usr/bin/python3-native/python3: can't open file 'setup.py': [Errno 2] No such file or directory
| ERROR: 'python3 setup.py install --root=/home/ubuntu/environment/imx-yocto-bsp/build-yocto/tmp/work/aarch64-poky-linux/neo-ai-dlr/1.7.0-r0/image     --prefix=/usr     --install-lib=/usr/lib/python3.7/site-packages     --install-data=/usr/share' execution failed.
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/ubuntu/environment/imx-yocto-bsp/build-yocto/tmp/work/aarch64-poky-linux/neo-ai-dlr/1.7.0-r0/temp/run.do_install.29252' failed with exit code 1:
| /home/ubuntu/environment/imx-yocto-bsp/build-yocto/tmp/work/aarch64-poky-linux/neo-ai-dlr/1.7.0-r0/recipe-sysroot-native/usr/bin/python3-native/python3: can't open file 'setup.py': [Errno 2] No such file or directory
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/home/ubuntu/environment/imx-yocto-bsp/sources/meta-aws/recipes-devtools/amazon-sagemaker-neo/neo-ai-dlr_1.7.0.bb:do_install) failed with exit code '1'
NOTE: Tasks Summary: Attempted 9249 tasks of which 9240 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/ubuntu/environment/imx-yocto-bsp/sources/meta-aws/recipes-devtools/amazon-sagemaker-neo/neo-ai-dlr_1.7.0.bb:do_install
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

rpcme, 2021-01-06:
setuptools changed between zeus/dunfell and what's on master now. 

Please pull and switch to Zeus branch.

rpcme, 2021-01-06:
This is verified fixed (messaged directly to me)

closed 2021-01-06 by rpcme