aws4embeddedlinux/meta-aws #95: Missing python3-distro dependency for neo-ai-dlr

dhwalters423, 2021-01-11:
```
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "imx8mqevk"
DISTRO               = "fsl-imx-wayland"
DISTRO_VERSION       = "5.4-zeus"
TUNE_FEATURES        = "aarch64 cortexa53 crc crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:5408e48d374fc741fa46b57ddfd8a537843bda6f"
meta-oe              
meta-multimedia      
meta-python          = "HEAD:2b5dd1eb81cd08bc065bc76125f2856e9383e98b"
meta-freescale       = "HEAD:14f1a630a47375432f93c556927b879b51d84c4e"
meta-freescale-3rdparty = "HEAD:dbcc686f52c3c84db8cb86aa8973a4e373651b98"
meta-freescale-distro = "HEAD:ca27d12e4964d1336e662bcc60184bbff526c857"
meta-aws             = "HEAD:4d0bdcdf0994804f7159e12a36298a11291e9a62"
meta-scipy           = "zeus:757b20376ed7cb8e4497c6b9bdc5444a5b5a4e81"
meta-virtualization  = "zeus:48ec1b49871fef7d331ac1d0680a48d362c7482b"
meta-bsp             
meta-sdk             
meta-ml              = "HEAD:780b023a4c77ac9c0011d858ad377f4a9593bfea"
meta-nxp-demo-experience = "HEAD:c7263d9f3cc7bbf44e7164ffeda494cf283d3dec"
meta-browser         = "HEAD:ee3be3b5986a4aa0e73df2204a625ae1fe5df37e"
meta-rust            = "HEAD:7f235b6f8973cc5269448375f2a8f9867bb2a369"
meta-clang           = "HEAD:711e593d5984aad3bf35c51b7ac4482982bc16c7"
meta-gnome           
meta-networking      
meta-filesystems     = "HEAD:2b5dd1eb81cd08bc065bc76125f2856e9383e98b"
meta-qt5             = "HEAD:5144a99a696a8a72c0765ea0e8be8ceafaa78f85"
meta-python2         = "HEAD:4400f9155ec193d028208cf0c66aeed2ba2b00ab"
```

**Missing ‘distro’ package**

Run the following from device after build:
``import dlr``  throws the following error: 
```
ERROR:root:linux 64 api exception occurred
ERROR:root:unable to create system class instance
```

To fix, ``python3 -m pip install distro``'

*Adding recipe for ‘distro’*
It’s not included in meta-openembedded/meta-python in zeus branch, but was added with this PR to meta-openembedded and merged on master: https://github.com/openembedded/meta-openembedded/pull/213/files

You can manually add this recipe to your yocto layer:

Add ``python3-distro_1.5.0.bb`` to <yocto root>/sources/meta-aws/recipes-devtools/amazon-sagemaker-neo 

```
SUMMARY = "Distro is an OS platform information API"
SECTION = "devel/python"
LICENSE = "Apache-2.0"
LIC_FILES_CHKSUM = "file://LICENSE;md5=d2794c0df5b907fdace235a619d80314"

PYPI_PACKAGE = "distro"

SRC_URI[md5sum] = "0ed68b4064709bdaaf6cce69780ddc51"
SRC_URI[sha256sum] = "0e58756ae38fbd8fc3020d54badb8eae17c5b9dcbed388b17bb55b8a5928df92"

inherit pypi setuptools3
```

Add it to your local.conf:
IMAGE_INSTALL_append = " python3-distro"

rpcme, 2021-01-11:
Hello @dhwalters423, the plan is to put recipes that need to be upstream and upstream will likely not ship them to release branches (like zeus), is to put them into a `recipes-external` directory off the root.  If you would like to submit this as a PR for zeus and dunfell (if it's not on dunfell or will not be pulled down to dunfell) then we can maintain it here until further notice.

rpcme, 2021-01-20:
This issue has been resolved by #99 and #100

PR #100 (2021-01-11): Add python3-distro dependency to Zeus
**Issue #, if available:**
Fixes #95

**Description of changes:**
Added python3-distro package to zeus as an external recipe

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
closed 2021-01-20 by rpcme