aws4embeddedlinux/meta-aws #96: Missing test package dependencies for neo-ai-dlr

dhwalters423, 2021-01-11:
The packages are not required for DLR to build and run successfully on the device, but in order to complete the [DLR integration tests](https://github.com/neo-ai/neo-ai-dlr/tree/release-1.7.0/tests/python/integration) the following packages are required:

 * pytest
 * sklearn (scikit-learn)
 * scipy (dependency of sklearn)

Scipy is an RDEPEND of sklearn, but there seems to be issues xcompiling: 
https://github.com/scipy/scipy/issues/8226

Build info:

```
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "imx8mqevk"
DISTRO               = "fsl-imx-wayland"
DISTRO_VERSION       = "5.4-zeus"
TUNE_FEATURES        = "aarch64 cortexa53 crc crypto"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:5408e48d374fc741fa46b57ddfd8a537843bda6f"
meta-oe              
meta-multimedia      
meta-python          = "HEAD:2b5dd1eb81cd08bc065bc76125f2856e9383e98b"
meta-freescale       = "HEAD:14f1a630a47375432f93c556927b879b51d84c4e"
meta-freescale-3rdparty = "HEAD:dbcc686f52c3c84db8cb86aa8973a4e373651b98"
meta-freescale-distro = "HEAD:ca27d12e4964d1336e662bcc60184bbff526c857"
meta-aws             = "HEAD:4d0bdcdf0994804f7159e12a36298a11291e9a62"
meta-scipy           = "zeus:757b20376ed7cb8e4497c6b9bdc5444a5b5a4e81"
meta-virtualization  = "zeus:48ec1b49871fef7d331ac1d0680a48d362c7482b"
meta-bsp             
meta-sdk             
meta-ml              = "HEAD:780b023a4c77ac9c0011d858ad377f4a9593bfea"
meta-nxp-demo-experience = "HEAD:c7263d9f3cc7bbf44e7164ffeda494cf283d3dec"
meta-browser         = "HEAD:ee3be3b5986a4aa0e73df2204a625ae1fe5df37e"
meta-rust            = "HEAD:7f235b6f8973cc5269448375f2a8f9867bb2a369"
meta-clang           = "HEAD:711e593d5984aad3bf35c51b7ac4482982bc16c7"
meta-gnome           
meta-networking      
meta-filesystems     = "HEAD:2b5dd1eb81cd08bc065bc76125f2856e9383e98b"
meta-qt5             = "HEAD:5144a99a696a8a72c0765ea0e8be8ceafaa78f85"
meta-python2         = "HEAD:4400f9155ec193d028208cf0c66aeed2ba2b00ab"
```

rpcme, 2021-01-11:
Hello @dhwalters423 we can handle this like #95 - put the recipes into `recipes-external` until they can be shipped upstream.  There is another effort that needs these recipes as well.

jackmitch, 2021-01-14:
Without wanting to intrude, there is a common effort going on at the moment to try get scikit-learn building under OE. Please see the repo https://github.com/tuxable-ltd/meta-scikit-learn, and take note of issue https://github.com/tuxable-ltd/meta-scikit-learn/issues/1, while it looks like you're targeting a 64bit system you will probably get away with it, but there may be other underlying issues with improper cross compilation.

rpcme, 2021-04-01:
@jackmitch can you let us know what's going on with that? I am going to add the recipes under recipes-external so we can support customers, but when it gets upstream and it supports all out targets, I'm happy to pull it out.  Especially if it's going into something like meta-python.

jackmitch, 2021-04-06:
Hi @rpcme, the layer is currently in a working state and is being used on both arm and arm64 targets in a production setting, however the project only uses a small subset of functionality so it's not exhaustive testing. I will tidy up the layer and push a gatesgarth branch which coincides with what my client is using. As for pushing upstream I currently have no plans due to the requirement for Fortran support to be enabled in order to test, and there is also some changes for OpenBLAS support. What would need to happen is for blas support to become a virtual provider upstream and then the distro be able to select the blas/cblas provider. I would like to work on this but I need to find the time.

I hope this answers your questions, and feel free to ping me with any others.

rpcme, 2021-04-08:
Hello @jackmitch thank you for your input.  Indeed, there are very specific requirements for these libraries that will be difficult to be shipped upstream.

we need to support more than gatesgarth.  However, it is not cool to create fragmentation here.  Can our engineer get the required dependencies into your layer, and then have the layer indexed by openembedded if it's not already?  I will then put the layer in as an optional dependency on my side.  I am only hoping that the layer will be actively maintained?

jackmitch, 2021-04-08:
@rpcme, to go further back from Gatesgarth I believe will require including a newer version of numpy as they implemented a packaging feature that I use in order to successfully cross compile. I would be happy to take patches for older branches in order to implement this.

With regards to maintenance the layer was created on behalf of a client and is being shipped in production for arm and arm64 based systems. I have intentions to keep the layer updated as I tend to track close to OE master HEAD so fixes will most likely come when we bump OE versions for the product.

rpcme, 2021-04-08:
@jackmitch I have this PR here: https://github.com/aws/meta-aws/pull/119

I asked Srikanth to take out the dependency in layer.conf because 1/ I didn't want the hosttools dependency there for everyone and 2/ I didn't have the time to think about tuning it.  So, I asked him to document it.  I think it might have been forked from your work or someone else's.  If we can get this into your repo I think it would be fantastic.  What do you think?

jackmitch, 2021-04-08:
Yes, throw a pull request over, I'll also try whip the actual Gatesgarth layer into shape. I was also undecided on the hosttools and you've made it easy by proposing to be a user of the layer.

rpcme, 2021-04-08:
awesome @jackmitch I spoke with @srikanth311 and he will do the PR when he gets a moment.

srikanth311, 2021-04-10:
Thank you @rpcme. I just tested it again by forking https://github.com/tuxable-ltd/meta-scikit-learn and adding my changes. I created a new branch with `dunfell` in my repo - https://github.com/srikanth311/meta-scikit-learn/tree/dunfell. @jackmitch - Can you create `dunfell` branch in your repo, so that I can push it to that branch?

jackmitch, 2021-04-12:
@srikanth311 I've created that branch for you.

rpcme, 2021-04-24:
@srikanth311 @jackmitch I see that the PR was approved - awesome teamwork!! I'm going to close this issue now.

closed 2021-04-24 by rpcme