aws4embeddedlinux/meta-aws #97: Unable to locate package aws-iot-device-sdk-cpp-v2

Zeph1990, 2021-01-11:
Hi,
we have a problem on aws-iot-device-sdk-cpp-v2 recipe from the master branch when building our device image based on yocto dunfell. local.conf's IMAGE_INSTALL_APPEND  includes aws-iot-device-sdk-cpp-v2.

ERROR: st-image-eglfs-qt-1.0-r0 do_rootfs: Unable to install packages. Command '/home/alex/st_dunfell_ota/build-openstlinuxeglfs-stm32mp1/tmp-glibc/work/stm32mp1_disco-ostl-linux-gnueabi/st-image-eglfs-qt/1.0-r0/recipe-sysroot-native/usr/bin/apt-get  install --force-yes --allow-unauthenticated --no-remove apt aws-iot-device-sdk-cpp-v2 cmake curl curl-dev dpkg enablewifi git mender-client nano libnghttp2-14 packagegroup-base-extended packagegroup-core-boot packagegroup-core-eclipse-debug packagegroup-core-sdk packagegroup-core-ssh-dropbear packagegroup-core-standalone-sdk-target packagegroup-core-tools-profile packagegroup-framework-core packagegroup-framework-core-base packagegroup-framework-core-extra packagegroup-framework-sample-qt packagegroup-framework-sample-qt-extra packagegroup-framework-tools packagegroup-framework-tools-base portaudio-v19 resize-helper run-postinsts tzdata vim' returned 100:
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package aws-iot-device-sdk-cpp-v2

if we launch bitbake aws-iot-device-sdk-cpp-v2 all seems to work properly. The version is : 1.10.0

Any suggestion?

Thank you

rpcme, 2021-01-11:
Yes, I haven't tested it on dunfell specifically yet. Once done, hopefully today, I will commit to dunfell.  Sorry for that @Zeph1990.

rpcme, 2021-01-11:
Hello @Zeph1990 I have pulled down the recipes to dunfell.

Note that in so doing I found an incorrectly configured flag and shared libraries were not being generated.  They are now generated and put to /usr/lib properly.

Note that I have not tested SDK generation yet so I am unsure if headers are being propagated. Please let me know your experience and will help to overcome any shortcoming you might experience.

Zeph1990, 2021-01-12:
Hi @rpcme,

Thank you so much, i will test it as soon as possible and let you know.

Best Regards

rpcme, 2021-01-20:
Closing - please reopen if error is reproduced.

closed 2021-01-20 by rpcme