aws4embeddedlinux/meta-aws #102: Official buildroot support?

aduskett, 2021-01-22:
I'm not sure where to put this, so I figured I would put this here.
I was able to port all of the packages used in the Zeus branch over to buildroot and make a repository:
https://github.com/aduskett/buildroot-aws-iot

Suppose you want to take over and make an official repository that would be awesome. If not, no big deal either. I figured you guys would like to know!

Thanks!

aduskett, 2021-01-23:
The main branch also has the latest versions of the various libraries including many of the sdk's that are in the issues list.

rpcme, 2021-01-23:
Hello @aduskett This is superb! Let me check internally how we would go about doing this.  Hope to get an answer for you by the end of this coming week!

aduskett, 2021-01-23:
> Hello @aduskett This is superb! Let me check internally how we would go about doing this. Hope to get an answer for you by the end of this coming week!

Thanks! I am going to also include the video stream packages as well.

aduskett, 2021-01-25:
@rpcme I was able to update all of the packages to their latest releases in the main branch, and I added the amazon-kinesis-video-streams packages as well.

rpcme, 2021-01-29:
Hello @aduskett for clarity: if we were to "take over" the maintenance, and since this would be MIT License, your expectation would only be to carry over the copyright correct? We are not able to grant commit bits for individuals outside our Github orgs.  Please confirm so I can effectively "work the process".

aduskett, 2021-01-29:
Hey @rpcme That's usually the standard license I attach to any repository. If you want I would gladly change it over to apache-2.0 like the rest of your repos.

rpcme, 2021-01-30:
Hello @aduskett the license is not the issue.  I just wanted to make sure you were OK with the commit bit situation.  Just trying to ensure there isn't any misunderstanding :)  I would have to fork the repo internally, do testing, and then go through our normal internal open source process to instantiate the repo in one of our organizations.  So, it's a process, but I feel it's totally worth it.

aduskett, 2021-01-30:
Thanks, @rpcme ! I really appreciate you supporting Buildroot! It's not often I see companies supporting external buildroot tree's, let alone a company so large as Amazon! It really means a lot!

rpcme, 2021-01-30:
Hi @aduskett can you contact me on https://www.linkedin.com/in/richardelberger/ so I can meet you formally and setup a virtual  meet and greet for next week?  I want to ensure we have the same expectations.

aduskett, 2021-01-30:
Can do!

rpcme, 2021-02-06:
@aduskett as discussed we will continue to work on this offline and I will close this issue.  Thank you so much for this great work!!

closed 2021-02-06 by rpcme