aws4embeddedlinux/meta-aws #103: Greengrass pkg_postinst_ontarget_${PN} prevents read-only-rootfs

kennu, 2021-01-25:
The Greengrass recipe is using pkg_postinst_ontarget_${PN} to perform some configuration on first-boot: https://github.com/aws/meta-aws/blob/master/recipes-greengrass/greengrass-core/greengrass.inc#L40

This prevents building a read-only root filesystem using IMAGE_FEATURES += "read-only-rootfs".

It causes this error at build-time:

    The following packages could not be configured offline and rootfs is read-only: ['100-greengrass']

I would propose replacing the pkg_postinst_ontarget_${PN} configuration with some alternative method of doing the customizations that are currently done on first boot.

rpcme, 2021-01-29:
@kennu Thank you for your feedback.  I will evaluate the proper fix and get back to you hopefully within 3 business days.

kennu, 2021-01-29:
Thanks @rpcme. Btw, I wonder why the Greengrass recipe is removing /var/volatile from fstab? This is also needed for a read-only filesystem, so that /var/volatile is a writable tmpfs. It seems a little aggressive to force this on the OS build.

In my current workaround, I implemented these alternatives to eliminate pkg_postinst_ontarget:
* Don't modify /etc/sysctl.conf but instead add a /etc/sysctl.d/greengrass.conf file
* Let /var/volatile be tmpfs
* Add base-files_%.bbappend with do_install_append() to add the cgroup mount to /etc/fstab

This way I can build a read-only rootfs but I'm only beginning to test it now.

rpcme, 2021-03-24:
Hello @kennu somehow I missed your last message and finally getting back on these after Embedded World chaos.

I thought this change was only for the Qualcomm BSP which had some ... interesting challenges.  I need to re-evaluate and figure out how it got snuck into all targets.

njgudman-reach, 2021-07-01:
Hi @rpcme! I'm also using a read-only rootfs, but with dunfell. Could this fix be merged into the dunfell branch? I tried out the master branch with my dunfell build, and Greengrass installed and ran fine.

rpcme, 2021-07-02:
Hello @njgudman-reach I am currently on leave so my workstation is turned off.  Let me see if I can get someone to help you in the meantime ok?