aws4embeddedlinux/meta-aws #104: aws-crt-python: build fail in zeus branch

gianlucacornacchia, 2021-01-26:
In zeus branch build fails

branch: zeus
command: bitbake aws-iot-device-sdk-python-v2
error:
WARNING: aws-crt-python-0.5.11-r0 do_fetch: Failed to fetch URL git://github.com/awslabs/aws-crt-python.git;branch=master;name=aws-crt-python, attempting MIRRORS if available
ERROR: aws-crt-python-0.5.11-r0 do_fetch: Fetcher failure: Unable to find revision 2d19abb7fc360416202f9c590971c91c84dc2c72 in branch master even from upstream
ERROR: aws-crt-python-0.5.11-r0 do_fetch: Fetcher failure for URL: 'git://github.com/awslabs/aws-crt-python.git;branch=master;name=aws-crt-python'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /yocto/yocto-output/imx6ul-var-dart/tmp/work/cortexa7t2hf-neon-poky-linux-gnueabi/aws-crt-python/0.5.11-r0/temp/log.do_fetch.18887
ERROR: Task (/yocto/projects/neutronic/sources/meta-aws/recipes-crt/aws-crt-python/aws-crt-python_0.5.11.bb:do_fetch) failed with exit code '1'

The error is due to the missing branch 'master' in the repository git://github.com/awslabs/aws-crt-python.git
In this repo the master branch has been renamed to 'main'

I got the build finish without errors changing the branch name from 'master' to 'main' in the file _recipes-crt/aws-crt-python/aws-crt-python.h_:
```
@@ -2,11 +2,11 @@ SUMMARY = "AWS CRT Python"
 DESCRIPTION = "Python bindings for the AWS Common Runtime"
 HOMEPAGE = "https://github.com/awslabs/aws-crt-python"
 LICENSE = "Apache-2.0"
 LIC_FILES_CHKSUM = "file://LICENSE;md5=3b83ef96387f14655fc854ddc3c6bd57"
 
-BRANCH ?= "master"
+BRANCH ?= "main"
 
 SRC_URI = "git://github.com/awslabs/aws-crt-python.git;branch=${BRANCH};name=aws-crt-python"
 SRCREV = "2d19abb7fc360416202f9c590971c91c84dc2c72"
 S = "${WORKDIR}/git"
```

rpcme, 2021-01-29:
Thank you for submitting this issue.  I will evaluate the proper fix and get back to you hopefully within 3 business days.

rpcme, 2021-02-13:
Hello @gianlucacornacchia I have pulled down my fixes from master.  Can you please confirm? I will automatically close this issue in 7 days.

rpcme, 2021-02-23:
Closing as solved.  Please let me know if otherwise.

closed 2021-02-23 by rpcme