aws4embeddedlinux/meta-aws #107: Greengrass startup script /etc/greengrass breaks when /greengrass is a symlink

kennu, 2021-02-01:
The startup script at https://github.com/aws/meta-aws/blob/master/recipes-greengrass/greengrass-core/greengrass/greengrass-init#L3 uses the command `find /greengrass/ -type f -name greengrassd` to find the Greengrass executable. This breaks when /greengrass is a symlink (in our case, it's a symlink to /data/greengrass, because our root partition is read-only and Greengrass has to be on a writable partition).

As a fix, I would suggest one of these:
1. Change the command to `find /greengrass/ -type f -name greengrassd` so that the symlink is resolved
2. Change the command to `find -L /greengrass -type f -name greengrassd` so that find follows symlinks

rpcme, 2021-02-06:
This is definitely a new case.  Thank you for the report.  I need will need to thoroughly test this one to ensure I don't get any regression.

Are you only reporting this and already created a bbappend for the recipe or is this blocking you?

kennu, 2021-02-06:
We are actually patching files at build time and replacing greengrass.inc and greengrass_1.11.0.bb with our own versions.  So it's not blocking our project, but of course it would be nice to use the official versions in the long run.

commit d66de6ed (2021-06-10): add the -L flag to overcome symlink challenges.  Fixes #107
closed 2021-06-10 by rpcme