aws4embeddedlinux/meta-aws #108: Building latest C++ SDK with old Yocto (Pyro)

angrdtsk, 2021-02-02:
I am trying to build meta-aws from master branch with Yocto Pyro. I know Pyro is old so I am ready to accept that I may not get it to work, but I have a reason to try. What I need is the C++ SDK v2, preferably recent enough to come with the changes that brought in the support for building your own application outside the SDK source tree a couple of months ago.

I have added the layer into my bblayers.conf and IMAGE_INSTALL_append = " aws-iot-device-sdk-cpp-v2" to my local.conf. That makes Yocto attempt to build the SDK, but I am getting errors.

When Yocto run cmake, there was an issue related pthread:
| -- Looking for pthread.h
| -- Looking for pthread.h - found
| -- Looking for pthread_create
| -- Looking for pthread_create - not found
| -- Check if compiler accepts -pthread
| CMake Error: TRY_RUN() invoked in cross-compiling mode, please set the following cache variables appropriately:
|    THREADS_PTHREAD_ARG (advanced)

Based on my research on Google, I added set( THREADS_PTHREAD_ARG "2" CACHE STRING "Result from TRY_RUN" FORCE) to the SDK's toolchain.cmake. That was enough to get Yocto to try to actually build the SDK.

Now I am getting this error:
aws-iot-device-sdk-cpp-v2/1.10.0-r0/git/crt/aws-crt-cpp/crt/aws-checksums/source/arm/crc32c_arm.c:1:0: error: switch -mcpu=cortex-a8 conflicts with -march=armv8-a+crc switch [-Werror]

Does this information point to any problem that is easy to solve or at least known?

angrdtsk, 2021-02-03:
I remember that when buidling the C++ SDK v2 without Yocto, there was need to have libssl-dev installed. I also remember that that might have done some relation to checksums which this error here seems to be about too, at least based on the name of the file. I noticed that the meta-aws layer has no reference to libssl. Could these details have any role in this issue?

rpcme, 2021-02-06:
@angrdtsk  I am not entirely certain why this would be happening but I would need to test it.  Is it that you just want to tack on cloudy capabilities to an existing app?

Pyro is so old I am wondering if I will have do dig up an Ubuntu 16.04 image to build it.  Of course this is kind of a big ask to create another maintenance branch but I will give it a go if this is really blocking.  Please let me know.

angrdtsk, 2021-02-08:
@rpcme Thanks for response! I will check a few more things this week and get back to this after that.

angrdtsk, 2021-02-12:
Hi again!

@rpcme I have some new information. First, using Pyro is no longer in plans. Target is now Thud. Also, it seems that this issue is more about hardware support than about versions. There seems to be something in the code that prevents me from building it for ARMv7. Is that something you could somehow help with?

rpcme, 2021-02-13:
hello @angrdtsk arm32 might be a bit rough going.  Definitely something I can help navigate and get a definite answer.  Can you let me know what the error signature is or is there cmake output you can share?

angrdtsk, 2021-02-15:
This is where the build process stops at:
`aws-iot-device-sdk-cpp-v2/1.10.0-r0/git/crt/aws-crt-cpp/crt/aws-checksums/source/arm/crc32c_arm.c:1:0: error: switch -mcpu=cortex-a8 conflicts with -march=armv8-a+crc switch [-Werror]`

I no longer have logs more than that, but I can get them if they are needed.

rpcme, 2021-02-15:
@angrdtsk working on it now. Hopefully can get you unblocked soon ok?

rpcme, 2021-02-15:
Hello @angrdtsk I pulled down recipes from master and needed to do modification for cmake build options. anyway, I could build fine under 'qemuarm'.  if the issue persists, it might be related to the bsp, in which case please let me know what you are using.

angrdtsk, 2021-02-16:
Hi! Sounds good @rpcme ! Would it be possible for you to share those modifications for example as a patch?

rpcme, 2021-02-16:
@angrdtsk I already pushed to the thud branch.  Sorry I wasn't clear: when I say "pulled down" I meant pulled down changes from master to thud.

Btw, do not use the greengrass version on that branch.  it's not supported anymore. I need to go back and update that too.

rpcme, 2021-02-23:
Closing as solved.  Please let me know if otherwise.

closed 2021-02-23 by rpcme

angrdtsk, 2021-02-24:
@rpcme Yes, this is solved. It just had a little delay in informing about that. Thanks for help!