aws4embeddedlinux/meta-aws #109: awsiot install issue in yocto build

elockman, 2021-02-12:
I have a meta/aws layer in the yocto build, with:

> git clone git://github.com/aws/meta-aws -b dunfell yocto/meta-aws

but it appears it wasn't built into the image because there aren't any dependencies asking for it.

I didn't realize there was a recipe for aws-iot-device-sdk-python-v2, but I found it here:

> https://layers.openembedded.org/layerindex/recipe/120360/

I added the recipe to local.conf with 

> CORE_IMAGE_EXTRA_INSTALL += "aws-iot-device-sdk-python-v2"

After Flashing the new image, this is what I see for aws python packages:

> root@sama5d2-xplained:/usr/lib/python3.8/site-packages# ls | grep aws
> _awscrt.cpython-38-arm-linux-gnueabi.so
> awscrt
> awscrt-0.5.11-py3.8.egg-info
> awsiot
> awsiotsdk-1.0.0.dev0-py3.8.egg-info

However, checking versions, it appears awsiot did not install?

> root@sama5d2-xplained:/usr/lib/python3.8/site-packages# pip3 show awscrt | grep Version
> Version: 0.5.11
> root@sama5d2-xplained:/usr/lib/python3.8/site-packages# pip3 show awsiot | grep Version
> WARNING: Package(s) not found: awsiot

elockman, 2021-02-12:
@rpcme can you explain the differences between the following yocto recipes:

aws-iot-device-sdk-python-v2
https://layers.openembedded.org/layerindex/recipe/125260/

python-aws-iot-device-sdk-python
https://layers.openembedded.org/layerindex/recipe/125150/

python3-aws-iot-device-sdk-python
https://layers.openembedded.org/layerindex/recipe/124151/

Should I have used a different recipe with a dunfell build (python 3.8)?

rpcme, 2021-02-13:
Hello @elockman meta-aws is the official AWS device software layer for Yocto. Other layers can implement recipes if they like, but we cannot reassure correctness of those recipes.

The first on your list is the V2 SDK - this is what we would like customers to use when using the device SDK for Python.
The second on your like is the old V1 SDK - please consider it deprecated.
The third on your list is in the meta-python layer.  We don't contribute to that and we cannot reassure its correctness.

State at the moment: we ended up getting into some weird situations when running the recipe with recursive checkout - I am trying to correct this issue on this weekend. I have all the CRT for C components done - all of the language specific SDKs are bindings on top of this.  When I get it sorted on master, I will push to release branches.  Sorry for the delay.

rpcme, 2021-02-13:
Hello @elockman I have fixed the issue and you should have no problem on the dunfell branch.

elockman, 2021-02-15:
@rpcme thanks for the help.  I rebuilt yocto, after seeing this message, but after FLASHING the image, I am still seeing awsiotsdk v1.0.0dev0?

> root@sama5d2-xplained:~# pip3 show awsiotsdk
> Name: awsiotsdk
> Version: 1.0.0.dev0
> Summary: AWS IoT SDK based on the AWS Common Runtime
> Home-page: https://github.com/aws/aws-iot-device-sdk-python-v2
> Author: AWS SDK Common Runtime Team
> Author-email: None
> License: UNKNOWN
> Location: /usr/lib/python3.8/site-packages
> Requires: awscrt
> Required-by:
> root@sama5d2-xplained:~# pip3 show awscrt
> Name: awscrt
> Version: 0.5.11
> Summary: A common runtime for AWS Python projects
> Home-page: https://github.com/awslabs/aws-crt-python
> Author: Amazon Web Services, Inc
> Author-email: aws-sdk-common-runtime@amazon.com
> License: UNKNOWN
> Location: /usr/lib/python3.8/site-packages
> Requires:
> Required-by: awsiotsdk

I see that you moved the **awsiotsdk** commit reference to the head of the main branch (v1.5.4, commit a7739a6f980292ae146e064b18f3f2723019261e), That looks right, but the previous commit reference was only from October 28th (v1.5.3, commit 844bf38ebb13e316ea8100e364121e616c9e71df), so I would have expected the previous recipe to yield v1.5.3 instead of v1.0.0dev0.

I see you also moved the **awscrt** commit reference to the head of the master branch (v0.10.3, commit d020399725065de2999c6743063d7fabf5e0dd9a)  when installing **awsiotsdk** with pypi, **awsiotsdk** requests **awscrt** version v0.9.15.  Should the **awscrt** commit reference point there? (v0.9.15, commit 2584495f0103cc5056c11823ee87b352625e8356)?

Is it possible the install is failing on dependencies, and somehow reverting to v1.0.0dev0.

rpcme, 2021-02-15:
I think I see what's going on here.  The default version for setup.py is that version number.   The code itself should be right, I just need to tweak the recipe so pip emits properly.  I never install pip on embedded linux because it should "just run" so I never saw this. I see the value in what you're doing so let me close out another thing I'm working on and then I will see what I can do about the label ok?

rpcme, 2021-02-15:
Hello @elockman I have pushed changes.  I verified with pip in qemu.

[image]

It's on dunfell but still need to propagate to other branches.

elockman, 2021-02-15:
@rpcme Thanks, again.

Will I need to add **IMAGE_INSTALL_append = "python3-distro"** to local.conf in my dunfell build?

rpcme, 2021-02-15:
I didn't.  the only thing I had in my local.conf is:
`IMAGE_INSTALL_append = " aws-iot-device-sdk-python-v2 python3-pip"`

rpcme, 2021-02-15:
Also, I noted the license wasn't showing properly in the output so I've supplied pull requests to both SDK repos to fix that.

elockman, 2021-02-16:
**GOLD STAR for @rpcme!**

I now have AWSIOTSDK installed on my Linux Build.

> root@sama5d2-xplained:~ # pip3 show awscrt | grep Version
> Version: 0.10.3
> root@sama5d2-xplained:~ # pip3 show awsiotsdk | grep Version
> Version: 1.5.4
> 

Thank You!

rpcme, 2021-02-16:
@elockman I am very happy we got this resolved for you!  Would like to see what you're building on the SAMA5D2 (I'm a Microchip fan) - drop me a note on LinkedIn when you're ready.

closed 2021-02-16 by rpcme

nealdeore, 2023-06-02:
I am trying to install the aws-iot sdk on my yocto build.

IMAGE_INSTALL_append = " aws-iot-device-sdk-python-v2 python3-pip"

but i get below error - 
ERROR: Nothing PROVIDES 'aws-iot-device-sdk-python-v2' (but /home/.../poky/recipes-core/core-imgae-minimal.bb RDEPENDS on or otherwise requires it)

ERROR: Required build target 'core-image-minimal' has no buildable providers/
Missing or unbuildable dependency chain was: ['core-image-minimal', 'aws-iot-device-sdk-python-v2']

Please help I am new to yocto trying to learn.

building image for Raspberrypi4.