aws4embeddedlinux/meta-aws #110: [greengrass-bin] The first deployment fails with NoSuchFileException

kmeshcheryakov, 2021-02-26:
When the Greengrass receives the first deployment it fails with the following exception: 

```
java.nio.file.NoSuchFileException: /init
        at java.base/sun.nio.fs.UnixException.translateToIOException(UnixException.java:92)  
        at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:111)   
        at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:116)   
        at java.base/sun.nio.fs.UnixFileAttributeViews$Basic.readAttributes(UnixFileAttributeViews.java:55)
        at java.base/sun.nio.fs.UnixFileSystemProvider.readAttributes(UnixFileSystemProvider.java:149)
        at java.base/sun.nio.fs.LinuxFileSystemProvider.readAttributes(LinuxFileSystemProvider.java:99)
        at java.base/java.nio.file.Files.readAttributes(Files.java:1763)                     
        at java.base/java.nio.file.FileTreeWalker.getAttributes(FileTreeWalker.java:219)     
        at java.base/java.nio.file.FileTreeWalker.visit(FileTreeWalker.java:276)             
        at java.base/java.nio.file.FileTreeWalker.walk(FileTreeWalker.java:322)              
        at java.base/java.nio.file.Files.walkFileTree(Files.java:2716)                       
        at java.base/java.nio.file.Files.walkFileTree(Files.java:2796)                       
        at com.aws.greengrass.util.Utils.copyFolderRecursively(Utils.java:673)               
        at com.aws.greengrass.lifecyclemanager.KernelAlternatives.prepareBootstrap(KernelAlternatives.java:267)
        at com.aws.greengrass.deployment.activator.KernelUpdateActivator.activate(KernelUpdateActivator.java:77)
        at com.aws.greengrass.deployment.DeploymentConfigMerger.updateActionForDeployment(DeploymentConfigMerger.java:128)
        at com.aws.greengrass.deployment.DeploymentConfigMerger.lambda$mergeInNewConfig$0(DeploymentConfigMerger.java:91)
        at com.aws.greengrass.lifecyclemanager.UpdateSystemPolicyService.runUpdateActions(UpdateSystemPolicyService.java:94)
```

The problem seems to be caused by the incorrect initial `${GG_ROOT}/alts/current` symbolic link. When it should use an absolute path, it uses the relative:

greengrass-bin_2.0.3.bb, line 42: 
```
ln -s init current
```

The possible fix is to change the line to

```
ln -s ${GG_ROOT}/alts/init current
```

rpcme, 2021-02-26:
@kmeshcheryakov Thank you for the feedback.  Were you able to confirm this is the fix?  I will not have the opportunity to do a full pass until next Tuesday.

lukaszs-lars, 2021-03-03:
I had a similar issue and the solution was:
`ln -s /${GG_BASENAME}/alts/init current`

because `${GG_ROOT}` points to a directory on the host

The deployment ended with success however I got the following error and I don't know if it will not produce further errors in the future:

```
[ERROR] (pool-2-thread-11) com.aws.greengrass.deployment.DeploymentService: Failed to reset Nucleus activate directory. {serviceName=DeploymentService, currentState=RUNNING}

java.nio.file.DirectoryNotEmptyException: /greengrass/v2/alts/init/distro
        at sun.nio.fs.UnixFileSystemProvider.implDelete(Unknown Source)
        at sun.nio.fs.AbstractFileSystemProvider.deleteIfExists(Unknown Source)
        at java.nio.file.Files.deleteIfExists(Unknown Source)
        at com.aws.greengrass.lifecyclemanager.KernelAlternatives.cleanupLaunchDirectorySingleLevel(KernelAlternatives.java:313)
        at com.aws.greengrass.lifecyclemanager.KernelAlternatives.activationSucceeds(KernelAlternatives.java:225)
        at com.aws.greengrass.deployment.DeploymentService.finishCurrentDeployment(DeploymentService.java:329)
        at com.aws.greengrass.deployment.DeploymentService.startup(DeploymentService.java:187)
        at com.aws.greengrass.lifecyclemanager.Lifecycle.lambda$handleStateTransitionStartingToRunningAsync$9(Lifecycle.java:531)
        at java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
        at java.util.concurrent.FutureTask.run(Unknown Source)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
        at java.lang.Thread.run(Unknown Source)
```

rpcme, 2021-03-24:
@lukaszs-lars I am sorry for the late response. Just now recovering from Embedded World.  Working on this now.

rpcme, 2021-08-05:
@lukaszs-lars modern versions of greengrass v2 recipes have landed.  Please let us know if this works for you.  Will close this issue since it's kind of old, please reopen if you're still facing the issue ok?

asymptotic, 2023-01-07:
I know this is really old now, but I'm still having this same problem with the current greengrass-bin recipe on kirkstone. I didn't see this until just know, I tried changing the ${GG_ROOT} to a relative link to fix the host link problem and when I got greengrass errors that's what brought me here. Same issue as in #269. 

Looks like it's the same in more up to date branches, can confirm that the ${GG_BASENAME} change above works. I'm doing this with a bbappend, appending to do_install a line of `cd /${GG_ROOT}/alts && rm current && ln -s /${GG_BASENAME}/alts/init /${GG_BASENAME}/alts/current`. But it would be great to have the fix in the main recipe! Asking instead of sending a PR because it looks like there's a lot of process to PRs :)

asymptotic, 2023-01-09:
@thomas-roos I started writing my post and then came to the same conclusion as you, thought I edited it, but apparently didn't from what I posted above. Ugh, sorry about that.

This is my actual working bbappend:

```
do_install:append() {
  cd /${GG_ROOT}/alts && rm current && ln -s /${GG_BASENAME}/alts/init /${GG_ROOT}/alts/current
}
```

So I can confirm changing the line that you pointed out will fix it. Thank you!

closed 2023-01-09 by thomas-roos