aws4embeddedlinux/meta-aws #126: Secure Tunneling Local Proxy requires Boost 1.69.0

dhwalters423, 2021-05-12:
The Secure Tunneling local proxy build fails with several errors from a clean environment. In order to successfully build, you need to downgrade the recipe for Boost to [Boost 1.69.0](https://github.com/aws-samples/aws-iot-securetunneling-localproxy#2-download-and-install-boost-dependency). The standard distribution from [Poky is version 1.76](https://git.yoctoproject.org/cgit.cgi/poky/plain/meta/recipes-support/boost/).

```
Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-fsl-linux"
MACHINE              = "s32g274ardb2"
DISTRO               = "fsl-auto"
DISTRO_VERSION       = "1.0"
TUNE_FEATURES        = "aarch64"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:0275e68b0576cdefc19718e4b1977044f9f43f48"
meta-oe              
meta-multimedia      
meta-python          
meta-networking      
meta-gnome           
meta-filesystems     
meta-webserver       
meta-perl            
meta-xfce            = "HEAD:aad5b3d070cd8c58828b0975cf861d8ebc90f460"
meta-virtualization  = "HEAD:04e0d9de8cbaeb76aff6e78b90941a6e9a359b87"
meta-optee           = "HEAD:7603aadcc0c1544588203b6a13ecbf28c50d158d"
meta-freescale       = "HEAD:0eb21f9d0d023fbb054e284986815510a84dbd66"
meta-alb             = "HEAD:b45a1fdc08da8b0df3871bb03126379bbec58c1f"
meta-aws             = "master:5bbe0d21b67f3b0451149ea99d600f53f6030d34"
meta-java            = "HEAD:95d56a63cdbf1ed9db99a1f02075568fd45a668b"
meta-nxp-greengrass  = "HEAD:b94a8dacc6733b9871467d4dcf8c42f59c38b401"
meta-tensorflow      = "HEAD:17fe99fa101a51d62e55e0d4372b9301f1f5fbe3"
meta-linaro-toolchain = "HEAD:7603aadcc0c1544588203b6a13ecbf28c50d158d"`
```

Work around to downgrade to Boost 1.69.0:

1.  rename sources/poky/meta/recipes-support/boost/bjam-native_1.69.0
2. rename sources/poky/meta/recipes-support/boost/boost-1.69.0.inc
3. rename sources/poky/meta/recipes-support/boost/boost-1.69.0.bb
4. Remove patch ' file://0001-Don-t-set-up-arch-instruction-set-flags-we-do-that-o.patch' from boost-1.69.0.bb
5. Update md5 and sha256 checkums in boost-1.69.0:
``
SRC_URI[md5sum] = "a1332494397bf48332cb152abfefcec2"
SRC_URI[sha256sum] = "8f32d4617390d1c2d16f26a27ab60d97807b35440d45891fa340fc2648b04406"
``

rpcme, 2021-06-03:
Hello @dhwalters423 it seems as though I have something working.  The instructions you provided were a bit incomplete so I needed to trudge my way through it a bit.  Can you please check it on the mainline?  The changes are committed.

Summarized: boost 1.69.0 was added to meta-aws/recipes-support/boost, propagated from zeus.  I set the preferred version in the recipe.  There are warnings about multiple providers for some boost items.

https://github.com/aws/meta-aws/commit/de4889367780f8403895bdc904a0061497ed56cf
https://github.com/aws/meta-aws/commit/f7b728017d632f44079109489018f845a598eace

In the meantime, I will try to clean up the warnings.  But please check it.

rpcme, 2021-06-04:
hello @dhwalters423 

Due to issues with layercheck trying to resolve this issue, I have removed the recipe from the mainline.

An issue has been posted to the repo maintainer for the software:

https://github.com/aws-samples/aws-iot-securetunneling-localproxy/issues/51

When the project starts keeping boost up to date we can continue to maintain the recipe in this project.

Closing this issue.

closed 2021-06-04 by rpcme

kareali, 2021-07-26:
Just updated localproxy to bump up boost to use 1.76