aws4embeddedlinux/meta-aws #130: Greengrass V2 support in zeus branch

kennu, 2021-05-25:
Hi! I'm working on migrating a zeus-based project from Greengrass V1 to V2. Currently it looks like zeus only has Greengrass V1, while V2 support is in master. I wonder if we could have it in zeus too?

rpcme, 2021-05-25:
Hello @kennu 

I will discuss in the team when we can get it propagated and tested.  When doing that should probably refactor the recipe-* dirs as well. Otherwise it might be a bit difficult to track.

If you would like to pull down the recipe from master (you will need corretto as well - see recipe requirements) and do a PR I'm ok with that too.  I don't have any reason to believe ggv2 will not work on zeus.

kennu, 2021-05-25:
Thanks @rpcme I will experiment with in the near future and see if I can get our stuff running.

kennu, 2021-05-26:
So far I have Greengrass V2 running on the device, but I'm having some trouble configuring it.

I tried to put `posixUser: ggc_user:ggc_group` and `awsRegion: eu-west-1` into https://github.com/aws/meta-aws/blob/master/recipes-iot/aws-iot-greengrass/files/greengrassv2-init.yaml but it seems that Greengrass is not picking those up. After booting the device I see `awsRegion: "us-east-1"` and `runWithDefault: {}` in effectiveConfig.yaml. Greengrass then dies, being unable to call `usermod` because we don't have that command in the Linux image.

If you can offer any guidance into the configuration process @rpcme, I'd be very thankful. Official Greengrass documentation talks about running a Java installer, but I'm not sure how that works in Yocto Linux.

rpcme, 2021-08-05:
Hello @kennu can this be closed?  There are modern Greengrass recipes on master and dunfell.  I am planning on propagating once more to zeus as a final active maintenance run on that release branch.

kennu, 2021-08-23:
I guess I can close this since the support was added. I can make more tickets if I run into any trouble.

closed 2021-08-23 by kennu