aws4embeddedlinux/meta-aws #131: greengrass-bin recipe should generate config.yaml (not config.yaml.clean)

kennu, 2021-05-26:
In https://github.com/aws/meta-aws/blob/master/recipes-iot/aws-iot-greengrass/greengrass-bin_2.0.3.bb#L35 the recipe is creating `config.yaml.clean` which has no effect. I was wondering why my changes to greengrassv2-init.yaml didn't work. Greengrass will only read initial settings from `config.yaml`.

Thanks to @MikeDombo in https://github.com/aws-greengrass/aws-greengrass-nucleus/issues/830 for explaining the config loading process. :-)

MikeDombo, 2021-05-26:
I believe it is getting setup here: https://github.com/aws/meta-aws/blob/master/recipes-iot/aws-iot-greengrass/greengrass-bin-demo_2.0.3.bb#L30

kennu, 2021-05-26:
Yeah it's a separate recipe for greengrass-bin-demo vs. greengrass-bin. I've been using greengrass-bin. I just assumed demo is not for real use..

kennu, 2021-05-26:
If the intent is to create my own recipe or bbappend greengrass-bin, that's fine too, but needs some documentation / instructions. :-)

rpcme, 2021-05-26:
The purpose of -bin-demo is for persons creating a one-off demo.  bridging the gap between install and a working install in -bin was TBD - since customers can implement any number of provisioning schemes.  Sorry for the confusion and non-documentation on this point.

MikeDombo, 2021-05-26:
You should look at https://docs.aws.amazon.com/greengrass/v2/developerguide/manual-installation.html

kennu, 2021-05-26:
From end-user perspective, the challenge is that I didn't know which parts of https://docs.aws.amazon.com/greengrass/v2/developerguide/ apply when using the meta-aws recipe. I couldn't be sure whether any of the Installation sections apply, since I would have assumed the installation to happen while BitBake builds the image. 

But now I understand that there is no Greengrass installation during BitBake. The systemd service just launches Greengrass.jar at boot and Greengrass performs an auto-installation on every startup. So essentially, https://docs.aws.amazon.com/greengrass/v2/developerguide/manual-installation.html#run-greengrass-core-v2-installer-manual occurs again on every boot.

First it was a complete black box for me how greengrassv2-init.yaml ends up into effectiveConfig.yaml. Now  I understand  that the greengrass-bin recipe writes it into /greengrass/v2/config/config.yaml, and Greengrass picks it up from there (when config.tlog doesn't yet exist). During future boots, there will already be a config.tlog and Greengrass will ignore the original config.yaml.

I'm certainly going to document all this internally for our other guys. :-)

rpcme, 2021-05-26:
@kennu Thank you for this - also there is a README you can edit if you like with the details.  We are happy to learn through this. I think my problem is I always used -bin-demo and never experienced the merry-go-round.  Now we know what requires targeting but still can use the feedback.

rpcme, 2021-06-02:
@kennu Do you have any update?  I am doing some refactoring on the zeus branch (specifically to include Iot Device Client there).  I might seed the greengrass recipes just for completion but I don't want to mess your work up.

kennu, 2021-06-02:
@rpcme I've created my own local recipe based on greengrass-bin and made small changes to it. I'm still working on getting all the policies and other configurations right in AWS so it's taking me a while to get our apps running. When the whole thing is finally working, I can check what the situation is with the official recipe and compare my changes. I think we're also using a locally modified version of the greengrass (V1) recipe to support read-only rootfs, so changes in meta-aws aren't directly affecting that either.

philcali, 2021-08-05:
Hey there @kennu, I'm in the process of updating the [README][1] for using the `bin-demo` approach for a pre-installed config, and I was wondering if there was any additional clarifications that would have helped you out to what you described [here][2].

[1]: https://github.com/aws/meta-aws/tree/master/recipes-iot/aws-iot-greengrass
[2]: https://github.com/aws/meta-aws/issues/131#issuecomment-849161452

kennu, 2021-08-23:
@philcali In general, I think it is useful to document how various files are created on the Yocto target image. This helps when the recipe is not just copying a directory hierarchy verbatim, but auto-generating files using some logic and using various source files.

The situation I face is usually that I ssh into the newly built Linux image on a test device, and take a look at what files are there. Then I start wondering how did they get there from the recipe, and how can I modify them. It's tricky if the file is not mentioned anywhere. In cases like effectiveConfig.yaml, it's useful to document that it's not created by the recipe, but it's created by Greengrass at runtime based on another file that is created by the recipe. It took a while to figure that out..

(Sorry to take long to respond, I've been involved in some time-consuming extra projects recently.)

rpcme, 2021-08-23:
@kennu thank you for the valuable feedback.

@philcali would love to see how we can improve the devex here - should there be a table naming volatile/non-volatile files?  would like to hear better approaches.

rpcme, 2021-09-01:
@kennu - I've spoken with the Greengrass team and they will look towards centralizing this information in the product documentation.  Please reopen this issue if necessary.