aws4embeddedlinux/meta-aws #133: Fatal error condition occurred in /usr/src/debug/aws-c-mqtt/0.5.5-r0/git/aws-c-mqtt

sskrn, 2021-06-08:
Hello..
I am using meta-aws layer (dunfell) for the Yocto image on my Nano Pi Duo2 Board. I did compiling this layer with configuration in my image.conf with configuration: 

IMAGE_INSTALL_append = " aws-iot-device-sdk-python-v2"

then, the image compiled successfully and on my board, i can import awscrt on Python3 shell. But when i tried to run sample pubsub.py with proper cert files, it shows this error : 

Running pub/sub sample application...
Connecting to xxxxx-ats.iot.ap-southeast-1.amazonaws.com with client ID 't
est-6d6b1c72-47ef-4d6a-967a-ef3b46ee85a8'...
Fatal error condition occurred in /usr/src/debug/aws-c-mqtt/0.5.5-r0/git/aws-c-mqtt
/source/packets.c:158: username.len > 0
Exiting Application
################################################################################
Resolved stacktrace:
################################################################################
################################################################################
Raw stacktrace:
################################################################################
Signal received: 0, errno: 0
################################################################################
Resolved stacktrace:
################################################################################
################################################################################
Raw stacktrace:
################################################################################

I'm confused because i can run the sdk and my custom scripts on my PC and vendor-built Linux Image just fine.

rpcme, 2021-06-08:
Hello @sskrn - I don't have this board and not sure of the BSP you're using.  Can you please let me know? I can try to repro on dunfell for raspberry pi3 32 bit?

rpcme, 2021-06-08:
Hello @sskrn 

I was unable to reproduce the issue.  I used the tip revisions of the dunfell branches for the following layers:

```text
BBLAYERS ?= " \
  /worktree/poky-dunfell/meta \
  /worktree/poky-dunfell/meta-poky \
  /worktree/poky-dunfell/meta-yocto-bsp \
  /worktree/poky-dunfell/meta-openembedded/meta-oe \
  /worktree/poky-dunfell/meta-openembedded/meta-python \
  /worktree/poky-dunfell/meta-openembedded/meta-filesystems \
  /worktree/poky-dunfell/meta-openembedded/meta-networking \
  /worktree/poky-dunfell/meta-raspberrypi \
  /worktree/poky-dunfell/meta-virtualization \
  /worktree/poky-dunfell/meta-aws \
  "
```

I used the following conf:

```text
MACHINE = "raspberrypi4-64"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
IMAGE_INSTALL_append = " aws-iot-device-sdk-python-v2"
IMAGE_FSTYPES = "rpi-sdimg"

IMAGE_INSTALL_append = " strace"
DISTRO_FEATURES += "systemd"
DISTRO_FEATURES += "virtualization"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = ""

AWS_IOT_GG = "openssh                                \
              ntp                                    \
              docker                                 \
              python3-docker-compose                 \
              "

RPI4_WIFI =  "crda                                   \
              iw                                     \
              linux-firmware-rpidistro-bcm43455      \
              wpa-supplicant                         \
              "

RPI4_BT =    "bluez-firmware-rpidistro-bcm4345c0-hcd \
              bluez5                                 \
              bluez5-dev                             \
              bluez5-noinst-tools                    \
              bluez5-obex                            \
              bluez5-testtools                       \
              python3-pybluez                        \
              "

RPI4_NET =   "i2c-tools                              \
              python3-smbus                          \
              bridge-utils                           \
              hostapd                                \
              dhcpcd                                 \
              iptables                               \
              wpa-supplicant                         \
              ca-certificates                        \
              "

IMAGE_INSTALL_append = " ${AWS_IOT_GG}"
IMAGE_INSTALL_append = " ${RPI4_WIFI}"
IMAGE_INSTALL_append = " ${RPI4_BT}"
IMAGE_INSTALL_append = " ${RPI4_NET}"

IMAGE_INSTALL_append = "kernel-modules"
```

and invoked the pubsub script with no issues.  I did need to modify the policy based on the instruction here.  https://github.com/aws/aws-iot-device-sdk-python-v2/tree/main/samples#pubsub

Having said that, I am a little concerned that in a number of the recipes we are not passing ```-DCMAKE_BUILD_TYPE=Release``` so I will revisit that, but it does not seem to causing any issue with cloud connectivity.

sskrn, 2021-06-09:
Thank you for the quick and detail response.. I am using this meta-allwinner layer: https://gitlab.com/dimtass/meta-allwinner-hx . Also i made my own layer for customization in image recipe as shown below:

`SUMMARY = "Allwinner console image"
LICENSE = "MIT"
AUTHOR = "Dimitris Tassopoulos"

inherit allwinner-base-image

inherit extrausers
EXTRA_USERS_PARAMS = "usermod -P qwer1234 root"

#DISTRO_FEATURES_append += "bluez5"
MACHINE_FEATURES_append += "bluetooth"

SSG_BT = " bluez5                                 \
            bluez5-dev                             \
            bluez5-noinst-tools                    \
            bluez5-obex                            \
            bluez5-testtools                       \
            python3-pybluez                        \
          "

#IMAGE_INSTALL += "htop"
IMAGE_INSTALL_append = " alsa-lib alsa-utils alsa-plugins alsa-state alsa-tools"
IMAGE_INSTALL_append = " screen"
IMAGE_INSTALL_append = " i2c-tools vim git"
IMAGE_INSTALL_append = " ${SSG_BT}"
#IMAGE_INSTALL_append = " simplehello"
IMAGE_INSTALL_append = " aws-iot-device-sdk-python-v2"
IMAGE_FEATURES = "ssh-server-dropbear"
`

is there any dependencies that should be included in IMAGE_INSTALL variable or IMAGE_FEATURES for this library? Sorry I'm still literally new to Yocto and perhaps still trying to understand in adding custom packages to the image.

rpcme, 2021-06-09:
I think this looks ok.

For some reason I initially thought this board was a Raspberry Pi clone but it's not.  It's totally its own thing.

I am not aware of this BSP - first time seeing it.  Never worked with Allwinner chips before.

I would like to confirm that this is the board:

https://www.armbian.com/nanopi-duo-2/

We will have to get a board to try to reproduce.

sskrn, 2021-06-10:
Yes correct, that link refers to the board. Is there possibility that if I develop the script using C++ this error won't occur? I would appreciate it if you can give me some suggestions in using AWS IoT in Yocto Project.

rpcme, 2021-06-10:
You're using the preferred SDK for Python.  The C++ SDK uses the same underlying CRT so we need to narrow what the potential root cause might be.

While we get the board on hand, let me ask a few more questions.

The error infers that you have reached this block of code:

https://github.com/awslabs/aws-c-mqtt/blob/8984c51bc834598e3cb1e73ae5d7d614b445ce48/source/packets.c#L158

Which is strange given that /usually/ an autogenerated client id will be made if you didn't provide one.  Maybe the demo script for whatever reason was unable to provide one.  I see that in the output of the script you provided in the first post.

Can you give me the command line of how you're invoking the script?

Also, I will try invoking the script for the target region just to cover off that difference in my own testing.

sskrn, 2021-06-11:
Here is the bash script i used  when invoking the script. Sorry i have to block some text because it's my company's server.

```
#!/usr/bin/env bash
printf "\nRunning pub/sub sample application...\n"
python3 pubsub.py --endpoint xxxxxxxx-ats.iot.ap-southeast-1.amazonaws.com --root-ca root-CA.crt --cert xxxxxxxx-certificate.pem.crt --key xxxxxxx-private.pem.key
```
Ok i will try the C++ SDK later, the PC i usually used for building Yocto is still used for other projects.

rpcme, 2021-07-17:
@sskrn were you able to move forward here?

rpcme, 2021-08-12:
@sskrn please reeopen if needed.

WolfWalter, 2021-12-02:
I am stuck at the same problem with a raspi 4 honister image.

```
python3 pubsub.py --endpoint xxxxxxxxx-ats.iot.eu-central-1.amazonaws.com --cert 1f677c799e-certificate.pem.crt --key
  1f677c799e-private.pem.key --root-ca AmazonRootCA1.pem --client-id test-1 --topic test/topic --message "test" --verbosity Trace
[INFO] [2021-12-02T19:20:21Z] [0000007fa26ce010] [event-loop] - id=0x556e43cda0: Initializing edge-triggered epoll
[INFO] [2021-12-02T19:20:21Z] [0000007fa26ce010] [event-loop] - id=0x556e43cda0: Using eventfd for cross-thread notifications.
[TRACE] [2021-12-02T19:20:21Z] [0000007fa26ce010] [event-loop] - id=0x556e43cda0: eventfd descriptor 5.
[INFO] [2021-12-02T19:20:21Z] [0000007fa26ce010] [event-loop] - id=0x556e43cda0: Starting event-loop thread.
[INFO] [2021-12-02T19:20:21Z] [0000007fa26ce010] [dns] - id=0x556e437db0: Initializing default host resolver with 16 max host entries.
[INFO] [2021-12-02T19:20:21Z] [0000007fa0c9e150] [event-loop] - id=0x556e43cda0: main loop started
[TRACE] [2021-12-02T19:20:21Z] [0000007fa0c9e150] [event-loop] - id=0x556e43cda0: subscribing to events on fd 5
[INFO] [2021-12-02T19:20:21Z] [0000007fa26ce010] [channel-bootstrap] - id=0x556e3dc6e0: Initializing client bootstrap with event-loop group 0x556e43cd20
[INFO] [2021-12-02T19:20:21Z] [0000007fa0c9e150] [event-loop] - id=0x556e43cda0: default timeout 100000, and max events to process per tick 100
[TRACE] [2021-12-02T19:20:21Z] [0000007fa0c9e150] [event-loop] - id=0x556e43cda0: waiting for a maximum of 100000 ms
[DEBUG] [2021-12-02T19:20:22Z] [0000007fa26ce010] [tls-handler] - ctx: Certificate and key have been set, setting them up now.
[DEBUG] [2021-12-02T19:20:22Z] [0000007fa26ce010] [tls-handler] - ctx: Setting ALPN list x-amzn-mqtt-ca
[DEBUG] [2021-12-02T19:20:22Z] [0000007fa26ce010] [mqtt-client] - client=0x556e415f30: Initalizing MQTT client
[DEBUG] [2021-12-02T19:20:22Z] [0000007fa26ce010] [mqtt-client] - id=0x556e3dc750: Creating new connection
[DEBUG] [2021-12-02T19:20:22Z] [0000007fa26ce010] [mqtt-topic-tree] - tree=0x556e3dc9b8: Creating new topic tree
[TRACE] [2021-12-02T19:20:22Z] [0000007fa26ce010] [mqtt-client] - id=0x556e3dc750: Setting connection interrupted and resumed handlers
Connecting to a92fa5i0qisfk-ats.iot.eu-central-1.amazonaws.com with client ID 'test-1'...
[TRACE] [2021-12-02T19:20:22Z] [0000007fa26ce010] [mqtt-client] - id=0x556e3dc750: Setting reconnect timeouts min: 5 max: 60
[TRACE] [2021-12-02T19:20:22Z] [0000007fa26ce010] [mqtt-client] - id=0x556e3dc750: Setting username and password
[TRACE] [2021-12-02T19:20:22Z] [0000007fa26ce010] [mqtt-client] - id=0x556e3dc750: Opening connection
[DEBUG] [2021-12-02T19:20:22Z] [0000007fa26ce010] [mqtt-client] - id=0x556e3dc750: Begin connecting process, switch state to CONNECTING.
[INFO] [2021-12-02T19:20:22Z] [0000007fa26ce010] [mqtt-client] - id=0x556e3dc750: using ping timeout of 3000000000 ns
[TRACE] [2021-12-02T19:20:22Z] [0000007fa26ce010] [channel-bootstrap] - id=0x556e3dc6e0: attempting to initialize a new client channel to a92fa5i0qisfk-ats.iot.eu-central-1.amazonaws.com:443
[DEBUG] [2021-12-02T19:20:22Z] [0000007fa26ce010] [dns] - id=0x556e437db0: Host resolution requested for a92fa5i0qisfk-ats.iot.eu-central-1.amazonaws.com
[DEBUG] [2021-12-02T19:20:22Z] [0000007fa26ce010] [dns] - id=0x556e437db0: No cached entries found for a92fa5i0qisfk-ats.iot.eu-central-1.amazonaws.com starting new resolver thread.
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolving host a92fa5i0qisfk-ats.iot.eu-central-1.amazonaws.com
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 3.126.105.68
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 3.120.20.65
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 18.194.128.105
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 3.125.78.235
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 18.196.221.90
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 35.157.100.95
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 18.185.187.250
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 35.156.171.146
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 2a01:578:13::37d:4eeb
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 2a01:578:13::12b9:bbfa
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 2a01:578:13::12c4:e5aa
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 2a01:578:13::37e:6944
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 2a01:578:13::239d:645f
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 2a01:578:13::12c4:dd5a
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 2a01:578:13::12c0:59b6
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static: resolved record: 2a01:578:13::37a:c699
[DEBUG] [2021-12-02T19:20:22Z] [0000007f9bfff150] [dns] - static, resolving host a92fa5i0qisfk-ats.iot.eu-central-1.amazonaws.com successful, returned 16 addresses
[... cut ...]
/usr/lib/libaws-c-io.so.1.0.0(+0xd3e8) [0x7fa1c633e8]
/usr/lib/libaws-c-io.so.1.0.0(+0x246b4) [0x7fa1c7a6b4]
/usr/lib/libaws-c-io.so.1.0.0(+0x24d14) [0x7fa1c7ad14]
/usr/lib/libaws-c-io.so.1.0.0(+0x276b4) [0x7fa1c7d6b4]
/usr/lib/libaws-c-io.so.1.0.0(+0x27958) [0x7fa1c7d958]
/usr/lib/libaws-c-io.so.1.0.0(+0x233d4) [0x7fa1c793d4]
/usr/lib/libaws-c-io.so.1.0.0(+0x19920) [0x7fa1c6f920]
/usr/lib/libaws-c-common.so.1(+0x1da50) [0x7fa1c15a50]
/lib/libc.so.6(+0x803d8) [0x7fa21e53d8]
/lib/libc.so.6(+0xe6f1c) [0x7fa224bf1c]
```

WolfWalter, 2021-12-04:
Anything I could do to track down the issue @rpcme?

rpcme, 2021-12-05:
Hi @WolfWalter sorry about the delay.  I was at re:Invent and wasn't checking messages, and this issue is Closed so I only saw this when checking email.  I will reopen and see if we can repro.  @RephaelCongmon

RephaelCongmon, 2021-12-08:
Hello @WolfWalter , may I ask you to share your local.conf that you used to build the image (don't include any confidential information)? It can help us reproduce the issue faster.

WolfWalter, 2021-12-09:
Hi @RephaelCongmon,
my local.conf:
```
# debug-tweaks
EXTRA_IMAGE_FEATURES = "debug-tweaks"

# diskmon
BB_DISKMON_DIRS = "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"

# standard
CONF_VERSION = "2"
PACKAGE_CLASSES = "package_rpm"
SDKMACHINE = "x86_64"
USER_CLASSES = "buildstats image-prelink"
PATCHRESOLVE = "noop"
RPI_EXTRA_CONFIG = ' \n \
  kernel=kernel_rpilinux.img\n \
  arm_64bit=1\n \
  enable_uart=1\n \
  '

MACHINE ??= "raspberrypi4-64"
DISTRO ??= "poky"
BBMULTICONFIG ?= ""
```

I am using kas to build the image. Currently the is nothing confidential in my setup. Just exploring aws iot for raspberry pi with yocto . So if it helps I could clean it up and share a repo to repoduce the issue.

RephaelCongmon, 2021-12-10:
Hello @WolfWalter ,
I was not able to reproduce the issue outside of kas. The next step for me to try to repro is to copy your exact configuration and build with kas and then see if I can repro. When you run a yocto build, can you please share with me the exact hashes for your configuration?

One clarifying question: are you using kas just to build and you're running it on your Raspberry Pi 4?

Also, sharing your repo could be helpful as well so it would be great if you could share that with me.

WolfWalter, 2021-12-11:
Hi @RephaelCongmon,
here is the [repo](https://github.com/WolfWalter/meta-aws-platform-bsp). Just playing around there. Nothing serious yet. Yes I am using a raspberry pi 4. 

Here ist the output of `kas for-all-repos description.yml 'git status'` on my machine:

```
2021-12-11 13:04:36 - INFO     - kas 2.5 started
2021-12-11 13:04:36 - INFO     - /home/wolf/development/meta-aws-platform-bsp$ git rev-parse --show-toplevel
2021-12-11 13:04:36 - INFO     - /home/wolf/development/meta-aws-platform-bsp$ git rev-parse --show-toplevel
2021-12-11 13:04:36 - INFO     - Using /home/wolf/development/meta-aws-platform-bsp as root for repository meta-solvispi
2021-12-11 13:04:36 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky$ git remote set-url origin https://git.yoctoproject.org/git/poky
2021-12-11 13:04:36 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-openembedded$ git remote set-url origin http://git.openembedded.org/meta-openembedded
2021-12-11 13:04:36 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-raspberrypi$ git remote set-url origin https://github.com/agherzan/meta-raspberrypi.git
2021-12-11 13:04:36 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-aws$ git remote set-url origin https://github.com/aws/meta-aws.git
2021-12-11 13:04:36 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky$ git cat-file -t honister
2021-12-11 13:04:36 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-aws$ git cat-file -t master
2021-12-11 13:04:36 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-raspberrypi$ git cat-file -t honister
2021-12-11 13:04:36 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-openembedded$ git cat-file -t honister
2021-12-11 13:04:36 - INFO     - Repository poky already contains honister as commit
2021-12-11 13:04:36 - INFO     - Repository meta-aws already contains master as commit
2021-12-11 13:04:36 - INFO     - Repository meta-raspberrypi already contains honister as commit
2021-12-11 13:04:36 - INFO     - Repository meta-openembedded already contains honister as commit
2021-12-11 13:04:36 - INFO     - /home/wolf/development/meta-aws-platform-bsp$ git rev-parse --show-toplevel
2021-12-11 13:04:36 - INFO     - Using /home/wolf/development/meta-aws-platform-bsp as root for repository meta-solvispi
2021-12-11 13:04:36 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky$ git status -s
2021-12-11 13:04:36 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky$ git rev-parse --verify -q origin/honister
2021-12-11 13:04:36 - INFO     - e3d86eb7383d32cacef82033ef52d646d2a7a0ea
2021-12-11 13:04:36 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky$ git checkout -q e3d86eb7383d32cacef82033ef52d646d2a7a0ea -B honister
2021-12-11 13:04:36 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-openembedded$ git status -s
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-openembedded$ git rev-parse --verify -q origin/honister
2021-12-11 13:04:37 - INFO     - ad52a41de8b4b7d619d1376d0a0090ebcfff56da
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-openembedded$ git checkout -q ad52a41de8b4b7d619d1376d0a0090ebcfff56da -B honister
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-raspberrypi$ git status -s
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-raspberrypi$ git rev-parse --verify -q origin/honister
2021-12-11 13:04:37 - INFO     - 1584bddcf31f4bf3acc2304aa8dae927e8bec970
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-raspberrypi$ git checkout -q 1584bddcf31f4bf3acc2304aa8dae927e8bec970 -B honister
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-aws$ git status -s
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-aws$ git rev-parse --verify -q origin/master
2021-12-11 13:04:37 - INFO     - 8893e0cd4c0981eeda941eaa9ad2eb9359670502
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-aws$ git checkout -q 8893e0cd4c0981eeda941eaa9ad2eb9359670502 -B master
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp$ git rev-parse --show-toplevel
2021-12-11 13:04:37 - INFO     - Using /home/wolf/development/meta-aws-platform-bsp as root for repository meta-solvispi
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp$ git rev-parse --show-toplevel
2021-12-11 13:04:37 - INFO     - Using /home/wolf/development/meta-aws-platform-bsp as root for repository meta-solvispi
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky$ /tmp/tmponpyilf2/get_bb_env /home/wolf/development/meta-aws-platform-bsp/build
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp$ git rev-parse --show-toplevel
2021-12-11 13:04:37 - INFO     - Using /home/wolf/development/meta-aws-platform-bsp as root for repository meta-solvispi
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp$ git rev-parse --show-toplevel
2021-12-11 13:04:37 - INFO     - Using /home/wolf/development/meta-aws-platform-bsp as root for repository meta-solvispi
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky$ git status
[... cut ...]
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-raspberrypi$ git status
On branch honister
nothing to commit, working tree clean
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp/poky/meta-aws$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
2021-12-11 13:04:37 - INFO     - /home/wolf/development/meta-aws-platform-bsp$ git status
On branch testing-meta-aws
Your branch is up to date with 'origin/testing-meta-aws'.
```

RephaelCongmon, 2021-12-15:
Hi @WolfWalter , writing this to let you know I'm still working on reproducing this. Just to clarify, are you building the image on the Pi itself using kas and then running the bash scripts to prepare the SD card on the Pi itself as well?

WolfWalter, 2021-12-15:
Hi @RephaelCongmon ,
no, I am building and preparing the SD card on a Ubuntu 20.04 Intel machine.

rpcme, 2022-01-25:
@WolfWalter were you able to overcome this issue? We were not able to reproduce it, I am not sure where there is a disconnect. It could be related to another recently fixed item?

dbouras, 2023-05-29:
Hi @rpcme,  
I'm having what seems to be the same issue. I'm also building using [Siemens kas](https://github.com/siemens/kas), since it provides a handy container image that can be easily deployed on a development host or a build machine, without having to worry about Linux distribution specifics.  
What I see is the client crashing on various instances, complaining about null strings. One example is not being able to start it via configuration file; I always get:

```
Fatal error condition occurred in /usr/src/debug/aws-c-common/0.8.20-r0/git/source/string.c:217: allocator && aws_byte_cursor_is_valid(cursor)
Exiting Application
################################################################################
Stack trace:
################################################################################
/usr/lib/libaws-c-common.so.1(aws_backtrace_print+0x62) [0x7fced5dee4c2]
/usr/lib/libaws-c-common.so.1(aws_fatal_assert+0x48) [0x7fced5dd3228]
/usr/lib/libaws-c-common.so.1(+0x35d89) [0x7fced5df4d89]
/usr/lib/libaws-c-common.so.1(aws_json_value_get_from_object+0x29) [0x7fced5de5fc9]
/usr/lib/libaws-crt-cpp.so(_ZNK3Aws3Crt8JsonView11ValueExistsEPKc+0x1d) [0x7fced5f5539d]
aws-iot-device-client(+0x385dc) [0x55fefb4a95dc]
aws-iot-device-client(+0x3a233) [0x55fefb4ab233]
aws-iot-device-client(+0x3d64d) [0x55fefb4ae64d]
aws-iot-device-client(+0x1e121) [0x55fefb48f121]
/lib/libc.so.6(+0x2d57b) [0x7fced594057b]
/lib/libc.so.6(__libc_start_main+0x80) [0x7fced5940630]
aws-iot-device-client(+0x1ff45) [0x55fefb490f45]
Aborted
```

Also when testing secure tunnelling, if the tunnel is open at the Cloud end, I see:

```
2023-05-29T23:02:57.744Z [INFO]  {FileUtils.cpp}: Successfully create directory /home/root/.aws-iot-device-client/sample-shadow/ with required permissions 700
2023-05-29T23:02:57.744Z [INFO]  {Config.cpp}: ~/.aws-iot-device-client/sample-shadow/default-sample-shadow-document
2023-05-29T23:02:57.744Z [INFO]  {Config.cpp}: Succesfully create default file: /home/root/.aws-iot-device-client/sample-shadow/default-sample-shadow-document required for storage of shadow document
2023-05-29T23:02:57.744Z [DEBUG] {Config.cpp}: Did not find a runtime configuration file, assuming Fleet Provisioning has not run for this device
2023-05-29T23:02:57.744Z [DEBUG] {Config.cpp}: Did not find a http proxy config file /home/root/.aws-iot-device-client/http-proxy.conf, assuming HTTP proxy is disabled on this device
2023-05-29T23:02:57.744Z [DEBUG] {EnvUtils.cpp}: Updated PATH environment variable to: /usr/local/bin:/usr/bin:/bin:/usr/local/sbin:/usr/sbin:/sbin:/home/root/.aws-iot-device-client:/home/root/.aws-iot-device-client/jobs:/edge/aws/client:/edge/aws/client/jobs
2023-05-29T23:02:57.744Z [DEBUG] {LockFile.cpp}: creating lockfile
2023-05-29T23:02:57.744Z [INFO]  {Main.cpp}: Now running AWS IoT Device Client version v1.8.0-bb7ff67
2023-05-29T23:02:57.748Z [INFO]  {SharedCrtResourceManager.cpp}: SDK logging is enabled. Check /edge/aws/mqtt/logs/sdk.log for SDK logs.
2023-05-29T23:02:57.749Z [DEBUG] {Retry.cpp}: Retryable function starting, it will retry until success
2023-05-29T23:02:57.753Z [INFO]  {SharedCrtResourceManager.cpp}: Establishing MQTT connection with client id XXXXXXXX...
2023-05-29T23:02:58.229Z [INFO]  {SharedCrtResourceManager.cpp}: MQTT connection established with return code: 0
2023-05-29T23:02:58.229Z [INFO]  {SharedCrtResourceManager.cpp}: Shared MQTT connection is ready!
2023-05-29T23:02:58.230Z [INFO]  {Main.cpp}: Provisioning with Secure Elements is disabled
2023-05-29T23:02:58.230Z [INFO]  {Main.cpp}: Config shadow is disabled
2023-05-29T23:02:58.230Z [INFO]  {Main.cpp}: Jobs is enabled
2023-05-29T23:02:58.230Z [INFO]  {Main.cpp}: Secure Tunneling is enabled
2023-05-29T23:02:58.230Z [INFO]  {Main.cpp}: Device Defender is disabled
2023-05-29T23:02:58.230Z [INFO]  {Main.cpp}: Sample shadow is disabled
2023-05-29T23:02:58.230Z [INFO]  {Main.cpp}: Pub Sub is disabled
2023-05-29T23:02:58.230Z [INFO]  {Main.cpp}: Sensor Publish is disabled
2023-05-29T23:02:58.230Z [INFO]  {SharedCrtResourceManager.cpp}: Starting Device Client features.
2023-05-29T23:02:58.230Z [DEBUG] {FeatureRegistry.cpp}: Attempting to start Jobs
2023-05-29T23:02:58.230Z [INFO]  {Main.cpp}: Client base has been notified that Jobs has started
2023-05-29T23:02:58.230Z [DEBUG] {FeatureRegistry.cpp}: Attempting to start Secure Tunneling
2023-05-29T23:02:58.230Z [INFO]  {SecureTunnelingFeature.cpp}: Running Secure Tunneling!
2023-05-29T23:02:58.230Z [INFO]  {Main.cpp}: Client base has been notified that Secure Tunneling has started
2023-05-29T23:02:58.230Z [INFO]  {JobsFeature.cpp}: Running Jobs!
2023-05-29T23:02:58.230Z [DEBUG] {JobsFeature.cpp}: Attempting to subscribe to startNextPendingJobExecution accepted and rejected
2023-05-29T23:02:58.231Z [DEBUG] {JobsFeature.cpp}: Attempting to subscribe to nextJobChanged events
2023-05-29T23:02:58.231Z [DEBUG] {JobsFeature.cpp}: Attempting to subscribe to updateJobExecutionStatusAccepted for jobId +
2023-05-29T23:02:58.288Z [DEBUG] {SecureTunnelingFeature.cpp}: Received MQTT Tunnel Notification
terminate called after throwing an instance of 'std::logic_error2023-05-29T23:02:58.288Z [DEBUG] {SecureTunnelingFeature.cpp}: Region=us-east-1, Service=SSH'
  what():  basic_string::_M_construct null not valid
Aborted
```

complaining about  `basic_string::_M_construct null not valid` , again a null string issue.

Any ideas? Might this be a compiler version issue? In the kas container:

```
[... cut ...] fload-targets=nvptx-none=/build/gcc-10-Km9U7s/gcc-10-10.2.1/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-10-Km9U7s/gcc-10-10.2.1/debian/tmp-gcn/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-mutex
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.1 20210110 (Debian 10.2.1-6) 
```

Happy to help debug this - please let me know

dbouras, 2023-05-30:
Thank you for the follow-up @thomas-roos, really appreciated!  
Since I'm a bit in a hurry, I'm thinking of playing around with the compiler settings; I'm almost certain something has changed in g++ v10.x that triggers the bug. Unless you suggest otherwise, I'll move further comments to awslabs/aws-iot-device-client#411.

thomas-roos, 2023-05-30:
ok, you can also try to go back in time to beginning of March. This should work.

dbouras, 2023-05-30:
Tried 912bb30 (from March 1st) but same issue...

thomas-roos, 2023-05-31:
ok, strange - at least the ptests will run sucessfully with exact this commit: 
```
BB_VERSION           = "2.0.0"                                                                                                                                                                                                                                                            
BUILD_SYS            = "x86_64-linux"                                                                                                                                                                                                                                                     
NATIVELSBSTRING      = "universal"                                                                                                                                                                                                                                                        
TARGET_SYS           = "x86_64-poky-linux"                                                                                                                                                                                                                                                
MACHINE              = "qemux86-64"                                                                                                                                                                                                                                                       
DISTRO               = "poky"                                                                                                                                                                                                                                                             
DISTRO_VERSION       = "4.0.10"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "kirkstone:2afcf3e39f9016f590e597cee4b71166c982d58d"
meta-oe              
meta-python          
meta-networking      
meta-multimedia      = "kirkstone:5f120a926b0fcd55cfe7565bb7ddf23661cad498"
meta-aws             = "HEAD:912bb30955c7b28e70d3440dc6b24d8794aa7042"
```

thomas-roos, 2023-06-05:
did create a new issue for this - reopening this old one is confusing: https://github.com/aws4embeddedlinux/meta-aws/issues/4063

closed 2023-06-05 by thomas-roos