aws4embeddedlinux/meta-aws #134: High CPU utilization of Device Client

dhwalters423, 2021-06-18:
There has been a critical fix in the Device Client that should be in the next minor release of Device Client (1.3). Please update to the newest Device Client as soon as its feasible. Details in this ticket: https://github.com/awslabs/aws-iot-device-client/issues/147

rpcme, 2021-07-17:
@dhwalters423 I see this is fixed but the team hasn't yet made another release.  I will need to wait for a tagged, semantically versioned contiguous set is available before taking the fix.

rpcme, 2021-09-01:
@dhwalters423 we are in plan to put in the latest device client but still facing aws-lc not being tagged.  Hopefully this will be resolved over the next 2 weeks.

rpcme, 2021-09-09:
@dhwalters423 New version of Device Client (1.3.4) is staged on master-next and will likely be pushed to master in the morning.  all currently maintained release branches will fast follow.

closed 2021-09-09 by rpcme