aws4embeddedlinux/meta-aws #157: Failed to build aws-iot-securetunneling-localproxy on zeus branch

borisstein, 2021-08-12:
Describe the bug

I had issue with compiling "aws-iot-securetunneling-localproxy" in yocto

To Reproduce

added:
IMAGE_INSTALL_append += "
...........
aws-iot-device-client
aws-iot-securetunneling-localproxy"
+
BBLAYERS += "${BSPDIR}/sources/meta-aws"

Environment
Ubuntu 20.04

Yocto:
Build Configuration:
BB_VERSION = "1.44.0"
BUILD_SYS = "x86_64-linux"
NATIVELSBSTRING = "ubuntu-20.04"
TARGET_SYS = "aarch64-poky-linux"
MACHINE = "iot-gate-imx8"
DISTRO = "fsl-imx-xwayland"
DISTRO_VERSION = "5.4-zeus"
TUNE_FEATURES = "aarch64"

GIT vesion of meta-aws
meta-aws git version:
boris@boris-ThinkPad-P14s-Gen-1:~/yocto/compulab_test_manifest/sources/meta-aws$ git log
commit 4f7f7c08691c6947778f7fece2a3c7a0c59e058f (HEAD, origin/zeus)

Logs
Multiple error while compiling "aws-iot-securetunneling-localproxy" package:

DEBUG: Executing shell function do_compile
NOTE: VERBOSE=1 cmake --build /home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/build --target all -- -j 16
[1/12] cd /home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/build && /home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/recipe-sysroot-native/usr/bin/protoc --cpp_out /home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/build -I /home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/git/resources /home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/git/resources/Message.proto
[2/12] /home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++ -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_LOG_DYN_LINK -DBOOST_LOG_SETUP_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -I. -I/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/git/src -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/recipe-sysroot -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0=/usr/src/debug/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0 -fdebug-prefix-map=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0=/usr/src/debug/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0 -fdebug-prefix-map=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/recipe-sysroot= -fdebug-prefix-map=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/recipe-sysroot-native= -fvisibility-inlines-hidden -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/recipe-sysroot -O2 -D_FORTIFY_SOURCE=2 -fPIE -Wall -Werror -D_AWSIOT_TUNNELING_NO_SSL -std=c++14 -MD -MT CMakeFiles/localproxytest.dir/Message.pb.cc.o -MF CMakeFiles/localproxytest.dir/Message.pb.cc.o.d -o CMakeFiles/localproxytest.dir/Message.pb.cc.o -c Message.pb.cc
[3/12] /home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++ -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_LOG_DYN_LINK -DBOOST_LOG_SETUP_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -I. -I/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/git/src -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/recipe-sysroot -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0=/usr/src/debug/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0 -fdebug-prefix-map=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0=/usr/src/debug/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0 -fdebug-prefix-map=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/recipe-sysroot= -fdebug-prefix-map=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/recipe-sysroot-native= -fvisibility-inlines-hidden -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/recipe-sysroot -O2 -D_FORTIFY_SOURCE=2 -fPIE -Wall -Werror -std=c++14 -MD -MT CMakeFiles/localproxy.dir/Message.pb.cc.o -MF CMakeFiles/localproxy.dir/Message.pb.cc.o.d -o CMakeFiles/localproxy.dir/Message.pb.cc.o -c Message.pb.cc
[4/12] /home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++ -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_LOG_DYN_LINK -DBOOST_LOG_SETUP_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -I. -I/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/git/src -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/recipe-sysroot -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0=/usr/src/debug/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0 -fdebug-prefix-map=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0=/usr/src/debug/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0 -fdebug-prefix-map=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/recipe-sysroot= -fdebug-prefix-map=/home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/recipe-sysroot-native= -fvisibility-inline [... cut ...] E=2 -fPIE -Wall -Werror -D_AWSIOT_TUNNELING_NO_SSL -std=c++14 -MD -MT CMakeFiles/localproxytest.dir/test/AdapterTests.cpp.o -MF CMakeFiles/localproxytest.dir/test/AdapterTests.cpp.o.d -o CMakeFiles/localproxytest.dir/test/AdapterTests.cpp.o -c /home/boris/yocto/compulab_test_manifest/build-cmdline/tmp/work/aarch64-poky-linux/aws-iot-securetunneling-localproxy/1.0+gitAUTOINC+880b7dad54-r0/git/test/AdapterTests.cpp
ninja: build stopped: subcommand failed.
WARNING: exit code 1 from a shell command.

borisstein, 2021-08-12:
****

borisstein, 2021-08-12:
My question is: do i need "aws-iot-securetunneling-localproxy" executable on my imx.8 linux ebedded device to be able to make secure ssh to it?

rpcme, 2021-08-12:
@borisstein your intent is to use this IoT Core feature, correct? https://docs.aws.amazon.com/iot/latest/developerguide/secure-tunneling.html

In the meantime I will look at the issue.

borisstein, 2021-08-12:
> @borisstein your intent is to use this IoT Core feature, correct? https://docs.aws.amazon.com/iot/latest/developerguide/secure-tunneling.html
Correct, use aws-iot-device-client to connect to AWS service and then use secure tunneling feature, but perhaps i do not need  aws-iot-securetunneling-localproxy package at all and there is secure tunninling handling inside device-client

rpcme, 2021-08-12:
I see that I have not yet merged zeus-next to zeus.  This recipe had been broken on zeus for quite some time.  I am doing the build check once more of what's on zeus-next and then push the changes.

rpcme, 2021-08-12:
You can use either tool. IIRC both are reference implementations which usually infers customers would want/need to fork the code and modify to their needs but it should be sufficient for prototyping (or might even work for your production use case).

rpcme, 2021-08-12:
@borisstein you should be all set now.  I know how frustrating it can be when you go to build a recipe and it just pops so thank you for your patience here ok?

borisstein, 2021-08-12:
> @borisstein you should be all set now. I know how frustrating it can be when you go to build a recipe and it just pops so thank you for your patience here ok?

Many thanks!!

rpcme, 2021-08-12:
Ok going to close this now. Please let us know if you face any more challenges ok?

closed 2021-08-12 by rpcme