aws4embeddedlinux/meta-aws #160: Component upgrades?

denix0, 2021-08-31:
Are there any plans for component upgrades at least in master branch?

As of right now, I see component recipes in master being the same version as in dunfell, e.g. AWS IoT Device SDK for C++ is 1.10.5 or AWS IoT Device SDK for Python is 1.5.4, etc.

Corresponding components are being updated all the time and their versions are much newer than corresponding recipes... So, wondering what's the plan for recipe updates?

rpcme, 2021-09-01:
@denix0 we are planning a new rev to support the latest version of Device Client which uses the C++ CLR and Device SDK.  There are significant changes in the CLR dependencies that we need to work through.   I am hoping we can get this done by the end of next week.

rpcme, 2021-09-01:
To be more precise, we now have these versions on deck:

AWS IoT Device Client 1.3.4
AWS IoT Device SDK for C++ 1.14.1
AWS IoT Device SDK for Python v1.7.0
aws-crt-cpp 0.15.0
aws-crt-python 0.12.1
aws-c-iot 0.0.7
aws-c-auth v0.6.1
aws-c-cal v0.5.11
aws-c-common v0.6.8
aws-c-compression v0.2.14
aws-c-event-stream v0.2.7
aws-c-http v0.6.5
aws-c-io v0.10.7
aws-c-mqtt v0.7.6
aws-c-s3 v0.1.23
aws-checksums v0.1.11
aws-lc d0a5455 (new component - waiting for release tag)
s2n v1.0.13

rpcme, 2021-09-09:
Hi @denix0  - the upgrades have been staged to master-next and should be on the mainline tomorrow.  Dunfell, Gatesgarth, and Hardknott will fast follow.

denix0, 2021-09-09:
Thanks for the heads-up, @rpcme!

rpcme, 2021-09-11:
Hello @denix0 I have completed the merge.  I also did proper package splitting so if you encounter any issue with that and SDK/eSDK please let us know.

closed 2021-09-11 by rpcme