aws4embeddedlinux/meta-aws #171: aws-crt-python_0.10.3 breakage in Dunfell

denix0, 2021-09-13:
| x86_64-poky-linux-gcc -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot -Wno-unused-result -Wsign-compare -DNDEBUG -g -O3 -Wall -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/python3/3.8.11-r0=/usr/src/debug/python3/3.8.11-r0 -fdebug-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/python3/3.8.11-r0=/usr/src/debug/python3/3.8.11-r0 -fdebug-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot= -fdebug-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot-native= -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/python3/3.8.11-r0=/usr/src/debug/python3/3.8.11-r0 -fdebug-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/python3/3.8.11-r0=/usr/src/debug/python3/3.8.11-r0 -fdebug-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot= -fdebug-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot-native= -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0=/usr/src/debug/aws-crt-python/0.10.3-r0 -fdebug-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0=/usr/src/debug/aws-crt-python/0.10.3-r0 -fdebug-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot= -fdebug-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot-native= -Wl,-Bsymbolic -fPIC -I/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot/usr/include/python3.8 -I/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0/git/build/deps/install/include -c source/http_connection.c -o /OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0/build/temp.linux-x86_64-3.8/source/http_connection.o -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0=/usr/src/debug/aws-crt-python/0.10.3-r0 -fdebug-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0=/usr/src/debug/aws-crt-python/0.10.3-r0 -fdebug-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot= -fdebug-prefix-map=/OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot-native= -Wl,-Bsymbolic -Wextra -Werror -Wno-strict-aliasing -std=gnu99
| source/http_connection.c: In function 'aws_py_http_client_connection_new':
| source/http_connection.c:218:35: error: storage size of 'proxy_options_storage' isn't known
|   218 |     struct aws_http_proxy_options proxy_options_storage;
|       |                                   ^~~~~~~~~~~~~~~~~~~~~
| source/http_connection.c:218:35: error: unused variable 'proxy_options_storage' [-Werror=unused-variable]
| cc1: all warnings being treated as errors
| error: command 'x86_64-poky-linux-gcc' failed with exit status 1
| WARNING: /OE/poky-dunfell/build/tmp/work/core2-64-poky-linux/aws-crt-python/0.10.3-r0/temp/run.do_compile.21854:1 exit 1 from 'exit 1'
| 
ERROR: Task (/OE/poky-dunfell/meta-aws/recipes-sdk/aws-crt-python/aws-crt-python_0.10.3.bb:do_compile) failed with exit code '1'

rpcme, 2021-09-13:
@denix0  I'm on it.

denix0, 2021-09-13:
Same issue on master:

| aarch64-poky-linux-gcc -mcpu=cortex-a57 -march=armv8-a+crc -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time --sysroot=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot -Wno-unused-result -Wsign-compare -DNDEBUG -g -O3 -Wall -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/python3/3.9.6-r0=/usr/src/debug/python3/3.9.6-r0 -fdebug-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/python3/3.9.6-r0=/usr/src/debug/python3/3.9.6-r0 -fdebug-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot= -fdebug-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot-native= -DHAVE_BROKEN_POSIX_SEMAPHORES -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/python3/3.9.6-r0=/usr/src/debug/python3/3.9.6-r0 -fdebug-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/python3/3.9.6-r0=/usr/src/debug/python3/3.9.6-r0 -fdebug-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot= -fdebug-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot-native= -DHAVE_BROKEN_POSIX_SEMAPHORES -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0=/usr/src/debug/aws-crt-python/0.10.3-r0 -fdebug-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0=/usr/src/debug/aws-crt-python/0.10.3-r0 -fdebug-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot= -fdebug-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot-native= -Wl,-Bsymbolic -fPIC -I/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot/usr/include/python3.9 -I/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/git/build/deps/install/include -c source/http_connection.c -o /OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/build/temp.linux-x86_64-3.9/source/http_connection.o -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0=/usr/src/debug/aws-crt-python/0.10.3-r0 -fdebug-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0=/usr/src/debug/aws-crt-python/0.10.3-r0 -fdebug-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot= -fdebug-prefix-map=/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot-native= -Wl,-Bsymbolic -Wextra -Werror -Wno-strict-aliasing -std=gnu99
| source/http_connection.c: In function 'aws_py_http_client_connection_new':
| source/http_connection.c:218:35: error: storage size of 'proxy_options_storage' isn't known
|   218 |     struct aws_http_proxy_options proxy_options_storage;
|       |                                   ^~~~~~~~~~~~~~~~~~~~~
| source/http_connection.c:218:35: error: unused variable 'proxy_options_storage' [-Werror=unused-variable]
| cc1: all warnings being treated as errors
| error: command '/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcc' failed with exit code 1
| ERROR: 'python3 setup.py build ' execution failed.
| WARNING: /OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/temp/run.do_compile.38920:181 exit 1 from 'exit 1'
| WARNING: Backtrace (BB generated script):
|       #1: bbfatal_log, /OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/temp/run.do_compile.38920, line 181
|       #2: distutils3_do_compile, /OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/temp/run.do_compile.38920, line 167
|       #3: do_compile, /OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/temp/run.do_compile.38920, line 156
|       #4: main, /OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/temp/run.do_compile.38920, line 181
| ERROR: Execution of '/OE/poky-master/build/tmp/work/cortexa57-poky-linux/aws-crt-python/0.10.3-r0/temp/run.do_compile.38920' failed with exit code 1
ERROR: Task (/OE/poky-master/meta-aws/recipes-sdk/aws-crt-python/aws-crt-python_0.10.3.bb:do_compile) failed with exit code '1'

rpcme, 2021-09-13:
@denix0 I'm upgrading the recipes right now.  I did only C++ on Friday.  I just finished CRT and working on the device sdk right now.

rpcme, 2021-09-13:
@denix0 please check Dunfell.

rpcme, 2021-09-13:
@denix0 I have updated the mainline as well.

We currently have engineers working on CI hopefully these types of issues will cease in the future.

denix0, 2021-09-13:
Thanks, the build now works, but there are other issues - I'll file separately.

closed 2021-09-13 by denix0