aws4embeddedlinux/meta-aws #178: cmake files are in the wrong package, depend on custom BUILD_SHARED_LIBS, which is not defined

denix0, 2021-09-13:
1. cmake files are being packaged into ${PN}-dbg, but instead should be part of ${PN}-dev

2. each AWS cmake file has a condition on BUILD_SHARED_LIBS, which seems to be AWS-specific and is not defined by default in 3rd-party apps, hence falls back to static path, even though only shared variant is provided by AWS components by default:

if (BUILD_SHARED_LIBS)
    include(${CMAKE_CURRENT_LIST_DIR}/shared/aws-c-s3-targets.cmake)
else()
    include(${CMAKE_CURRENT_LIST_DIR}/static/aws-c-s3-targets.cmake)
endif()

The workaround is to pass -DBUILD_SHARED_LIBS=On to CMake when building an app using AWS, but maybe this should NOT be based on a variable if only one variant (either shared or static) is provided anyway.

rpcme, 2021-09-13:
Hello @denix0 - I am going to put /2/ in a different issue so they can be tracked separately.

For /2/ I will bring in the CRT and SDK teams to evaluate.

rpcme, 2021-09-13:
I think I need to document that the default is ON and implement a variable to override, but I need to coordinate this flexibility across all dependent recipes.  Can you let me know the temperature of this issue?  I will need to shuffle some things around if this is immediately breaking

denix0, 2021-09-13:
@rpcme sounds good, thanks!

Maybe it would make sense to build and package both static and shared libs and then use a more AWS-specific variable (instead of a more generically-named BUILD_SHARED_LIBS) to select which one an app should use and link against...

If only one variant is provided, then you might want to replace BUILD_SHARED_LIBS in those cmake files with its value at the time AWS components are built.

rpcme, 2021-09-14:
Hi @denix0 our SDK team pointed me to this cmake documentation.

https://cmake.org/cmake/help/git-stage/variable/BUILD_SHARED_LIBS.html

I am trying to figure out how we can clarify this to everyone.  I did note that it isn't called out in our documentation as a flag that needs to be used when choosing shared or static.  Do you think this would have helped?  Or is there something else that would have helped?

rpcme, 2021-09-20:
Hello @denix0 please let me know if you have any further thoughts / questions on this ok?

rpcme, 2021-09-21:
Hi @denix0 will be closing this for now.  Please re-open if this did not explain the SDK team's intent by using BUILD_SHARED_LIBS.  Would still like to get feedback on what you think would have made the realization more natural.

closed 2021-09-21 by rpcme

denix0, 2021-09-21:
Hi @rpcme sorry for the delay - been really busy lately.

BUILD_SHARED_LIBS variable makes sense as a global variable for a project being built. So when you build one of the AWS components, it controls the output. The resulting AWS binary package at the end only has one variant of the library and since BUILD_SHARED_LIBS is set to On in AWS recipes, only shared library gets packaged.

So, first issue is that CMake files that those prebuilt AWS packages provide still contain a condition on BUILD_SHARED_LIBS - it should no longer be conditional, as the package has already been built and only contains a shared library.

The second issue is that those AWS libraries are meant to be consumed by another project (a user app), which may not define BUILD_SHARED_LIBS or set it to Off in order to build own libraries statically. That flag no longer controls how AWS libraries were built (as those are now being consumed in binary form), but instead it controls how user app project's own libraries are being built.

The workaround I used for now is to force BUILD_SHARED_LIBS=On in the user app, which is incorrect.

One of the solutions I see is to do a quick "sed" command in recipe's do_install() on corresponding *-config.cmake file to replace "if (BUILD_SHARED_LIBS)" with the corresponding value at the time of the build, which would be "if (On)".

rpcme, 2021-09-21:
@denix0 I am going to file this to the SDK team internally and come to a resolution on this.

denix0, 2021-09-21:
BTW, another option would be to build and package both dynamic and static versions of each AWS SDK library. Then the condition on BUILD_SHARED_LIBS in the cmake files would make sense and it would be up to consuming user app to set it and select one or another.