aws4embeddedlinux/meta-aws #180: aws-c-s3_0.1.23 fails with ParseError

gsalomao, 2021-09-14:
The build using the latest commit on `hardknott` branch fails with `ParseError` due to `aws-c-s3_0.1.23`.

```bash
ERROR: ParseError at /src/meta-aws/recipes-sdk/aws-c-s3/aws-c-s3_0.1.23.bb:26: unparsed line: 'CFLAGS:append = " -Wl,-Bsymbolic"'
```

rpcme, 2021-09-14:
Hello @gsalomao  - I am unable to reproduce the error. I also ran `-c cleansstate` on the target just to make sure.  This is the first override in the file though.  Note that (this is my understanding) that the override syntax is available on hardknott but only after the change was made to the mainline (honister release).  Is it possible that your yocto tree isn't up to date?  Or are you stuck on a specific commit hash?

gsalomao, 2021-09-14:
@rpcme I have all layers on hardknott and I saw that the build fails everytime in a different recipe:
```bash
ERROR: ParseError at /src/meta-aws/recipes-sdk/aws-c-io/aws-c-io_0.10.7.bb:28: unparsed line: 'CFLAGS:append = " -Wl,-Bsymbolic"'
...
ERROR: ParseError at /src/meta-aws/recipes-sdk/aws-c-compression/aws-c-compression_0.2.14.bb:24: unparsed line: 'RDEPENDS:${PN} = "s2n aws-c-common aws-c-cal aws-c-io"'
```

Running the `git bisect` I found the following commit that breaks my build:
```bash
$ git bisect good
fa98622c6af551d0eb30ef9b0a0f360ecbf840f0 is the first bad commit
commit fa98622c6af551d0eb30ef9b0a0f360ecbf840f0
Author: rpcme <rich@richelberger.com>
Date:   Sat Sep 11 13:11:35 2021 -0400

    update hardknott with upgrade to CRT C++, Device SDK C++ v2, IoT Device Client
```

rpcme, 2021-09-14:
Can you please give me your build configuration, i.e. What is output when you start the build, such as:

```text
YP:hardknott-next > bitbake aws-iot-device-sdk-cpp-v2
Loading cache: 100% |################################################################################################| Time: 0:00:00
Loaded 3708 entries from dependency cache.
Parsing recipes: 100% |##############################################################################################| Time: 0:00:00
Parsing of 2348 .bb files complete (2332 cached, 16 parsed). 3709 targets, 85 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.50.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "qemux86-64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.3.3"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "hardknott:0a2ca9d60f3851515a79d5aa9ddd8b4069b5a206"
meta-oe              
meta-python          
meta-networking      = "hardknott:7bd7e1da9034e72ca4262dba55f70b2b23499aae"
meta-aws             = "hardknott-next:5dae5e6f9b82018f13c0fb2df151e6659dab3cdd"

Initialising tasks: 100% |###########################################################################################| Time: 0:00:01
Sstate summary: Wanted 46 Local 0 Network 0 Missed 46 Current 604 (0% match, 92% complete)
Removing 31 stale sstate objects for arch core2-64: 100% |###########################################################| Time: 0:00:00
Removing 15 stale sstate objects for arch qemux86_64: 100% |#########################################################| Time: 0:00:00
NOTE: Executing Tasks
NOTE: Tasks Summary: Attempted 2050 tasks of which 2003 didn't need to be rerun and all succeeded.
YP:hardknott-next > git add .
YP:hardknott-next > git commit -m"move cmake files from -dbg to -dev, fixes #179"
[hardknott-next 0292c7f] move cmake files from -dbg to -dev, fixes #179
 14 files changed, 23 insertions(+), 25 deletions(-)
YP:hardknott-next > git push
Counting objects: 31, done.
Delta compression using up to 16 threads.
Compressing objects: 100% (30/30), done.
Writing objects: 100% (31/31), 2.53 KiB | 517.00 KiB/s, done.
Total 31 (delta 23), reused 0 (delta 0)
remote: Resolving deltas: 100% (23/23), completed with 23 local objects.
To github.com:aws/meta-aws.git
   5dae5e6..0292c7f  hardknott-next -> hardknott-next
YP:hardknott-next > cd /worktree/poky-hardknott/
YP:hardknott-next > which bitbake
/worktree/poky-hardknott/bitbake/bin/bitbake
YP:hardknott-next > git branch
* hardknott
YP:hardknott-next > 

```

rpcme, 2021-09-20:
Hi @gsalomao looking forward to future communication on this issue so we can get it resolved to your satisfaction.

gsalomao, 2021-09-20:
Hi @rpcme, sorry for the long delay.

The output of the command:
```
user@3801cfef6a79:/opt/yocto/build$ bitbake aws-iot-device-sdk-cpp-v2
Loading cache: 100% |################################################################################################################################################################################| Time: 0:00:00
Loaded 176 entries from dependency cache.
ERROR: ParseError at /src/meta-aws/recipes-sdk/aws-c-http/aws-c-http_0.6.5.bb:24: unparsed line: 'RDEPENDS:${PN} = "s2n aws-c-common aws-c-cal aws-c-io aws-c-compression"'                          | ETA:  0:00:31

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
user@3801cfef6a79:/opt/yocto/build$ 
user@3801cfef6a79:/opt/yocto/build$ bitbake aws-iot-device-sdk-cpp-v2
Loading cache: 100% |################################################################################################################################################################################| Time: 0:00:00
Loaded 493 entries from dependency cache.
ERROR: ParseError at /src/meta-aws/recipes-sdk/aws-c-io/aws-c-io_0.10.7.bb:24: unparsed line: 'RDEPENDS:${PN} = "s2n aws-c-common aws-c-cal"'                                                       | ETA:  --:--:--

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
user@3801cfef6a79:/opt/yocto/build$ 
user@3801cfef6a79:/opt/yocto/build$ 
user@3801cfef6a79:/opt/yocto/build$ bitbake aws-iot-device-sdk-cpp-v2
Loading cache: 100% |################################################################################################################################################################################| Time: 0:00:00
Loaded 501 entries from dependency cache.
ERROR: ParseError at /src/meta-aws/recipes-sdk/aws-iot-device-sdk-cpp-v2/aws-iot-device-sdk-cpp-v2_1.14.1.bb:23: unparsed line: 'RDEPENDS:${PN} = "aws-crt-cpp aws-c-iot"'                          | ETA:  --:--:--

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
user@3801cfef6a79:/opt/yocto/build$ 
user@3801cfef6a79:/opt/yocto/build$ bitbake aws-iot-device-sdk-cpp-v2
Loading cache: 100% |################################################################################################################################################################################| Time: 0:00:00
Loaded 520 entries from dependency cache.
ERROR: ParseError at /src/meta-aws/recipes-sdk/aws-c-common/aws-c-common_0.6.8.bb:19: unparsed line: 'CFLAGS:append = " -Wl,-Bsymbolic"'                                                            | ETA:  --:--:--

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
user@3801cfef6a79:/opt/yocto/build$
user@5be11e31f5dd:/opt/yocto/build$ which bitbake
/src/poky/bitbake/bin/bitbake
user@5be11e31f5dd:/opt/yocto/build$ exit
```

Note that running it multiple times gives different errors.

```
$ git log --oneline --decorate
cad1c71 (HEAD -> hardknott, origin/hardknott) Merge pull request #184 from aws/hardknott-next
0292c7f (origin/hardknott-next) move cmake files from -dbg to -dev, fixes #179
294a598 Merge pull request #176 from aws/hardknott-next
5dae5e6 upgrade support for aws iot device sdk python v2
...
```

It seems that the problem is with the poky 3.3.2 because once updated to 3.3.3, the build finished without error.
As the error seems not to be in the meta-aws, I think we can close this ticket.

rpcme, 2021-09-21:
@gsalomao YP added a new change to the override syntax and the recipes use the override syntax.  My first response noted this.  Thank you for the confirmation though. Will close the ticket.

closed 2021-09-21 by rpcme