aws4embeddedlinux/meta-aws #186: Error in Greengrass logs - Unable to set up Nucleus from build recipe file.

givenand, 2021-09-20:
Hi All, 

We are seeing this issue:

 2021-09-01T10:25:51.553Z [ERROR] (main) com.aws.greengrass.deployment.DeviceConfiguration: Unable to set up Nucleus from build recipe file. {}
    java.nio.file.DirectoryNotEmptyException: /data/greengrass/v2/alts/init/distro
        at java.base/sun.nio.fs.UnixFileSystemProvider.implDelete(UnixFileSystemProvider.java:247)
        at java.base/sun.nio.fs.AbstractFileSystemProvider.deleteIfExists(AbstractFileSystemProvider.java:110)
        at java.base/java.nio.file.Files.deleteIfExists(Files.java:1180)
        at com.aws.greengrass.lifecyclemanager.KernelAlternatives.relinkInitLaunchDir(KernelAlternatives.java:197)
        at com.aws.greengrass.deployment.DeviceConfiguration.initializeComponentStore(DeviceConfiguration.java:352)
        at com.aws.greengrass.deployment.DeviceConfiguration.initializeNucleusFromRecipe(DeviceConfiguration.java:383)
        at com.aws.greengrass.lifecyclemanager.Kernel.parseArgs(Kernel.java:645)
        at com.aws.greengrass.easysetup.GreengrassSetup.performSetup(GreengrassSetup.java:296)
        at com.aws.greengrass.easysetup.GreengrassSetup.main(GreengrassSetup.java:269)

It seems like GG expects that the directory /data/greengrass/v2/alts/init/distro is empty, but it is not, as this is where we initially run GG from.  Can we modify the recipe to not error on this distro being empty or is there another place where we can run GG from?

K-AWS, 2021-09-23:
Hi Andrew,

Thanks for reporting the issue. We're trying to replicate and investigate the issue. Are you using Gatesgarth? Which version of GG are you using?

Thanks,

rpcme, 2021-11-05:
Hello @givenand can you please take a moment to get back to @K-AWS ?  Thank you.

hancin, 2021-11-29:
The greengrass Yocto recipe expects you to configure greengrass manually (more information can be obtained [in the manual installation documentation](https://docs.aws.amazon.com/greengrass/v2/developerguide/manual-installation.html) instead of using easy setup. Easy Setup will show this error because the folder structure it creates is already present but not the configuration.

The greengrass-bin-demo image shows how to add the configuration at build time so that you do not get this error. You can also add some logic on first device boot to do the manual configuration before Greengrass runs, so that greengrass is configured at run time.

rpcme, 2022-01-15:
Closing this issue based on lack of activity.

ahmedmuzammilAI, 2023-07-18:
I am trying to set a WSL device as a Greengrass core device for testing.. maybe this way you can replicate the error. please let me know if there is a solution or a work around. Thank you.

thomas-roos, 2023-07-18:
@nenadilic84 might you have an answer?

nateglims, 2023-07-18:
To get better attention from the GreenGrass support apparatus, I recommend asking on [re:post](https://repost.aws/) or the relevant repo in https://github.com/aws-greengrass 

This repository just deals with the Yocto delivery channel

Dizzzmas, 2024-03-22:
I've bumped into the very same error today.

I'm following a tutorial https://dev.to/iotbuilders/fleet-provisioning-for-embedded-linux-devices-with-aws-iot-greengrass-4h8b on building a yocto image with greengrass fleet provisioning baked in.

The only difference - I'm trying to build it for a generic x86-64 target (currently qemux86-64 for testing), not a raspberry pi. Once I boot and check the logs - that's the error I see.