aws4embeddedlinux/meta-aws #234: The Python IoT v2 SDK require a function that is deprecated and has been removed from .so file

gustavqmatic, 2021-11-03:
Current rev on master branch of meta-aws has these versions
aws-crt-python 0.12.1
aws-c-mqtt 0.7.6

aws-crt-python is set to use .so file versions of aws-c-mqtt (with patch fix-library-suffix.patch)

aws-crt-python in mqtt_client_connection.c calls function aws_mqtt_client_connection_reconnect (from aws-c-mqtt) but this function is deprecated and not exported to the .so file (see client.h in  aws-c-mqtt where it does not have the AWS_MQTT_API flag).

How should this work?

rpcme, 2021-11-04:
Hello @gustavqmatic let us review what happened here and will report back or make fix.

graebm, 2021-11-04:
tldr; it's weird, but fine

The python function `awscrt.mqtt.Connection.reconnect()` is also marked [DEPRECATED](https://github.com/awslabs/aws-crt-python/blob/aa4511c1cea931c30051b2fe43be76f86b6a2ba2/awscrt/mqtt.py#L400). Users should not call it, but we exposed it early on, and so the function remains in the codebase for now.

The missing AWS_MQTT_API flag in aws-c-mqtt is weird, but it shouldn't cause any problems. The missing flag was probably just an accident that we've never noticed, and since it's deprecated I don't see a strong reason to fix it. The missing AWS_MQTT_API flag only controls symbol visibility when aws-c-mqtt is turned into a standalone .so file. 
This doesn't affect aws-crt-python which builds aws-c-mqtt (and many others) as static libraries, and links them all together into one shared library: `_awscrt.so`. All symbols in static libs are visible, so the missing AWS_MQTT_API flag doesn't matter here.

Here's how it goes, for those interested:

`awscrt.mqtt.py:Connection.reconnect()` <-- python code
CALLS publicly visible function in C shared lib:
`aws_py_mqtt_client_connection_reconnect()` <-- from _awscrt.so
CALLS function that it linked statically:
`aws_mqtt_connection_reconnect()` <-- from aws-c-mqtt.a

graebm, 2021-11-04:
Is this causing an issue?

RephaelCongmon, 2021-11-04:
The missing macro is added to [client.h](https://github.com/awslabs/aws-c-mqtt/blob/main/include/aws/mqtt/client.h).

rpcme, 2021-11-05:
@gustavqmatic to be clear, we are identifying the most friction free fix implementation for you.  Likely, we will patch the current version until we are prepared to move the version of the entire CRT because we must keep the versions contiguous.  @graebm thank you so much for your feedback - always appreciated!

gustavqmatic, 2021-11-05:
We are happy with the above fix, thank you for the quick response.

Just a final question, what is the reason behind patch https://github.com/aws/meta-aws/blob/master/recipes-sdk/aws-crt-python/files/fix-library-suffix.patch? My understanding is that this change it from static linking to shared and seems to cause this problem when using meta-aws but not when everything is built into one _awscrt.so file as described above.

WolfWalter, 2021-11-21:
Also ran into this yesterday when adding `aws-iot-device-sdk-python-v2` to my image and trying to run the [sample](https://github.com/aws/aws-iot-device-sdk-python-v2/blob/main/samples/pubsub.py). The error was `undefined symbol: aws_mqtt_client_connection_reconnect`. But also a pyhton script with only `from awsiot import mqtt_connection_builder` lead to the same error.

My image is for a raspi 4 an im am currently testing on `hardknott`. I was able to fix the issue by adding `aws-c-mqtt_0.7.9.bb` to my layer with the same content as `aws-c-mqtt_0.7.6.bb`.

rpcme, 2022-01-20:
This issue seems resolved, please reopen if not.

closed 2022-01-20 by rpcme