aws4embeddedlinux/meta-aws #242: Request to publish your CodeBuild configuration

manuel-wagesreither, 2021-11-04:
Hi all,

I see you are building and testing this layer on AWS CodeBuild. As someone currently trying to set up cloud CI for his own Yocto project, it would be interested in seeing how you did it. Specifically if and how you are reusing your sstate cache and download directories. When the build is finished, do you push them to a S3 bucket which is made accessible via HTTP and set as `SSTATE_MIRRORS`/`MIRRORS`/`PREMIRRORS` inYocto? That's the idea I had, but perhaps you came up with something better. You are directly at the core of AWS knowledge, after all. :-)

Not sure if the "Issues" section is the right place for this, didn't see any other support channel.

Thanks!
Manuel

rpcme, 2021-11-05:
@manuel-wagesreither I am happy to answer here until we start using Github discussions or another venue.

Currently, all our CodeBuild configurations are in the meta-aws-ci repository here.  https://github.com/aws/meta-aws-ci

There is also workshop content there to help you get started.  The more interesting section for you is in Module 2 since Module 1 is building manually through Cloud9.

rpcme, 2021-11-05:
Please feel free to continue to ask questions here and either me or team members will help you.

rpcme, 2021-11-05:
Also, regarding finished product propagation - yes, in the class I gave at Embedded World earlier this year I showed this.  The pathway does differ between distros.  However, you will see how simple it is to do with CodeBuild.

manuel-wagesreither, 2021-11-05:
Thank you a lot! I will consume the material you mentioned and will give notice in case I need some advice. Much appreciated.

Shall I close this issue or leave it open? Would you prefer questions here or shall I raise them with individual tickets?

rpcme, 2021-11-06:
Hi @manuel-wagesreither please feel free to open another ticket as challenges arise. If it relates to content in meta-aws-ci feel free to open tickets there.  Hopefully in the very near future we will have a discussion board.

closed 2021-11-06 by rpcme