aws4embeddedlinux/meta-aws #261: No files of aws-c-common are packaged for native SDK

GonzalezSGA, 2021-11-15:
Working branch: Dunfell.
I want to include aws-crt-cpp in my image, and also have the libraries in the native SDK in order to develop and run test in the host.
I do generate correctly the image with aws-crt-cpp adding:
`IMAGE_INSTALL_append = " aws-crt-cpp"` to my image recipe.
Then I added:
`TOOLCHAIN_HOST_TASK_append = " aws-crt-cpp"` to my image recipe and executed:
`bitbake <image_name> -c populate_sdk`
And the build crashes because all the recipes depending on aws-c-common can't find the cmake configuration files.
If I isolate the problem and execute:
`bitbake nativesdk-aws-c-common` I get the following error:
`ERROR: nativesdk-aws-c-common-0.6.8-r0 do_package: QA Issue: nativesdk-aws-c-common: Files/directories were installed but not shipped in any package:
  /usr/include
  /usr/lib/libaws-c-common.so
  /usr/lib/libaws-c-common.so.1
  /usr/lib/libaws-c-common.so.1.0.0
  /usr/lib/aws-c-common
  /usr/lib/cmake
  /usr/lib/aws-c-common/cmake
  /usr/lib/aws-c-common/cmake/aws-c-common-config.cmake
  /usr/lib/aws-c-common/cmake/shared
  /usr/lib/aws-c-common/cmake/shared/aws-c-common-targets-release.cmake
  /usr/lib/aws-c-common/cmake/shared/aws-c-common-targets.cmake
  /usr/lib/cmake/AwsCheckHeaders.cmake
  /usr/lib/cmake/AwsCFlags.cmake
  /usr/lib/cmake/AwsSharedLibSetup.cmake
  /usr/lib/cmake/AwsLibFuzzer.cmake
  /usr/lib/cmake/AwsSIMD.cmake
  /usr/lib/cmake/AwsSanitizers.cmake
  /usr/lib/cmake/AwsFeatureTests.cmake
  /usr/lib/cmake/AwsTestHarness.cmake
  /usr/lib/cmake/AwsFindPackage.cmake
  /usr/include/aws
  /usr/include/aws/testing
  /usr/include/aws/common
  /usr/include/aws/testing/aws_test_harness.h
  /usr/include/aws/testing/aws_test_allocators.h
  /usr/include/aws/common/encoding.inl
  /usr/include/aws/common/array_list.inl
  /usr/include/aws/common/atomics.h
  /usr/include/aws/common/stdbool.h
  /usr/include/aws/common/atomics_gnu_old.inl
  /usr/include/aws/common/config.h
  /usr/include/aws/common/math.cbmc.inl
  /usr/include/aws/common/fifo_cache.h
  /usr/include/aws/common/platform.h
  /usr/include/aws/common/cache.h
  /usr/include/aws/common/time.h
  /usr/include/aws/common/thread_scheduler.h
  /usr/include/aws/common/xml_parser.h
  /usr/include/aws/common/byte_order.h
  /usr/include/aws/common/string.inl
  /usr/include/aws/common/command_line_parser.h
  /usr/include/aws/common/zero.h
  /usr/include/aws/common/allocator.h
  /usr/include/aws/common/date_time.h
  /usr/include/aws/common/math.gcc_overflow.inl
  /usr/include/aws/common/condition_variable.h
  /usr/include/aws/common/math.gcc_builtin.inl
  /usr/include/aws/common/zero.inl
  /usr/include/aws/common/ref_count.h
  /usr/include/aws/common/logging.h
  /usr/include/aws/common/ring_buffer.inl
  /usr/include/aws/common/clock.h
  /usr/include/aws/common/thread.h
  /usr/include/aws/common/uuid.h
  /usr/include/aws/common/exports.h
  /usr/include/aws/common/clock.inl
  /usr/include/aws/common/math.msvc.inl
  /usr/include/aws/common/math.inl
  /usr/include/aws/common/byte_buf.h
  /usr/include/aws/common/math.gcc_arm64_asm.inl
  /usr/include/aws/common/promise.h
  /usr/include/aws/common/atomics.inl
  /usr/include/aws/common/process.h
  /usr/include/aws/common/math.gcc_x64_asm.inl
  /usr/include/aws/common/string.h
  /usr/include/aws/common/package.h
  /usr/include/aws/common/lifo_cache.h
  /usr/include/aws/common/linked_hash_table.h
  /usr/include/aws/common/statistics.h
  /usr/include/aws/common/lru_cache.h
  /usr/include/aws/common/device_random.h
  /usr/include/aws/common/encoding.h
  /usr/include/aws/common/log_writer.h
  /usr/include/aws/common/atomics_msvc.inl
  /usr/include/aws/common/common.h
  /usr/include/aws/common/priority_queue.h
  /usr/include/aws/common/log_channel.h
  /usr/include/aws/common/math.h
  /usr/include/aws/common/error.inl
  /usr/include/aws/common/ring_buffer.h
  /usr/include/aws/common/hash_table.h
  /usr/include/aws/common/atomics_gnu.inl
  /usr/include/aws/common/error.h
  /usr/include/aws/common/mutex.h
  /usr/include/aws/common/environment.h
  /usr/include/aws/common/linked_list.h
  /usr/include/aws/common/file.h
  /usr/include/aws/common/assert.h
  /usr/include/aws/common/cpuid.h
  /usr/include/aws/common/stdint.h
  /usr/include/aws/common/linked_list.inl
  /usr/include/aws/common/predicates.h
  /usr/include/aws/common/system_info.h
  /usr/include/aws/common/macros.h
  /usr/include/aws/common/math.fallback.inl
  /usr/include/aws/common/resource_name.h
  /usr/include/aws/common/rw_lock.h
  /usr/include/aws/common/atomics_fallback.inl
  /usr/include/aws/common/byte_order.inl
  /usr/include/aws/common/log_formatter.h
  /usr/include/aws/common/array_list.h
  /usr/include/aws/common/task_scheduler.h
  /usr/include/aws/common/posix
  /usr/include/aws/common/posix/common.inl
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
nativesdk-aws-c-common: 104 installed and not shipped files. [installed-vs-shipped]
ERROR: nativesdk-aws-c-common-0.6.8-r0 do_package: Fatal QA errors found, failing task.`
I could see that the files have been installed correctly, but the do_package task fails to copy to the package staging directories.
Can you please help me?

GonzalezSGA, 2021-11-17:
Ok, I think that I have found what the problem is, the recipe is forcing the install prefix to `/usr` while the native SDK is installed in the default SDK path `/opt/dey/...`.
I have fixed it by:
1. setting the CMAKE_PREFIX_PATH variable to `$D${prefix}`, during the configure phase which sets the correct path depending on the package type (target or nativesdk).
2. Changing the library names from lib${PN}.so to lib${BPN}.so in order to user "aws-c-common" instead of "nativesdk-aws-c-common".

This is how it looks like:
`
# -*- mode: Conf; -*-
SUMMARY = "AWS C Common"
DESCRIPTION = "Core c99 package for AWS SDK for C. Includes cross-platform primitives, configuration, data structures, and error handling."

HOMEPAGE = "https://github.com/awslabs/aws-c-common"
LICENSE = "Apache-2.0"
PROVIDES = "aws/crt-c-common"

inherit cmake

LIC_FILES_CHKSUM = "file://LICENSE;md5=3b83ef96387f14655fc854ddc3c6bd57"

BRANCH ?= "main"
TAG ?= "v${PV}"
SRC_URI = "git://github.com/awslabs/aws-c-common.git;protocol=https;branch=${BRANCH};tag=${TAG};"

S = "${WORKDIR}/git"

CFLAGS:append = " -Wl,-Bsymbolic"
EXTRA_OECMAKE:append = " -DCMAKE_INSTALL_PREFIX=$D${prefix}"
EXTRA_OECMAKE += "-DBUILD_TESTING=OFF"
EXTRA_OECMAKE += "-DBUILD_SHARED_LIBS=ON"
EXTRA_OECMAKE += "-DCMAKE_BUILD_TYPE=Release"
OECMAKE_BUILDPATH = "${WORKDIR}/build"
OECMAKE_SOURCEPATH = "${S}"

FILES:${PN}     = "${libdir}/lib${BPN}.so.1.0.0 \
                   ${libdir}/lib${BPN}.so.1"
FILES:${PN}-dev = "${includedir}/aws/common/* \
                   ${includedir}/aws/testing/* \
                   ${libdir}/cmake/* \
                   ${libdir}/aws-c-common/* \
                   ${libdir}/lib${BPN}.so"
FILES:${PN}-dbg = "/usr/src/debug/aws-c-common/* \
                   ${libdir}/.debug/lib${BPN}.so.1.0.0"

BBCLASSEXTEND = "native nativesdk"
`
This change has to be extended to every recipe, and also change the CMAKE_PREFIX_PATH variable from `$D/usr` to `$D${prefix}`

nateglims, 2021-12-03:
Thanks for the issue, I will take a look. Is this using the Digi Embedded Yocto distribution?

rpcme, 2022-01-12:
Hello @GonzalezSGA could you kindly let us know @nateglims query? We want to ensure we're looking at this from the right working set.

rpcme, 2022-01-15:
Hello @GonzalezSGA please reopen this issue if it persists. I assume that it is resolved.

closed 2022-01-15 by rpcme