aws4embeddedlinux/meta-aws #262: Support for AWS KVS in meta-aws

qraleq, 2021-11-24:
Hi, I'm interested in whether there's a timeline for AWS KVS support in meta-aws?

aduskett, 2021-12-01:
Hello, while I can't directly answer that question, if you are willing to use buildroot I have a buildroot version of meta-aws that includes the kvs packages found here: https://github.com/aduskett/buildroot-aws-iot/

Might be worth a look! 

Adam

nateglims, 2021-12-02:
Hi Ivan, We do not have a timeline for KVS support right now. If you're able to discuss what you are trying to do, we might be able to help you though.

qraleq, 2021-12-02:
Hi, thank you both for your answers. 

We're trying to build AWS KVS and AWS IoT Greengrass on IP cameras running Yocto (for example AXIS CCTV cameras). 

@aduskett I'll take a look into `buildroot-aws-iot`, and it would be great if we could connect to see if you can help us to achieve our goal.

Thank you!

rpcme, 2021-12-06:
@qraleq Thank you for asking about KVS support.  I'm reaching out to folks internally to see how we can bridge support for it sooner rather than later.

qraleq, 2021-12-16:
Hi @rpcme, do you have any update on this?

rpcme, 2022-01-12:
Hello @qraleq we are currently scoping this.  Please stay tuned. I would also love to hear the urgency.  If this is blocking you, this will lift the priority and we might be able to shift resources to accomodate.

qraleq, 2022-01-12:
Hi @rpcme, 

At the moment we're using NVIDIA edge devices in combination with AXIS cameras to stream and analyze the CCTV streams. Unfortunately, with the global chip shortage, NVIDIA devices are impossible to get and we're considering moving the streaming service directly to the camera. That's where `meta-aws` might be really helpful to support KVS on the cameras themselves.

So, this is pretty high priority for us. Thank you!

rpcme, 2022-01-12:
@qraleq I would like one more clarification/scope question:  You are asking for KVS SDK for Data Streams and you would also like Yocto SDK support for cross compile environment?  Want to ensure we have the requirements right so what is delivered meets / exceeds your expectations.

qraleq, 2022-01-13:
Hi @rpcme, we're interested in AWS Kinesis Video Streams Producer SDK, and it would be great to have support for cross-compile using Yocto SDK.

rpcme, 2022-01-13:
@qraleq perfectly understand and sorry for the data streams confusion, I have for another reason kinesis data streams on my mind :D

rpcme, 2022-01-15:
Hello @qraleq I have started the recipe development. There are many things in the upstream cmake that's not "Yocto ready" so it will take some time to complete - perhaps another week but I'll try to bring it in.

Which language are you targeting? This is the status:

1/ PIC (completed, not yet in master-next)
2/ C (in progress, depends on PIC)
3/ C++ (in progress, depends on C)
4/ Java - I plan to do this last but if this is the language you're really targeting let me know.

qraleq, 2022-01-15:
Hi @rpcme, we would be interested in C and C++. Thank you so much for your help.

rpcme, 2022-01-15:
Hi @qraleq I completed C but have a hard stop in about 30 minutes.  You might be aware it is public holiday in US on Monday so I will likely not have anyone to do CR on Monday. In such case, you will have at least C by COB Tuesday.

rpcme, 2022-01-16:
Hello @qraleq I have "completed" the initial work on the recipes and done a PR. In the meantime, you can try from my fork if you desire. 

https://github.com/rpcme/meta-aws/tree/master/recipes-sdk/amazon-kvs-producer-sdk

I have not yet runtime verified but will at least make sure over the next couple hours that all runtime dependencies are in place, I need to build an image for that, and then after build an example to ensure I get KVS running video to the cloud ok (I don't expect issues with that once deps are verified).

Also, another point is the CMake files are not shipping the SDK cmake files, so please let me know if you expect to lean on those in an SDK context. Customer do use them in our Device SDKs so I have that as a follow on task for the implementation.

rpcme, 2022-01-16:
Ok it looks decen
[image]
t

rpcme, 2022-01-20:
Hello @qraleq this work was put on master-next and should be put onto master branch today (PR is in progress). Closing this request.

closed 2022-01-20 by rpcme