aws4embeddedlinux/meta-aws #265: missing runtime dependencies for aws-iot-device-sdk-python-v2

WolfWalter, 2021-11-27:
After building a honister based raspberry pi image with `aws-iot-device-sdk-python-v2` there where no `awscrt` files in my rootfs. When building hardknot there was a `build/tmp/work/raspberrypi4_64-poky-linux/solvispi-image/1.0-r0/rootfs/usr/lib/python3.9/site-packages/awscrt` folder that is now missing. Could it be that the line `RDEPENDS_${PN} += "python3 aws-crt-python"` in the recipe shoulb de `RDEPENDS:${PN} += "python3 aws-crt-python" for honister?

I didn't use yocto for quite a wihile, so the problem could totally be on my end.

nateglims, 2021-12-02:
It appears you are correct. We will fix this in #273

WolfWalter, 2021-12-02:
Ok. Great. But I think there are more recipes that are affected:
https://github.com/aws/meta-aws/search?q=RDEPENDS_%24%7BPN%7D

nateglims, 2021-12-02:
Good catch. I've updated the PR.

rpcme, 2022-01-15:
@nateglims has this issue been resolved and pulled into the appropriate release branches yet?

rpcme, 2022-01-15:
This issue should be fixed by this commit here https://github.com/aws/meta-aws/commit/10bfe0479a32efbd25e7eedd67f522b0a4c87db7

Please let us know if the issue persists @WolfWalter - actually, the fix was made some time ago but we neglected to get back to you via this issue. I am sorry for that.

closed 2022-01-15 by rpcme