aws4embeddedlinux/meta-aws #269: Dunfell Greengrass v2 2.4.0 runtime failure during Greengrass deployments

sgreen-GC, 2021-12-01:
Hi, we are attempting to build greengrass V2 into the latest openstlinux distribution but we're having an issue when deploying greengrass components to the running build.

The image builds fine, greengrass is installed, and when the image is booted, it starts up without any problems and greengrass starts up and runs (as a systemd service). 

However, whenever any GG component deployment runs, the device picks up the deployment but fails when trying to check the dependencies with the following (from greengrass.log):

```
2021-11-30T17:11:21.771Z [INFO] (pool-2-thread-6) com.aws.greengrass.deployment.activator.DeploymentActivator: merge-config. Rolling back failed deployment. {deploymentId=117a2d6e-6fcd-4b3c-b080-f4c89ca09624}
java.nio.file.NoSuchFileException: /init
        at java.base/sun.nio.fs.UnixException.translateToIOException(UnixException.java:92)
        at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:111)
        at java.base/sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:116)
        at java.base/sun.nio.fs.UnixFileAttributeViews$Basic.readAttributes(UnixFileAttributeViews.java:55)
        at java.base/sun.nio.fs.UnixFileSystemProvider.readAttributes(UnixFileSystemProvider.java:149)
        at java.base/sun.nio.fs.LinuxFileSystemProvider.readAttributes(LinuxFileSystemProvider.java:99)
        at java.base/java.nio.file.Files.readAttributes(Files.java:1764)
        at java.base/java.nio.file.FileTreeWalker.getAttributes(FileTreeWalker.java:219)
        at java.base/java.nio.file.FileTreeWalker.visit(FileTreeWalker.java:276)
        at java.base/java.nio.file.FileTreeWalker.walk(FileTreeWalker.java:322)
        at java.base/java.nio.file.Files.walkFileTree(Files.java:2717)
        at java.base/java.nio.file.Files.walkFileTree(Files.java:2797)
        at com.aws.greengrass.util.Utils.copyFolderRecursively(Utils.java:674)
        at com.aws.greengrass.lifecyclemanager.KernelAlternatives.prepareBootstrap(KernelAlternatives.java:317)
        at com.aws.greengrass.deployment.activator.KernelUpdateActivator.activate(KernelUpdateActivator.java:77)
        at com.aws.greengrass.deployment.DeploymentConfigMerger.updateActionForDeployment(DeploymentConfigMerger.java:150)
        at com.aws.greengrass.deployment.DeploymentConfigMerger.lambda$mergeInNewConfig$0(DeploymentConfigMerger.java:102)
        at com.aws.greengrass.lifecyclemanager.UpdateSystemPolicyService.runUpdateActions(UpdateSystemPolicyService.java:95)
        at com.aws.greengrass.lifecyclemanager.UpdateSystemPolicyService.lambda$startup$0(UpdateSystemPolicyService.java:166)
        at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
        at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
        at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
        at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
        at java.base/java.lang.Thread.run(Thread.java:829)

2021-11-30T17:11:21.778Z [INFO] (pool-2-thread-6) com.aws.greengrass.deployment.DeploymentDirectoryManager: Persist deployment metadata. {file=/greengrass/v2/deployments/117a2d6e-6fcd-4b3c-b080-f4c89ca09624/deployment_metadata.json, deploymentId=117a2d6e-6fcd-4b3c-b080-f4c89ca09624}
```

This happens for any deployment, even if the deployment is an 'empty' one (deploying only the same Nucleus version as already installed).

This seems similar to https://github.com/aws/meta-aws/issues/110 from what I can tell.

------
System and install info as follows:
Image was built using OpenSTLinux version: openstlinux-5.10-dunfell-mp1-21-11-17

meta-aws was added with the following version: 

`
$ git show
commit 2f72301f5a73279c4d2f12fc6218876629666e06 (HEAD -> dunfell, origin/dunfell)
`

A new layer with recipe to add greengrass-bin was added which resulted in the 
following in the build manifest:

corretto-11-bin cortexa7t2hf-neon-vfpv4 11.0.12.7.1-r0.0
greengrass-bin cortexa7t2hf-neon-vfpv4 2.4.0-r0.0

Image was copied to an SD card and boots normally. On the rootfs, the following can be seen:

`
root@stm32mp1:~# ls /greengrass/v2/
LICENSE  alts  bin  cli_ipc_info  config  deployments  ipc.socket  logs  packages  plugins  telemetry  work
`

And from systemctl:

```
root@stm32mp1:~# systemctl status greengrass --no-pager
● greengrass.service - Greengrass Core
     Loaded: loaded (/lib/systemd/system/greengrass.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2021-12-01 15:01:41 UTC; 2min 37s ago
   Main PID: 552 (sh)
     Memory: 98.6M
     CGroup: /system.slice/greengrass.service
             ├─552 /bin/sh /greengrass/v2/alts/current/distro/bin/loader
             └─560 java -Dlog.store=FILE -Dlog.store=FILE -Droot=/greengrass/v2 -jar /greengrass/v2/alts/current/distro/lib/Greengrass.jar --setup-system-service false

Dec 01 15:01:41 stm32mp1 sh[552]: JVM options: -Dlog.store=FILE -Droot=/greengrass/v2
Dec 01 15:01:41 stm32mp1 sh[552]: Nucleus options: --setup-system-service false
Dec 01 15:02:04 stm32mp1 sh[560]: SLF4J: Failed to load class "org.slf4j.impl.StaticMDCBinder".
Dec 01 15:02:04 stm32mp1 sh[560]: SLF4J: Defaulting to no-operation MDCAdapter implementation.
Dec 01 15:02:04 stm32mp1 sh[560]: SLF4J: See http://www.slf4j.org/codes.html#no_static_mdc_binder for further details.
Dec 01 15:02:12 stm32mp1 sh[560]: Launching Nucleus...
Dec 01 15:02:14 stm32mp1 sh[560]: AWS libcrypto resolve: searching process and loaded modules
Dec 01 15:02:14 stm32mp1 sh[560]: AWS libcrypto resolve: found static aws-lc HMAC symbols
Dec 01 15:02:14 stm32mp1 sh[560]: AWS libcrypto resolve: found static aws-lc libcrypto 1.1.1 EVP_MD symbols
Dec 01 15:02:47 stm32mp1 sh[560]: Launched Nucleus successfully.
```

Please let me know if I can provide further details.

hancin, 2021-12-01:
It seems like our greengrass images call `ln -r` to create a relative symlink while [nucleus assume it to be an absolute path](https://github.com/aws-greengrass/aws-greengrass-nucleus/blob/f14442c90d944dfc6d5288f9de8ba842b37acc27/src/main/java/com/aws/greengrass/lifecyclemanager/KernelAlternatives.java#L316) (as seen in #110)

To workaround you can recreate the symlink `/greengrass/v2/alts/current` so it points to the absolute path `/greengrass/v2/alts/init` instead of relative path.

sgreen-GC, 2021-12-02:
Great thanks! That's fixed the issue with deployments.
We're now getting a previously unseen error with our own component (compared to a manual GG install) but obviously that's for us to investigate!

rpcme, 2022-01-12:
Hello @sgreen-GC , closing this issue.  Although closed, I would love to hear how you got along in the integration.

sgreen-GC, 2022-01-13:
That's all good, thanks for the help everyone!
We've sidestepped using the greengrass-bin recipe for now for various reasons, but we're making good progress with a manually installed version on the same hardware. Our application seems to be working well in normal working conditions so we're now pushing at some kind of hardware/software limit with stress tests (The Nucleus Telemetry Emitter component has been a useful tool here).