aws4embeddedlinux/meta-aws #282: Is there any AMI ready to run Yocto linux on AWS?

t0ny-peng, 2021-12-27:
Hi there,

We are trying to move some of the local test farm to AWS, and we would like to run Yocto Linux on AWS. The only tutorial I found on line is [Getting Started with Cloud-Native Automotive Software Development](https://catalog.us-east-1.prod.workshops.aws/v2/workshops/12f31c93-5926-4477-996c-d47f4524905d/en-US/20-container-build-pipeline), however it only has AMI for `us-east-1`. There's also an article [TipsAndTricks/Running YP Image On AWS](https://wiki.yoctoproject.org/wiki/TipsAndTricks/Running_YP_Image_On_AWS). Before I dive deeper into it, just want to know if AWS already provides an Yocto Linux AMI?

Thanks

t0ny-peng, 2022-01-13:
@rpcme Thanks for the reply. Yes we do have out local `meta-` layer but as an appetizer I just want to build the vanilla Yocto image and deploy it on AWS. Will also include the `meta-aws` layer. Any recommended tutorial?

Thanks

rpcme, 2022-01-15:
For building AMI, the meta-aws project doesn't yet provide anything. However, there are tutorials for various device targets for meta-aws (mostly greengrass) in the meta-aws-demos and meta-aws-ci repos.

Please feel free to reach out directly to me on LinkedIn and will be able to carry on the convo from there?
https://www.linkedin.com/in/richardelberger/

I will resolve this issue for now since we will not be providing this mechanism from meta-aws itself, but there is potential to solve it via the meta-aws-ci repo and we can likely help with that.

closed 2022-01-15 by rpcme