aws4embeddedlinux/meta-aws #288: Greengrassd : failed to initialize with reason Address already in use

uyrusali, 2022-01-14:
## Greengrassd : failed to initialize with reason Address already in use

Greengrass Daemon starts with this error, also get the same when trying to deploy the group. ```runtime.log:```

```
2022-01-14T08:46:00.341+03:00][INFO]-File never started syncing or already stopped syncing.	{"src": "/etc/resolv.conf", "dst": "/home/nvidia/greengrass/ggc/packages/1.10.5/dns/etc/resolv.conf"}
[2022-01-14T08:46:00.342+03:00][INFO]-Finding mounted cgroups.
[2022-01-14T08:46:00.382+03:00][INFO]-Starting file sync.	{"src": "/etc/resolv.conf", "dst": "/home/nvidia/greengrass/ggc/packages/1.10.5/dns/etc/resolv.conf"}
[2022-01-14T08:46:00.971+03:00][INFO]-Created worker.	{"functionArn": "arn:aws:lambda:::function:GGShadowSyncManager", "workerId": "31e240b0-185c-4cc9-76ab-b394b365c434", "pid": 8506}
[2022-01-14T08:46:01.003+03:00][INFO]-Created worker.	{"functionArn": "arn:aws:lambda:::function:GGShadowService", "workerId": "f7b5289c-9685-4c69-630f-625f513aa077", "pid": 8578}
[2022-01-14T08:46:01.004+03:00][INFO]-Created worker.	{"functionArn": "arn:aws:lambda:::function:GGTES", "workerId": "c4e8865e-8a5b-49f1-6b33-8f3de414e884", "pid": 8463}
[2022-01-14T08:46:01.015+03:00][INFO]-Created worker.	{"functionArn": "arn:aws:lambda:::function:GGDeviceCertificateManager", "workerId": "4c507d75-0444-49c9-61ea-b7367177f78c", "pid": 8566}
[2022-01-14T08:46:01.019+03:00][INFO]-Created worker.	{"functionArn": "arn:aws:lambda:::function:GGStreamManager:1", "workerId": "7ebf7891-ba1a-489b-629a-3dc3d627eccd", "pid": 8559}
[2022-01-14T08:46:01.087+03:00][INFO]-Created worker.	{"functionArn": "arn:aws:lambda:::function:GGIPDetector:1", "workerId": "45517478-64b4-4090-4ba6-67312f197b53", "pid": 8669}
[2022-01-14T08:46:01.111+03:00][INFO]-Created worker.	{"functionArn": "arn:aws:lambda:::function:GGSecretManager:1", "workerId": "45a73cf8-e69a-40e4-6fda-4f180b9b0b71", "pid": 8658}
[2022-01-14T08:46:01.111+03:00][INFO]-Created worker.	{"functionArn": "arn:aws:lambda:::function:GGConnManager", "workerId": "b5feba9d-85a8-4f6e-542d-3c6a2bdf5a82", "pid": 8715}
[2022-01-14T08:46:01.119+03:00][INFO]-Created worker.	{"functionArn": "arn:aws:lambda:::function:GGCloudSpooler:1", "workerId": "ce8aa2b8-b504-4ffe-690f-43b4211ef53b", "pid": 8746}
[2022-01-14T08:46:05.101+03:00][INFO]-Handled functions request.	{"functionName": "arn:aws:lambda:::function:GGStreamManager:1", "invocationId": ""}
[2022-01-14T08:46:09.719+03:00][INFO]-Worker Memory Usage (KB).	{"workerId": "7ebf7891-ba1a-489b-629a-3dc3d627eccd", "funcArn": "arn:aws:lambda:::function:GGStreamManager:1", "memSize": 4194304, "memUsed": 130556}
[2022-01-14T08:46:09.736+03:00][ERROR]-Stopping all workers as lifecycle manager fails to start.	{"error": "worker with 7ebf7891-ba1a-489b-629a-3dc3d627eccd failed to initialize with reason Address already in use"}
[2022-01-14T08:46:09.736+03:00][INFO]-Stopping all workers.
[2022-01-14T08:46:09.776+03:00][WARN]-Connection was closed by client.	{"errorString": "EOF", "clientId": "4c507d75-0444-49c9-61ea-b7367177f78c", "invokerArn": "arn:aws:lambda:::function:GGDeviceCertificateManager"}
[2022-01-14T08:46:09.776+03:00][WARN]-Connection was closed by client.	{"errorString": "EOF", "clientId": "4c507d75-0444-49c9-61ea-b7367177f78c", "invokerArn": "arn:aws:lambda:::function:GGDeviceCertificateManager"}
[2022-01-14T08:46:09.776+03:00][WARN]-Connection was closed by client.	{"errorString": "EOF", "clientId": "45a73cf8-e69a-40e4-6fda-4f180b9b0b71", "invokerArn": "arn:aws:lambda:::function:GGSecretManager:1"}
[2022-01-14T08:46:09.776+03:00][WARN]-Connection was closed by client.	{"errorString": "EOF", "clientId": "c4e8865e-8a5b-49f1-6b33-8f3de414e884", "invokerArn": "arn:aws:lambda:::function:GGTES"}
[2022-01-14T08:46:09.777+03:00][WARN]-Connection was closed by client.	{"errorString": "EOF", "clientId": "f7b5289c-9685-4c69-630f-625f513aa077", "invokerArn": "arn:aws:lambda:::function:GGShadowService"}
[2022-01-14T08:46:09.783+03:00][WARN]-Connection was closed by client.	{"errorString": "EOF", "clientId": "ce8aa2b8-b504-4ffe-690f-43b4211ef53b", "invokerArn": "arn:aws:lambda:::function:GGCloudSpooler:1"}
[2022-01-14T08:46:09.783+03:00][WARN]-Connection was closed by client.	{"errorString": "EOF", "clientId": "ce8aa2b8-b504-4ffe-690f-43b4211ef53b", "invokerArn": "arn:aws:lambda:::function:GGCloudSpooler:1"}
[2022-01-14T08:46:09.784+03:00][WARN]-Connection was closed by client.	{"errorString": "EOF", "clientId": "45517478-64b4-4090-4ba6-67312f197b53", "invokerArn": "arn:aws:lambda:::function:GGIPDetector:1"}
[2022-01-14T08:46:09.918+03:00][INFO]-Worker Max Memory Usage (KB).	{"workerId": "4c507d75-0444-49c9-61ea-b7367177f78c", "funcArn": "arn:aws:lambda:::function:GGDeviceCertificateManager", "memUsed": 22680}
[2022-01-14T08:46:10.115+03:00][INFO]-Worker Max Memory Usage (KB).	{"workerId": "ce8aa2b8-b504-4ffe-690f-43b4211ef53b", "funcArn": "arn:aws:lambda:::function:GGCloudSpooler:1", "memUsed": 20104}
[2022-01-14T08:46:10.315+03:00][INFO]-Worker Memory Usage (KB).	{"workerId": "45517478-64b4-4090-4ba6-67312f197b53", "funcArn": "arn:aws:lambda:::function:GGIPDetector:1", "memSize": 32768, "memUsed": 22296}
[2022-01-14T08:46:10.515+03:00][INFO]-Worker Max Memory Usage (KB).	{"workerId": "f7b5289c-9685-4c69-630f-625f513aa077", "funcArn": "arn:aws:lambda:::function:GGShadowService", "memUsed": 18248}
[2022-01-14T08:46:10.715+03:00][INFO]-Worker Max Memory Usage (KB).	{"workerId": "45a73cf8-e69a-40e4-6fda-4f180b9b0b71", "funcArn": "arn:aws:lambda:::function:GGSecretManager:1", "memUsed": 22492}
[2022-01-14T08:46:10.915+03:00][INFO]-Worker Max Memory Usage (KB).	{"workerId": "c4e8865e-8a5b-49f1-6b33-8f3de414e884", "funcArn": "arn:aws:lambda:::function:GGTES", "memUsed": 19812}
[2022-01-14T08:46:19.785+03:00][WARN]-Connection was closed by client.	{"errorString": "EOF", "clientId": "31e240b0-185c-4cc9-76ab-b394b365c434", "invokerArn": "arn:aws:lambda:::function:GGShadowSyncManager"}
[2022-01-14T08:46:19.785+03:00][WARN]-Connection was closed by client.	{"errorString": "EOF", "clientId": "31e240b0-185c-4cc9-76ab-b394b365c434", "invokerArn": "arn:aws:lambda:::function:GGShadowSyncManager"}
[2022-01-14T08:46:19.796+03:00][WARN]-Connection was closed by client.	{"errorString": "EOF", "clientId": "b5feba9d-85a8-4f6e-542d-3c6a2bdf5a82", "invokerArn": "arn:aws:lambda:::function:GGConnManager"}
[2022-01-14T08:46:19.916+03:00][INFO]-Worker Max Memory Usage (KB).	{"workerId": "b5feba9d-85a8-4f6e-542d-3c6a2bdf5a82", "funcArn": "arn:aws:lambda:::function:GGConnManager", "memUsed": 19244}
[2022-01-14T08:46:20.115+03:00][INFO]-Worker Max Memory Usage (KB).	{"workerId": "31e240b0-185c-4cc9-76ab-b394b365c434", "funcArn": "arn:aws:lambda:::function:GGShadowSyncManager", "memUsed": 20040}
[2022-01-14T08:46:20.118+03:00][INFO]-Stop syncing file.	{"src": "/etc/resolv.conf", "dst": "/home/nvidia/greengrass/ggc/packages/1.10.5/dns/etc/resolv.conf"}
[2022-01-14T08:46:20.118+03:00][INFO]-Lifecycle manager is stopped.
[2022-01-14T08:46:20.118+03:00][ERROR]-Deployment failed.	{"deploymentId": "b3bfde2d-37f3-4b42-aa0a-6092616fe9cb", "errorString": "worker with 7ebf7891-ba1a-489b-629a-3dc3d627eccd failed to initialize with reason Address already in use"}
[2022-01-14T08:46:20.261+03:00][INFO]-Updating deployment shadow.	{"deploymentId": "b3bfde2d-37f3-4b42-aa0a-6092616fe9cb", "groupId": "4f0c5c5a-8c10-494a-827f-ac7801e160dc"}
[2022-01-14T08:46:20.341+03:00][INFO]-Successfully updated deployment shadow.	{"deploymentId": "b3bfde2d-37f3-4b42-aa0a-6092616fe9cb", "groupId": "4f0c5c5a-8c10-494a-827f-ac7801e160dc"}
[2022-01-14T08:46:20.343+03:00][INFO]-Start to process new deployment messages from the cloud.
[2022-01-14T08:46:20.344+03:00][INFO]-End processing deployment messages from the cloud.
root@nvidia-desktop:/home/nvidia/greengrass/ggc/var/log/system# 
```

Checked the processes using this port and tried to kill as suggested in this (link)[https://www.intel.com/content/www/us/en/developer/articles/technical/aws-cloud-to-edge-pipeline.html] by:
```
sudo netstat -tulpn | grep 8000
sudo kill -9 <port-number>
```
But it happens that the process using this port is greengrassd itself.

The device is Nvidia Jetson Nano. Any help is appreciated.

rpcme, 2022-01-15:
Hello @uyrusali Can you please provide the bitbake output at the beginning of the build that includes all the repos and their associated hashes, as well as local.conf?  Although it's extremely unlikely the problem has anything to do with that, I want to ensure we're working from the same starting point.

rpcme, 2022-01-20:
Closing due to no response, please reopen if still an issue.

closed 2022-01-20 by rpcme