aws4embeddedlinux/meta-aws #305: Updating to aws-iot-device-sdk-python-v2_1.7.0 implicitly includes python3-modules

mlasch, 2022-02-23:
The update from `aws-iot-device-sdk-python-v2_1.5.4` to `aws-iot-device-sdk-python-v2_1.7.0` includes `python3-modules` into the build. I can't have the whole Python3 distribution in my image, because it simply won't fit. There seems to be some Yocto magic happen in the background including `python3-modules` since the dependencies did not change.

- I'm on Yocto 3.1.14 and I'm using the latest `dunfell` backport branch.
- My current workaround: Change `RDEPENDS` from depending on `python3` to `python3-core` then everything works as before.

rpcme, 2022-03-04:
@mlasch thank you for the notification, I apologize for the response delay. Will look at getting this sorted today.

mlasch, 2022-03-16:
@rpcme thanks for looking into the issue. It looks like there are two options:
1. All recipes which depend on Python3 should use explicitly `python3-core` + the relevant modules from the Python std. library. This would greatly help building constraint devices with limited flash, also would reduce the image size for everybody.
2. Keep `python3` as a dependency as it is now, but then everybody who uses a constraint device has to maintain separate `.ppappend` files in their distribution like:
 ```
RDEPENDS_${PN}_remove = "python3"
RDEPENDS_${PN}_append = " python3-core"
 ```
Option 2. is our current workaround.

thomas-roos, 2022-05-09:
Hi, now `python3-core` is used instead of python3 where possible.
https://github.com/aws4embeddedlinux/meta-aws/commit/de60da566a16b1af8d585ff7d4d48290169d8f46

closed 2022-05-09 by thomas-roos

mlasch, 2022-05-09:
Thanks a lot for your upstream fix, we can now remove our workarounds.