aws4embeddedlinux/meta-aws #313: Request to backport `device sdk: remove tag from SRC_URI to release branches`

mlasch, 2022-03-16:
In order to build an image which includes `aws-c-cal` changes from commit c6624193311f6cf191b2876e0828946005455c1d would be needed on Dunfell. Can we backport commit c6624193311f6cf191b2876e0828946005455c1d to earlier, still supported, releases e.g. `dunfell`?
I'm on Yocto 3.1.14 and when building with the `archiver` class the current `dunfell-next` branch fails:
```
ERROR: aws-c-cal-0.5.11-r0 do_ar_original: Cannot determine archive names for original source because 'name' URL parameter is unset in more than one URL. Add it to at least one of these: git://github.com/awslabs/aws-c-common.git;protocol=https;branch=main;tag=v0.6.8 git://github.com/awslabs/aws-c-cal.git;protocol=https;branch=main;tag=v0.5.11
```
However, there is one caveat: when cherry-picking, there are conflicts because `master` went on with other changes.

rpcme, 2022-03-18:
@mlasch Thank you for your report - will discuss with @nateglims today.

rpcme, 2022-04-22:
@thomas-roos could you look at this one as well please?

thomas-roos, 2022-05-09:
Hi, commit is backported to dunfell: https://github.com/aws4embeddedlinux/meta-aws/commit/1d926e9408434590b4d406acb8809ced558bd944
I hope this solves this issue.

closed 2022-05-09 by thomas-roos