aws4embeddedlinux/meta-aws #315: Incompatible layer Error with Yocto Rocko

bilalmalik76, 2022-03-25:
I am trying to integrate meta-aws layer with Yocto project which is based on Rocko branch. I am getting an error of incompatible with core layer. There is no branch for rocko in meta-aws. is it possible to use meta-aws with Yocto based on Rocko branch: 

```
ERROR: Unable to start bitbake server
ERROR: Last 10 lines of server log for this session (/home/bilal/firmware-gen2/poky/build/bitbake-cookerdaemon.log):
  File "/home/bilal/firmware-gen2/poky/bitbake/lib/bb/server/process.py", line 433, in _startServer
    self.cooker = bb.cooker.BBCooker(self.configuration, self.featureset)
  File "/home/bilal/firmware-gen2/poky/bitbake/lib/bb/cooker.py", line 197, in __init__
    self.initConfigurationData()
  File "/home/bilal/firmware-gen2/poky/bitbake/lib/bb/cooker.py", line 356, in initConfigurationData
    self.databuilder.parseBaseConfiguration()
  File "/home/bilal/firmware-gen2/poky/bitbake/lib/bb/cookerdata.py", line 316, in parseBaseConfiguration
    raise bb.BBHandledException
bb.BBHandledException
ERROR: Layer meta-aws is not compatible with the core layer which only supports these series: rocko (layer is compatible with kirkstone)
```

bilalmalik76, 2022-04-02:
Hi, @rpcme I am using MSM8909 from Qualcomm. I ported the master branch to Rocko and able to build it with Yocto Rocko version.

rpcme, 2022-04-22:
Okay this seems ok now. If you would like to contribute the Rocko port (just do PR to rocko release branch) we will review it.

closed 2022-04-22 by rpcme