aws4embeddedlinux/meta-aws #316: 'aws-c-iot' is unbuildable - dunfell

elockman, 2022-04-01:
Yocto is no longer building aws-c-iot, with the following error:

```
ERROR: Nothing RPROVIDES 'aws-c-iot' (but /home/circleci/project/yocto/meta-xeal/recipes-mesh/images/xeal-mesh.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'aws-c-iot' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['aws-c-iot']
```
aws-c-iot is added with **CORE_IMAGE_EXTRA_INSTALL += "aws-c-iot"**

This is on the **dunfell** branch

Is this failing due to the **git://github** URI [here](https://github.com/aws/meta-aws/blob/fd9c90772d8c11e7a589adf992a37dfcbd4ec90b/recipes-sdk/aws-c-iot/aws-c-iot_0.0.7.bb#L15)?  I've changed URI's to **https://github** in my recipes.

elockman, 2022-04-20:
When trying a bbappend, the original recipes are not found?

```
Parsing of 2271 .bb files complete (0 cached, 2271 parsed). 3378 targets, 361 skipped, 0 masked, 0 errors.
WARNING: No recipes available for:
  /home/circleci/project/yocto/meta-xeal/recipes-sdk/aws-c-iot/aws-c-iot_0.0.7.bbappend
  /home/circleci/project/yocto/meta-xeal/recipes-sdk/aws-crt-cpp/aws-crt-cpp_0.15.0.bbappend
NOTE: Resolving any missing task queue dependencies

```

elockman, 2022-04-20:
changing github repository didn't help either
```
#git clone https://github.com/aws/meta-aws -b dunfell yocto/meta-aws
git clone https://github.com/aws4embeddedlinux/meta-aws -b dunfell yocto/meta-aws

```

elockman, 2022-04-20:
To make this work, I had to add all aws .bb recipes to my build.
I'm still unsure why CORE_IMAGE_EXTRA_INSTALL no longer works.

rpcme, 2022-04-22:
Wow I don't know how I totally missed this, well I have been traveling heavy the last few weeks.

Let me see if Thomas can take a look.

thomas-roos, 2022-04-22:
@elockman tried to reproduce this error and did not succeed - may you try this simple steps if they work compare them with your yocto build environment:
```
mkdir yoctotest
cd yoctotest/
git clone git://git.yoctoproject.org/poky -b  dunfell
git clone https://github.com/aws4embeddedlinux/meta-aws -b dunfell
git clone https://github.com/openembedded/meta-openembedded.git -b dunfell
source poky/oe-init-build-env
bitbake-layers add-layer ../meta-openembedded/meta-oe
bitbake-layers add-layer ../meta-openembedded/meta-python
bitbake-layers add-layer ../meta-openembedded/meta-networking
bitbake-layers add-layer ../meta-aws
echo 'CORE_IMAGE_EXTRA_INSTALL += "aws-c-iot"' >> conf/local.conf 
bitbake core-image-minimal
```
```runqemu```
check if libs /usr/lib/libaws* are exsisting

elockman, 2022-05-16:
Thank you! This resolved the issue.  It may have been due to cloning **aws4embeddedlinux** instead of **aws**.

closed 2022-05-16 by elockman

thomas-roos, 2022-05-17:
Glad I could help you!