aws4embeddedlinux/meta-aws #335: aws-iot-device-client fails with "invalid pointer"

perezmeyer, 2022-05-17:
I have build the hardknott version on a VAR-DART-MX8M-PLUS board. Build was straightforward, but after configuring /etc/aws-iot-device-client.json and trying to run the app I get:

```bash
$ aws-iot-device-client --config-file /etc/aws-iot-device-client.json 
munmap_chunk(): invalid pointer
```
Is there anything I can do in order to debug this further?

thomas-roos, 2022-05-20:
Hi, we are currently try to reproduce and fix this error. Do you have a special aws-iot-device-client.json or is it the default one?

perezmeyer, 2022-05-20:
Hi Thomas! I had to create it by hand using the parameters specified by mi IT colleague.

Here is an anonymized copy:

```bash
cat /etc/aws-iot-device-client/aws-iot-device-client.conf 
{
        "endpoint": "a3kyhtb3rxud7g-ats.iot.us-east-1.amazonaws.com",
        "cert": "/home/root/keys/<a code>-certificate.pem.crt",
        "key": "/home/root/keys/<a code>-public.pem.key",
        "root-ca": "/home/root/keys/AmazonRootCA1.pem",
        "thing-name": "dart",
        "logging": {
                "level": "DEBUG",
                "type": "FILE",
                "file": "/var/log/aws-iot-device-client/aws-iot-device-client.log"
        },
        "jobs": {
                "enabled": false,
                "handler-directory": "/root/handler"
        },
        "tunneling": {
                "enabled": false
        },
        "device-defender": {
                "enabled": true,
                "interval": 300
        },
        "fleet-provisioning": {
                "enabled": false,
                "template-parameters": "<replace_with_template_parameters>",
                "template-name": "<replace_with_template_name>",
                "csr-file": "<replace_with_csr_file_path>",
                "device-key": "<replace_with_device_private_key_file_path>"
        },
        "samples": {
                "pub-sub": {
                        "enabled": false,
                        "publish-topic": "/topic/dc/pub",
                        "publish-file": "/var/log/pub.txt",
                        "subscribe-topic": "/topic/dc/sub",
                        "subscribe-file": "/var/log/sub.txt"
                }
        },
        "config-shadow": {
                "enabled": false
        },
        "sample-shadow": {
                "enabled": false,
                "shadow-name": "<replace_with_shadow_name>",
                "shadow-input-file": "<replace_with_shadow_input_file_path>",
                "shadow-output-file": "<replace_with_shadow_output_file_path>"
        }
}
```

perezmeyer, 2022-05-20:
Maybe worth to mention: I tried to debug it but the packaging does not creates a suitable -dbg package. Well the package is there but literally empty (no, there are no hidden files).

thomas-roos, 2022-05-27:
Could investigate on your problem in master and hardknott - both are working for me (after having same issues) if you:
- go to AWS IoT website
 download package and take the files to /etc/aws-iot-device-client/
- download root ca
` curl https://www.amazontrust.com/repository/AmazonRootCA1.pem > root-CA.crt` as well
- Thing name is in start.sh
- **IMPORTANT**: go to THINGS select Certificate edit policy - for testing * * should be ok otherwise will see disconnect ERROR in logfile
- have a nameserver eg. `echo nameserver 8.8.4.4 >> /etc/resolv.conf `

- have a minimal configuration like this (default one will cause that segmentation fault - as descriped [here](https://github.com/awslabs/aws-iot-device-client) between the lines): 
```
root@qemux86-64:/etc/aws-iot-device-client# cat aws-iot-device-client.conf 
{
        "endpoint": "XXX.amazonaws.com",
        "cert": "/etc/aws-iot-device-client/XXX.cert.pem",
        "key": "/etc/aws-iot-device-client/XXX.private.key",
        "root-ca": "/etc/aws-iot-device-client/root-CA.crt",
        "thing-name": "XXX",
        "logging": {
                "level": "DEBUG",
                "type": "FILE",
                "file": "/var/log/aws-iot-device-client.log"
        }
}
```

proper file/dir permissions:
```
chmod 700 /etc/aws-iot-device-client/
chmod 644 /etc/aws-iot-device-client/*
chmod 600 /etc/aws-iot-device-client/XXX.private.key
```
**Starting:**
`aws-iot-device-client --config-file /etc/aws-iot-device-client/aws-iot-device-client.conf  --enable-sdk-logging`

**Logfiles:**
tail -f /var/log/aws-iot-device-client.log 
tail -f /var/log/aws-iot-device-client/sdk.log

P.S: I plan an upgrade to 1.6. in master soon, I will fix the debug problem there as well. There must be `RelWithDebInfo`: Release optimizations with debug info as `CMAKE_BUILD_TYPE` in the recipe

thomas-roos, 2022-05-27:
please reopen if this will not fix your problem

closed 2022-05-27 by thomas-roos

perezmeyer, 2022-05-27:
Thanks, I'll try ASAP.

thomas-roos, 2022-06-03:
Hi, great to hear that things are improved. But as I'm not an expert for `aws-iot-device-client` I would friendly ask you to open a bug/discussion at their github page **https://github.com/awslabs/aws-iot-device-client**
The only thing that might help is the correct setting of this - but I guess you did that.
`IMPORTANT: go to THINGS select Certificate edit policy - for testing * * should be ok otherwise will see disconnect ERROR in logfile`

perezmeyer, 2022-06-03:
Thanks a lot for your help!